espressif/esp-idf #6096: tight loop in nvs_flash triggering WDT

mzimmers, 2020-11-09:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it): v4.1
- Build System:         [|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Windows|]
- (Windows only) environment type: [|ESP Command Prompt|].
- Using an IDE?: [No|]
- Power Supply:         [USB|external 5V||Battery]

## Problem Description

A tight loop in nvs_flash component triggers WDT. The loop can be found at:
C:\esp-idf\components\spi_flash\cache_utils.c:141-144.

### Expected Behavior

Should return before WDT triggers. I'd recommend a small delay in this loop. (vTaskDelay(1);)
### Actual Behavior
Triggers WDT on occasion. 
### Steps to reproduce

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

## Debug Logs

```
E (23767) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (23767) task_wdt:  - IDLE0 (CPU 0)
E (23767) task_wdt: Tasks currently running:
E (23767) task_wdt: CPU 0: Wifi
E (23767) task_wdt: CPU 1: IDLE1
E (23767) task_wdt: Aborting.
abort() was called at PC 0x400dce37 on core 0

ELF file SHA256: 1b5c05c3dc5a8fcf

Backtrace: 0x4008a1bd:0x3ffb0660 0x4008a579:0x3ffb0680 0x400dce37:0x3ffb06a0 0x4008507d:0x3ffb06c0 0x400856f3:0x3ffe1fa0 0x40086571:0x3ffe1fc0 0x40089f71:0x3ffe1fe0 0x4008637b:0x3ffe2000 0x4008655d:0x3ffe2030 0x4011b9d5:0x3ffe2050 0x4011a3d1:0x3ffe2070 0x4011a7a9:0x3ffe2090 0x401171cd:0x3ffe2100 0x40119960:0x3ffe2150 0x4011b8b3:0x3ffe21b0 0x40116719:0x3ffe21d0 0x40116ac9:0x3ffe2200 0x400e43c1:0x3ffe2220 0x400f8846:0x3ffe2370 0x400f8ab5:0x3ffe23c0 0x4008b831:0x3ffe23e0
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)
[files.zip](https://github.com/espressif/esp-idf/files/5513105/files.zip)

Alvin1Zhang, 2020-11-10:
Thanks for reporting, we will look into.

vanwykc, 2021-10-03:
Any update / workaround on this issue?

I'm experiencing this same issue. This exact same loop is the reason for random BtController / WiFi / fopen / fclose etc Task WDT issues prohibiting me from getting a product production ready for over a week now.

Please help!

fromeijn, 2023-04-19:
im still seeing the same problem in v4.3.4 when opening a (second) file with fopen

```
0x40081b9e: panic_abort at ~/esp-idf/components/esp_system/panic.c:404
0x40092d45: esp_system_abort at ~/esp-idf/components/esp_system/system_api.c:112
0x4009844a: abort at ~/esp-idf/components/newlib/abort.c:46
0x40159010: task_wdt_isr at ~/esp-idf/components/esp_common/src/task_wdt.c:182 (discriminator 1)
0x40083db1: _xt_lowint1 at ~/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x4008b932: spi_flash_disable_interrupts_caches_and_other_cpu at ~/esp-idf/components/spi_flash/cache_utils.c:160 (discriminator 1)
0x4008c825: cache_disable at ~/esp-idf/components/spi_flash/spi_flash_os_func_app.c:70
0x4008c831: spi1_start at ~/esp-idf/components/spi_flash/spi_flash_os_func_app.c:99
0x40090cbd: spiflash_start_default at ~/esp-idf/components/spi_flash/esp_flash_api.c:111
0x4008c641: esp_flash_read at ~/esp-idf/components/spi_flash/esp_flash_api.c:691
0x40150e5f: esp_partition_read at ~/esp-idf/components/spi_flash/partition.c:419
0x400ffaa3: spiffs_api_read at ~/esp-idf/components/spiffs/spiffs_api.c:36
0x40102ee9: spiffs_phys_rd at ~/esp-idf/components/spiffs/spiffs/src/spiffs_cache.c:161
0x40100ec3: spiffs_obj_lu_find_entry_visitor at ~/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:166
0x40102c15: spiffs_obj_lu_find_free_obj_id at ~/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:2212
0x40100006: SPIFFS_open at ~/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:250
0x400ff80d: vfs_spiffs_open at ~/esp-idf/components/spiffs/esp_spiffs.c:508
0x400d6153: esp_vfs_open at ~/esp-idf/components/vfs/vfs.c:376 (discriminator 3)
0x401c629c: _fopen_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fopen.c:129
0x401c6345: fopen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fopen.c:168
```

igrr, 2023-04-19:
@fromeijn could you please attach your sdkconfig file?

fromeijn, 2023-04-19:
Thanks igrr for looking in to this, see attached 
[sdkconfig.log](https://github.com/espressif/esp-idf/files/11272388/sdkconfig.log)

igrr, 2023-04-19:
```
CONFIG_SPI_FLASH_YIELD_DURING_ERASE=y
CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=20
CONFIG_SPI_FLASH_ERASE_YIELD_TICKS=1
CONFIG_SPI_FLASH_WRITE_CHUNK_SIZE=8192
```

It seems that the options required to yield from flash operations are enabled, so it's odd that the task watchdog is triggering.

Do you think you could reduce your application to a smaller example which reproduces this issue? This would help us recreate the issue on our end. Also, how frequently does the issue happen, or how much time it takes to reproduce it?

AxelLin, 2023-04-19:
> ```
> CONFIG_SPI_FLASH_YIELD_DURING_ERASE=y
> CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=20
> CONFIG_SPI_FLASH_ERASE_YIELD_TICKS=1
> CONFIG_SPI_FLASH_WRITE_CHUNK_SIZE=8192
> ```
> 
> It seems that the options required to yield from flash operations are enabled, so it's odd that the task watchdog is triggering.

The original issue report (https://github.com/espressif/esp-idf/issues/6096#issue-739336614) already has above settings (see the files.zip).
This issue has been there for years.

AxelLin, 2023-05-17:
https://github.com/espressif/esp-idf/issues/4984#issuecomment-1156445364

AxelLin, 2023-05-24:
> ```
> CONFIG_SPI_FLASH_YIELD_DURING_ERASE=y
> CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=20
> CONFIG_SPI_FLASH_ERASE_YIELD_TICKS=1
> CONFIG_SPI_FLASH_WRITE_CHUNK_SIZE=8192
> ```
> 
> It seems that the options required to yield from flash operations are enabled, so it's odd that the task watchdog is triggering.
> 

FYI, https://github.com/espressif/esp-idf/issues/11032#issuecomment-1560531918 has the same settings as above, but still hit WDT.

AxelLin, 2023-08-20:
@igrr @Alvin1Zhang 
Do you have any progress for this issue?
(This was reported since 2020, and there are several related reports linked in above comments)

grginho, 2023-08-24:
Hello everyone,

I'm also experiencing the watchdog timer getting triggered in read and fopen functions, using the LittleFS file system.
Backtrace says that the code is stuck at esp-idf/components/spi_flash/cache_utils.c::167
        while (!s_flash_op_can_start) {
            // Busy loop and wait for spi_flash_op_block_func to disable cache
            // on the other CPU
        }
Regards,

packn366, 2023-08-27:
I had a similar experience.
Reading NVS causes WDT elapsed at "while (!s_flash_op_can_start)".
What I found is it happens much less times after I updated IDF from 4.4 to 5.1.