espressif/esp-idf #6097: vTaskList does not show Affinity after upgrade to freertos-10

AxelLin, 2020-11-10:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v4.3-dev-1720-g494a124d961c
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

This looks like a regression after switch to freertos-10
Even when CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID is set, vTaskList does
not include information about Affinity.

### Expected Behavior
If build with older freertos, the vTaskList will also show affinity when CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID is set.
### Actual Behavior
affinity information is not display when CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID is set,

### Steps to reproduce
Set
CONFIG_FREERTOS_USE_TRACE_FACILITY=y
CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS=y
CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID=y
CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS=y

Then use examples/system/console to check.

Alvin1Zhang, 2020-11-10:
Thanks for reporting, we will look into.

AxelLin, 2020-11-10:
I sent the fix in https://github.com/espressif/esp-idf/pull/6099

Alvin1Zhang, 2020-11-10:
Thanks for the contribution.

AxelLin, 2020-12-14:
I thought sending a pull request can make it faster to fix the issue, but seems it does not help.

mahavirj, 2020-12-14:
> I thought sending a pull request can make it faster to fix the issue, but seems it does not help.

@AxelLin Apologies for the delay here. We always encourage contributions from community, so your PR is definitely very helpful. I am putting it under internal review queue, please expect update on issue/PR in next few weeks (probably with next github sync).

Mahavir

closed 2021-01-05 by espressif-bot