espressif/esp-idf #6100: Enhancement of heap_caps_print_heap_info()

ammaree, 2020-11-10:
**Is your feature request related to a problem? Please describe.**
The info presented by the default function is difficult to understand
The way the info is displayed makes it tedious to detect subtle changes in the memory map.
 
**Describe the solution you'd like**
Ability to write our own replacement for heap_caps_print_heap_info() allowing tabular layout of the info, easing debugging and optimisation of memory use.
Access to a header file that exposes the required definitions, macros, structures, variables and functions

**Describe alternatives you've considered**
Firstly, we have added a function to the end of heap_caps.c that solves the problem but brings it own complexities with keeping our local copy of the IDF in sync with master.
We have tried building the function in a module of our own, but that brought out the complexity/confusion of multiple different versions of "heap_t" defined in "multi_heap.c" and "heap_private.h" 

**Additional context**
The replacement function we coded are as follows:
```
void	halMCU_ReportMemory(uint32_t caps) {
    printfx("From Addr | Size  | Used  | Free  | MinFr | TotBlk|UsedBlk|FreeBlk| LF_Blk|  0x%04X\n", caps) ;
    multi_heap_info_t info ;
    heap_t * heap ;
    SLIST_FOREACH(heap, &registered_heaps, next) {
        if (heap_caps_match(heap, caps)) {
            multi_heap_get_info(heap->heap, &info);
            printfx("0x%08x %'#7u ",heap->start, heap->end - heap->start) ;
            printfx("%'#7u %'#7u %'#7u %'#7u %'#7u %'#7u %'#7u\n",
				info.total_allocated_bytes, info.total_free_bytes, info.minimum_free_bytes,
				info.total_blocks, info.allocated_blocks, info.free_blocks, info.largest_free_block);
        }
    }
    heap_caps_get_info(&info, caps);
    printfx("===== Totals ===== %'#7u %'#7u %'#7u %'#7u %'#7u %'#7u %'#7u\n\n",
			info.total_allocated_bytes, info.total_free_bytes, info.minimum_free_bytes,
			info.total_blocks, info.allocated_blocks, info.free_blocks, info.largest_free_block);
}
```

ammaree, 2020-11-10:
Example of the output being produced
[image]

Alvin1Zhang, 2020-11-11:
Thanks for raising this feature request.

projectgus, 2020-11-11:
Hi @ammaree,

Thanks for letting us know about this shortcoming, and sharing your fix.

If I understand correctly, to implement the `halMCU_ReportMemory` function using the ESP-IDF public API, the only required addition is a way to iterate all the registered heaps. Is that right?

Conceptually, the idea of "heaps caps" was to abstract the need to worry about individual regions of memory (the individual "heaps") in this case. In order for us to better improve the heap API in the future, can you please explain a bit about the "subtle changes in the memory map" and the problems you experienced that lead to needing to dig into each heap's individual statistics?

ammaree, 2020-11-12:
Hi @projectgus 

What is required you would know better than me, the iteration of heaps but also some clarity on the specific structure format to use (where duplicates exist).

Maybe my specific choice of words being "memory map" was not ideal, meant the memory in general as related to area addresses and sizes, as well as fragmentation ie blocks/sizes etc.

Everything goes back to the available memory, identifying when and by which process it is used blocks and (hopefully) released. Objective is to minimise dynamic allocation and have as few individual/separate blocks as possible, but with as large a size as possible, available after bootup.

Only issue we have encountered relating to memory map is still unclear to us but we have worked around it. Although this is not critical I can provide some background if you are interested.
* To improve detection of pointer errors we have a set of macros (INRANGE_SRAM, INRANGE_FLASH, INRANGE_MEM etc) that we use in debug code to evaluate the validity of pointers passed as parameters.
* To facilitate the capture and logging of most/all messages produced we have developed our own "syslog" module (replacing the 'log' component) that transforms message format and supplement info by adding the calling task name, function name and CoreID details.
* The messages are buffered in the unused RTC Slow RAM area and, once a UDP link is available to the syslog host, are also sent there for later analysis.
* Supporting the syslog module we have an enhanced & extended printfx module

The process flow that resulted in detecting the memory map anomaly is as follows.
1. ESP_LOGx messages are intercepted and forwarded to our syslog module.
2. Syslog module transforms (discard TAG etc) and enhance with additional info then..
3. Calls printfx() function with ends up in an xPrintString() function where the INRANGE_MEM() macro detects and invalid (from the best of our understanding) pointer.
4. The offending message originates from the wifi library, relates to LOGI, tag 'wifi' and content "rssi" but with a pointer value of 0xFFFFFFE6

Once again, the long memory map story is not serious, but any help you can provide in understanding will be of value.

Resolving the original feature request is of value, although we are currently working around it by adding the additional function at the end of heap_caps.c

projectgus, 2020-11-16:
Hi @ammaree ,

Thanks, this is very helpful. The features you've added sound very impressive, the process for checking any address-like things in log messages is clever!

Will take all of this into consideration when looking at what we can do to improve heap metadata functions.

ammaree, 2022-08-05:
@o-marshmallow 

Any update on this functionality. Currently modifying heap_caps.c by adding as last line....

```#include "../../../heap_caps_extend.inc"```

o-marshmallow, 2022-08-05:
Hi @ammaree ,

We currently don't have any plan to modify this function in the short term. However, a cleaner way to integrate your implementation would be wrap `heap_caps_print_heap_info()`. As it is a public function, it can be wrapped thanks to a linker flag that you can add in your project's `CMakeLists.txt` file:
```
idf_build_set_property(COMPILE_OPTIONS "-Wl,--wrap=heap_caps_print_heap_info" APPEND)
```

In one of your project's file, you can then define the following:
```
/* Add the following only if you plan to call the implementation, else, just discard it */
extern void __real_heap_caps_print_heap_info( uint32_t caps );

void _wrap_heap_caps_print_heap_info( uint32_t caps ) {
    /* Your implementation of  heap_caps_print_heap_info */
}

/* Other functions in the file */
```
This should let you transparently override the implementation without modifying IDF explicitely.

ammaree, 2022-08-06:
I do not believe wrapping the function will work since the function requires access to private functionality contained only within heap_caps.c

o-marshmallow, 2022-08-08:
Hi @ammaree ,

I see that you use `registered_heaps`, in theory this variable is private, but as you can see in `heap_caps_init.c` and `heap/heap_private.h`, it is not static, so you could use:
```
#include <sys/queue.h>

extern SLIST_HEAD(registered_heap_ll, heap_t_) registered_heaps;
``` 
In you file containing the wrapper.

Do you use any symbol that is completely private and prevent you from using wrappers?

ammaree, 2022-08-11:
@o-marshmallow 

I am trying to implement your suggestion, not much luck.
Added component ```heap``` as a PRIV_REQUIRE
Added ```#include <sys/queue.h>``` but had multiple compile errors. Replaced with ```#include "../heap_private.h"``` and some of the errors went away.

Now sitting with:

> In file included from /Users/andremaree/Dropbox/devs/ws/z-comp/hal_esp32/hal_mcu.c:137:
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/include/../heap_private.h:30:5: error: unknown type name 'multi_heap_lock_t'
   30 |     multi_heap_lock_t heap_mux;
      |     ^~~~~~~~~~~~~~~~~
In file included from /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/include/../heap_private.h:13,
                 from /Users/andremaree/Dropbox/devs/ws/z-comp/hal_esp32/hal_mcu.c:137:
/Users/andremaree/Dropbox/devs/ws/z-comp/hal_esp32/hal_mcu.c:139:19: error: redefinition of 'struct registered_heap_ll'
  139 | extern SLIST_HEAD(registered_heap_ll, heap_t_) registered_heaps;
      |                   ^~~~~~~~~~~~~~~~~~
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/include/../heap_private.h:41:19: note: originally defined here
   41 | extern SLIST_HEAD(registered_heap_ll, heap_t_) registered_heaps;
      |                   ^~~~~~~~~~~~~~~~~~
/Users/andremaree/Dropbox/devs/ws/z-comp/hal_esp32/hal_mcu.c:139:48: error: conflicting types for 'registered_heaps'; have 'struct registered_heap_ll'
  139 | extern SLIST_HEAD(registered_heap_ll, heap_t_) registered_heaps;
      |                                                ^~~~~~~~~~~~~~~~
In file included from /Users/andremaree/Dropbox/devs/ws/z-comp/hal_esp32/hal_mcu.c:137:
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/include/../heap_private.h:41:48: note: previous declaration of 'registered_heaps' with type 'struct registered_heap_ll'
   41 | extern SLIST_HEAD(registered_heap_ll, heap_t_) registered_heaps;
      |                                                ^~~~~~~~~~~~~~~~

Any suggestions?

ammaree, 2022-08-11:
@o-marshmallow 

Got the errors down to the single
>In file included from /Users/andremaree/Dropbox/devs/ws/z-comp/hal_esp32/hal_mcu.c:140:
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/include/../heap_private.h:30:5: error: unknown type name 'multi_heap_lock_t'
   30 |     multi_heap_lock_t heap_mux;
      |     ^~~~~~~~~~~~~~~~~

Now to resolve that one, needs```'MULTI_HEAP_FREERTOS``` to be defined....

ammaree, 2022-08-11:
@o-marshmallow 

Resolves last item, coded code from heap CmakeFiles.txt to my component cMakeFiles.txt to define ```MULTI_HEAP_RTOS```

Will close now.

closed 2022-08-11 by ammaree

o-marshmallow, 2022-08-12:
@ammaree Thanks for coming back with the solution :+1: