espressif/esp-idf #6101: emac_dm9051: no mem for receive buffer

pabs159, 2020-11-10:
Hello!

I am running the eth2ap example under the master branch using the esp32WROVER-E. I can get the "examples/ethernet/basic" to work with the DM9051 board. I get back a valid IP and my router shows the device as a listed device. 

When using the eth2ap example I also can see the ESP on the network along with my phone once I connect to the ESP. but shortly after any service tries to phone home I get this error:

"emac_dm9051: no mem for receive buffer" 

I have searched for this error code and have not found anything on it. I could easily be missing something since i dont see anyone else with this error. All I change in the example is the pins for the DM9051. Im using SPI host 2 (VSPI). This is my config example layout:

(2) SPI Host Number
(18) SPI SCLK GPIO number
(23) SPI MOSI GPIO number
(19) SPI MISO GPIO number
(5) SPI CS GPIO number
(20) SPI clock speed (MHz)
(4) Interrupt GPIO number
(15) PHY Reset GPIO number
(1) PHY Address
(eth2ap) Wi-Fi SSID
(password) Wi-Fi Password
(1) WiFi channel
(4) Maximum STA connections

Output after execution:

 (13682) eth_example: Wi-Fi AP got a station connected
W (17012) wifi:<ba-add>idx:4 (ifx:1, 7a:e0:aa:d2:f5:3c), tid:0, ssn:2, winSize:64
E (26932) eth_example: WiFi send packet failed: 12309
E (28052) emac_dm9051: no mem for receive buffer

Any insight is much appreciated. Thank you for your time.

suda-morris, 2020-11-11:
This is an "out of memory" error, according to code here: https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_mac_dm9051.c#L419
Although I don't recommend people to playing this `eth2ap` example with any SPI-Ethernet due to significant performance issue, still want to ask if changing `buffer = heap_caps_malloc(length, MALLOC_CAP_DMA);` to `buffer = malloc(length);` can help?

pabs159, 2020-11-11:
@suda-morris 

We are talking to Davicom about this too. We only need 1/2MB to get across it (is that even possible with SPI?). We did change the source file as you pointed out. Getting "wifi send packet failed" now. Which we also have gotten before, just eventually that DM9051 error your pointed to would throw. I know that issue is in reference to the macro "FLOW_CONTROL_WIFI_SEND_DELAY". I put that at 500ms which by any modern standard I think its getting too extreme. 

Would the LAN8270 work better? (RMI). We ordered one and if from the developers side that is better anyways we can just do that. 

Appreciate your time.

KaeLL, 2020-11-11:
I'd like to extend @pabs159's question by asking @suda-morris which Ethernet chip do you personally think pairs the most nicely with the ESP32? Currently I'm stuck with W5100, pabs159 is trying out a couple, but perhaps you got a __personal__ recommendation (personal in bold in order make it clear you're not talking on behalf of espressif, if that matters).

suda-morris, 2020-11-12:
@pabs159 Yes, sure, this `eth2ap` example is designed for ESP32 EMAC + PHY scheme.  Have you checked the error code regarding to the "wifi send packet failed"? i.e. the return value of `esp_wifi_internal_tx` function call.

@KaeLL For ESP32, I would say, we'd better use its internal EMAC if the GPIO numbers are not limited in your project. Internal EMAC is much powerful than any SPI-Ethernet module. For other ESP chips which doesn't embedded with a EMAC IP, SPI-Ethernet module is a good choice. The problem is, it's not an easy task to support as many modules as possible, currently the DM9051 driver is a good begging, showing out esp-eth component has the possibility to support different kinds of Ethernet solution. BTW, We already have some work done for W5500 module, but still need some time before it's released to Github.

closed 2021-06-23 by suda-morris

tsctrl, 2022-04-29:
LAN8270 with other 3 spi module. Having the same issue. dont know where to fix.
Did nott have issue with psram disabled
worver module devkit c idf 4.3 

E (177169) emac_esp32: no mem for receive buffer
E (177366) emac_esp32: no mem for receive buffer
E (177366) emac_esp32: no mem for receive buffer
E (177366) emac_esp32: no mem for receive buffer

king1688, 2022-08-22:
any update for this problem? I have the same problem with LAN8720 PHY

thiagohd, 2022-09-30:
> 

Same problem here

kostaond, 2022-10-10:
Duplicate of #9308

apple521, 2022-11-30:
> 

I have same problem. I used MicroPython. and create >3 udp thread,  run a moment out error.
E (13436) esp.emac: no mem for receive buffer
E (13536) esp.emac: no mem for receive buffer
E (13586) esp.emac: no mem for receive buffer
E (13656) esp.emac: no mem for receive buffer
E (13776) esp.emac: no mem for receive buffer
E (13886) esp.emac: no mem for receive buffer
E (14006) esp.emac: no mem for receive buffer
E (14116) esp.emac: no mem for receive buffer

kostaond, 2022-11-30:
@apple521 what IDF version do you use? Could you please try master and let me know if you still see the issue? The probability that this issue occurs was lowered by [this commit](https://github.com/espressif/esp-idf/commit/785f154f5652b353c460ec47cb90aa404207918c). However, the memory is limited resource and you can always get into this situation when your application uses a lot of memory as mentioned in my previous comment.

apple521, 2022-12-01:
> @apple521 what IDF version do you use? Could you please try master and let me know if you still see the issue? The probability that this issue occurs was lowered by [this [...]

I used LAN8720 and IDF v4.4.3. Only 3 socket thread is fail .
esp.emac: no mem for receive buffer
Myber ,run long time, out of memory ，always happen.
OMG!Doesn't anyone use this to make products?

kostaond, 2022-12-01:
> Only 3 socket thread is fail.

You can get fail with only one socket if there is high traffic and your application uses too much memory.

Try the ESP-IDF master branch where more optimized approach to allocate memory for receive frames is introduced as I mentioned above.

Or try to enable [external SPI RAM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html) if you have ESP32 module with PSRAM.

Please also check what is memory footprint of your Micropython application. You might import some big library. You can try ti optimize or can switch to plain C.

> OMG!Doesn't anyone use this to make products?

Memory has been always limited resource. Please read or [watch some documentary](https://youtu.be/sw0VfmXKq54) of how developers had to be smart during e.g. game development in old days to overcome such limitations. Please also understand that ESP32 is not (and it is not trying to be) PC with 128GB of memory.

apple521, 2022-12-01:
> You can get fail with only one socket if there is high traffic and your application uses too much memory.
>
> Try the ESP-IDF master branch where more optimized approach to allocate memory for receive [...]

my app is very easy. i test micropyhon.
C program is Ok .
And wifi is OK。
I try again.

apple521, 2022-12-01:
> You can get fail with only one socket if there is high traffic and your application uses too much memory.
>
> Try the ESP-IDF master branch where more optimized approach to allocate memory for receive [...]

I may have found the reason，
before: 
wlan = network.WLAN(network.STA_IF)
lan = network.LAN(mdc = Pin(23), mdio = Pin(18), power = Pin(5), phy_type = network.PHY_LAN8720, phy_addr=0) 
lan.active(True)  
now:
#wlan = network.WLAN(network.STA_IF)
lan = network.LAN(mdc = Pin(23), mdio = Pin(18), power = Pin(5), phy_type = network.PHY_LAN8720, phy_addr=0) 
lan.active(True)  
 I test 2 hours, it's OK .
by the way. 

Can I use wlan and lan at the same time?

kostaond, 2022-12-01:
Yes, you can. The issue is gone because you disabled WLAN, i.e. you utilize less memory since you don't process WLAN traffic. It is still about overall memory utilization. Please try the first two tips from [my previous comment](https://github.com/espressif/esp-idf/issues/6101#issuecomment-1333405699).

apple521, 2022-12-02:
> Yes, you can. The issue is gone because you disabled WLAN, i.e. you utilize less memory since you don't process WLAN traffic. It is still about overall memory utilization. Please try the first two [...]

Thank you for your help. My app is working properly. I can do other functions now.