espressif/esp-idf #6103: Unable to read json

solankigithub, 2020-11-11:
Hello,
I am trying to use the ESP-IDF json library but getting error. Below is my code:
```

#define CONFIG " {	\"config\":1, \
					\"uart\":{\"baud_rate\":9600, \"data_bits\":8,  \"parity\":0, \"stop_bit\":1}, \
					\"modbus_requests\": \
						[ \
						  {\"slave_id\":1, \"function_code\":4, \
							\"start_addressMSB\":0, \"start_addressLSB\":18, \
							\"rangeMSB\":0, \"rangeLSB\":1 \
							\"crcMSB\":36, \"crcLSB\":15} \
						] \
				  }"
void app_main(void)
{

cJSON *root = cJSON_Parse(CONFIG);
cJSON *uart = 	cJSON_GetObjectItem(root, "uart");
int baud_rate = cJSON_GetObjectItem(uart, "baud_rate")->valueint;
ESP_LOGI(TAG, "baud: %d", baud_rate);
}

```
The program compiles good. but when i run it, ESP32 crashes.

Please help.

Thanks
Sandeep

mcdavsco, 2020-11-11:
@solankigithub  Your JSON is invalid as you're missing a comma after ``"rangeLSB\":1`

Try:
`#define CONFIG " {	\"config\":1, \
					\"uart\":{\"baud_rate\":9600, \"data_bits\":8,  \"parity\":0, \"stop_bit\":1}, \
					\"modbus_requests\": \
						[ \
						  {\"slave_id\":1, \"function_code\":4, \
							\"start_addressMSB\":0, \"start_addressLSB\":18, \
							\"rangeMSB\":0, \"rangeLSB\":1, \
							\"crcMSB\":36, \"crcLSB\":15} \
						] \
				  }"`

Alvin1Zhang, 2020-11-18:
@mcdavsco Thanks for the notes and thanks @solankigithub for reporting, has your issue resolved? Thanks.

Alvin1Zhang, 2021-01-05:
Thanks for reporting, feel free to reopen with more updates.

closed 2021-01-05 by Alvin1Zhang