espressif/esp-idf #6105: Custom 128 bit UUID for Characteristics

Zeni241, 2020-11-12:
I am following the gatt_server_service_table example at :
    https://github.com/espressif/esp-‎idf/tree/release/v4.1/examples/bluetooth/bluedroid/ble/gatt_server_service_table
Though I am able to use custom 128 bit Service UUID (thanks to guidance by @chegewara in “_Custom ‎UUID for BLE services (IDFGH-4121) #5988_”), **I am unable to use custom UUID for Characteristics.** ‎
No matter what even value I set for “`GATTS_CHAR_UUID_TEST_A`” , I get the same UUID ‘`FF01`’. My ‎guess is it has something to do with :‎
`static const uint16_t character_declaration_uuid   = ESP_GATT_UUID_CHAR_DECLARE;‎`
but I don’t know how to manipulate it.‎

chegewara, 2020-11-12:
Maybe some code you have?

Zeni241, 2020-11-12:
I am using the almost all the code in the above example, except I changed 
`static const uint16_t GATTS_CHAR_UUID_TEST_A = 0xFF01; ` to
`static const uint16_t GATTS_CHAR_UUID_TEST_A = 0x1201`

chegewara, 2020-11-12:
One question, what device do you use to connect to esp32 during development?
Did you clear cache in that device or refresh services/characteristics UUID?

Zeni241, 2020-11-12:
I am using Web Bluetooth API to connect to ESP32. I use this page to discover device, services, characteristic. chrome://bluetooth-internals/#devices 
Besides I have developed my own web page using Web Bluetooth API to connect to bluetooth device.

Zeni241, 2020-11-12:
Besides the Characteristic UUIDs I see are using base service UUID :
0000ff01-0000-1000-8000-00805f9b34fb
0000ff02-0000-1000-8000-00805f9b34fb
While my service UUID is 020012ac-4202-c1ad-eb11-680fe014a236