espressif/esp-idf #6106: After burn security boot fw, then cann't burn again.

asd617140123, 2020-11-12:
## Environment

- Development Kit:   
ESP32-(POE version)
- Module or chip used:  
ESP-wridev-E
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
    v4.-dev-1561-g357a27760
- Operating System:     [Windows|Linux|macOS]
linux
## Problem Description
I use tools/testapps/security/secure_boot sample,I follow the README to config my sdk.  When I generate the .pem, then burn the image. I can not reburn again.

## Debug Logs

```
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05ff0 ( 24560) map
I (116) esp_image: segment 1: paddr=0x00016018 vaddr=0x3ffb0000 size=0x02794 ( 10132) load
I (121) esp_image: segment 2: paddr=0x000187b4 vaddr=0x40080000 size=0x00404 (  1028) load
I (125) esp_image: segment 3: paddr=0x00018bc0 vaddr=0x40080404 size=0x07458 ( 29784) load
I (147) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x1406c ( 82028) map
I (178) esp_image: segment 5: paddr=0x00034094 vaddr=0x4008785c size=0x0269c (  9884) load
I (183) esp_image: segment 6: paddr=0x00036738 vaddr=0x00000000 size=0x09898 ( 39064)
I (205) boot: Loaded app from partition at offset 0x10000
I (205) boot: Disabling RNG early entropy source...
I (217) cpu_start: Pro cpu up.
I (217) cpu_start: Starting app cpu, entry point is 0x40080eb8
I (0) cpu_start: App cpu up.
I (231) cpu_start: Pro cpu start user code
I (231) cpu_start: cpu freq: 160000000
I (231) cpu_start: Application information:
I (236) cpu_start: Project name:     secure_boot
I (241) cpu_start: App version:      v4.3-dev-1561-g357a27760-dirty
I (248) cpu_start: Compile time:     Nov 11 2020 21:42:58
I (254) cpu_start: ELF file SHA256:  d8b204ef7e25e7dc...
I (260) cpu_start: ESP-IDF:          v4.3-dev-1561-g357a27760-dirty
I (267) heap_init: Initializing. RAM available for dynamic allocation:
I (274) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (280) heap_init: At 3FFB2FB0 len 0002D050 (180 KiB): DRAM
I (286) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (293) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (299) heap_init: At 40089EF8 len 00016108 (88 KiB): IRAM
I (306) spi_flash: detected chip: generic
I (310) spi_flash: flash io: dio
W (314) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (328) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

Example to check Secure Boot status
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 3, 2MB external flash
I (359) example_secure_boot: Checking for secure boot v2..
I (369) example_secure_boot: Secure boot v2 not enabled. Enable Secure Boot V2 in menuconfig, build & flash again.
I (379) example_secure_boot: Checking for secure boot v1..
I (379) example_secure_boot: Secure Boot V1 not enabled. Enable Secure Boot in menuconfig, build & flash again.

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

projectgus, 2020-11-13:
Hi @asd617140123 ,

> tools/testapps/security/secure_boot 

The tools/testapps are unfortunately not examples. These are intended for use in our internal automated testing of ESP-IDF, not for users.

You can enable Secure Boot on any example (see under "examples" directory) by following the steps in the ESP-IDF documentation:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html

(This link is to the Secure Boot V2 docs, as I see you have an ESP32 revision 3 chip.)

> When I generate the .pem, then burn the image. I can not reburn again.

Can you please explain exactly what happens when you "can not reburn again"? i.e. exactly what commands do you run, and exactly what is the output?

The Secure Boot feature places some limitations on reflashing the firmware, once it's enabled the device is permanently locked into Secure Boot mode and will only boot correctly signed bootloader and app (for Secure Boot V2). However according to the log you posted, hardware Secure Boot is not actually enabled on the device.

One thing to note is that the bootloader binary not included in the default "idf.py flash" if Secure Boot is enabled. This is to avoid accidentally enabling Secure Boot. You need to run `idf.py bootloader-flash` or the command printed by `idf.py bootloader` in order to flash the bootloader.

asd617140123, 2020-11-16:
When I try to burn the image again, I see the below result. Device can halt at search /dev/ttyUSB1.
```
esptool.py --chip esp32 -p /dev/ttyUSB1 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 blink.bin
esptool.py v3.0-dev
Serial port /dev/ttyUSB1
Connecting........_____....._____.....
```

And I have try to press EN button then press BOOT button then release EN button, I can see the "Download mode is disabled."

```
Example to check Secure Boot status
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 3, 2MB external flash
I (359) example_secure_boot: Checking for secure boot v2..
I (369) example_secure_boot: Secure boot v2 not enabled. Enable Secure Boot V2 in menuconfig, build & flash again.
I (379) example_secure_boot: Checking for secure boot v1..
I (379) example_secure_boot: Secure Boot V1 not enabled. Enable Secure Boot in menuconfig, build & flash again.
Download mode is disabled. Restart with GPIO0 high.
Download mode is disabled. Restart with GPIO0 high.

```

Since I burn the sample Secure Boot what i thought. I can not into download mode.

thank your reply.

projectgus, 2020-12-01:
Hi @asd617140123 ,

Thanks for being patient while someone got back to you.

I'm afraid that by default when you enable Secure Boot V2 in menuconfig, the default configuration will "Permanently disable ROM download mode" (see screenshot):

[image]

This is intended to reduce the attack surface for any potential security issue. The idea is that this will be enabled when you go to production with your firmware, not during development.

It is possible to disable this flag in the config (under "Potentially Insecure Options"), but as you've already flashed the firmware I'm afraid it won't be able to undo this - you will need a new ESP32 to continue with.

projectgus, 2021-01-04:
Please let us know if you don't think the explanation above explains what you're seeing.

closed 2021-01-04 by projectgus

asd617140123, 2021-01-04:
Ok , Thanks. I will use another esp32 module to develop.