espressif/esp-idf #6107: Webserver doesn't work through SoftAp after reconnection (for certain devices)

will-emmerson, 2020-11-12:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32
- IDF version: master, v4.3-dev-1720-g494a124d9
- Build System:  idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:   Linux
- Using an IDE?: CLion
- Power Supply:  USB

## Problem Description

I have combined softAP and webserver examples so that webserver returns a simple response over the softAP. This initially works fine, but if I turn wifi off and on again on the client then reconnect, it can't access the server and just hangs indefinitely.

### Expected Behavior

Client should be able to access server after reconnection

### Actual Behavior

It reconnects fine to the softAP according to the logs but then can't see the server. When it's working I can ping the server from the client and see that port 80 is open, when it's not I can't do either of those things.
But this problem only seems to happen for certain devices - I've tested it on an android phone, an iphone and linux pc and the problem only happens on the two phones. It also doesn't happen 100% of the time, it sometimes takes a few goes to trigger it.
The problem can be 'fixed' by connecting to a different esp32 running the same firmware, then disconnecting and coming back to the original one, which will now connect properly.
Resetting the esp doesn't fix the problem.

There are a few other github issues that are similar but seem to be different problems:
https://github.com/espressif/esp-idf/issues/5768
https://github.com/espressif/esp-idf/issues/3851

It seems like a connection is being left open somewhere but I don't know why.

What I've tried:
- using `httpd_config_t lru_purge_enable`
- sending `Connection: close` with http response

I've tried viewing debug logs and setting CONFIG_LWIP_NETIF_DEBUG, CONFIG_LWIP_SOCKETS_DEBUG, CONFIG_LWIP_IP_DEBUG and packets do seem to be going between 192.168.4.1 and 192.168.4.2 when it's 'stuck' but I don't really understand it.

I feel like I must be doing something stupid because I can't be the only person who's trying to run a server with a softAP and connecting via a phone.

### Steps to reproduce

1. Connect to softap with a phone
2. Disable and re-enable wifi on client
3. Try and use server - may have to repeat steps a few times to trigger problem

### Code to reproduce this issue
https://github.com/will-emmerson/softap_server

## Debug Logs

- client connects to ap and access server ok
- client disconnects and reconnects and can still access it ok
- client disconnects and reconnects and now can't access server

```
I (770) wifi:mode : softAP (30:ae:a4:d3:48:5d)
I (770) wifi:Total power save buffer number: 16
I (770) wifi:Init max length of beacon: 752/752
I (780) wifi:Init max length of beacon: 752/752
I (4870) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (4870) wifi:station: 0c:cb:85:d4:a2:37 join, AID=1, bgn, 20
I (5180) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
W (5880) wifi:<ba-add>idx:4 (ifx:1, 0c:cb:85:d4:a2:37), tid:0, ssn:34, winSize:64
I (10510) WEBSERVER: handled root
I (19720) wifi:station: 0c:cb:85:d4:a2:37 leave, AID = 1, bss_flags is 134243, bss:0x3ffbabec
I (19730) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
W (19730) wifi:<ba-del>idx
I (28040) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (28040) wifi:station: 0c:cb:85:d4:a2:37 join, AID=1, bgn, 20
I (28350) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
W (29060) wifi:<ba-add>idx:4 (ifx:1, 0c:cb:85:d4:a2:37), tid:0, ssn:41, winSize:64
I (33450) WEBSERVER: handled root
I (54260) WEBSERVER: handled root
I (56850) wifi:station: 0c:cb:85:d4:a2:37 leave, AID = 1, bss_flags is 134243, bss:0x3ffbabec
I (56850) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
W (56850) wifi:<ba-del>idx
I (64360) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (64360) wifi:station: 0c:cb:85:d4:a2:37 join, AID=1, bgn, 20
I (64650) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
W (65390) wifi:<ba-add>idx:4 (ifx:1, 0c:cb:85:d4:a2:37), tid:0, ssn:25, winSize:64
----- can't access root -----
```

YouDONG-ESP, 2020-12-09:
@will-emmerson When the issue happened, it show no log right?
Could you tell me your phone modle? This issue wasn't reproduced with iphone 11 and oneplus 7 from my side.

will-emmerson, 2020-12-14:
There are packets coming to and from 192.168.4.1 when CONFIG_LWIP_IP_DEBUG is enabled but other than thatyt, nothing.
I have tried it with both Moto G6 Play and Nokia 5.3 and they both have the same problem.

YouDONG-ESP, 2020-12-15:
@will-emmerson Root cause found. Disconnect wifi won't clean tcp socket. So if you do it multiple times, new sockets are created but old socket didn't close, it will exceed the max listen sockets number.

AxelLin, 2020-12-15:
@YouDONG-ESP 
Can you post the fix for verify?

YouDONG-ESP, 2020-12-16:
You can print the `maxfd` inside httpd_server() before select() to verify this issue. It's not a bug, We want to keep the possibility that wifi is disconnected for a short time while tcp is not disconnected.  You should handle it in your applicaiton by closing the socket while `WIFI_EVENT_AP_STADISCONNECTED` is rasing in `wifi_event_handler()`

Alvin1Zhang, 2021-01-12:
Thanks for reporting, feel free to reopen.

closed 2021-01-12 by Alvin1Zhang

GreenLoofa, 2021-03-05:
@YouDONG-ESP Can you provide an example with how one would close the socket in the `wifi_event_handler()`? Would that be handled by `httpd_stop()`?

YouDONG-ESP, 2021-03-08:
Hi @GreenLoofa, `httpd_stop()` won't do the job. socketfd is stored in `handle` of ` httpd_start(httpd_handle_t *handle, const httpd_config_t *config)`.