espressif/esp-idf #6110: Cmake Error

Thoria94, 2020-11-12:
Hello, i am trying to create a custom ESP32 project on Eclipse IDE with ESP-IDF but when i tried to build main.c file i am getting this error , how can i solve this issue? I also added Cmake output log for you.

Traceback (most recent call last):
  File "C:\esp32\esp-idf\tools\kconfig_new\prepare_kconfig_files.py", line 96, in <module>
    _prepare_source_files(env)
  File "C:\esp32\esp-idf\tools\kconfig_new\prepare_kconfig_files.py", line 67, in _prepare_source_files
    _write_source_file(env_dict['COMPONENT_KCONFIGS'], env_dict['COMPONENT_KCONFIGS_SOURCE_FILE'])
  File "C:\esp32\esp-idf\tools\kconfig_new\prepare_kconfig_files.py", line 63, in _write_source_file
    with open(config_file, 'w', encoding='utf-8') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'F:/Programlar/Projeler/Eclipse_Proje/ESP32/KÃ¼tÃ¼phane/build/kconfigs.in'
Traceback (most recent call last):
  File "C:\esp32\esp-idf\tools\kconfig_new\confgen.py", line 625, in <module>
    main()
  File "C:\esp32\esp-idf\tools\kconfig_new\confgen.py", line 252, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "C:\esp32\esp-idf\tools\kconfig_new\kconfiglib.py", line 947, in __init__
    self._init(filename, warn, warn_to_stderr, encoding)
  File "C:\esp32\esp-idf\tools\kconfig_new\kconfiglib.py", line 1085, in _init
    self._parse_block(None, self.top_node, self.top_node).next = None
  File "C:\esp32\esp-idf\tools\kconfig_new\kconfiglib.py", line 2970, in _parse_block
    raise KconfigError(
kconfiglib.KconfigError: C:/esp32/esp-idf/Kconfig:166: 'F:/Programlar/Projeler/Eclipse_Proje/ESP32/KÃ¼tÃ¼phane/build/kconfigs_projbuild.in' not found (in 'source "$COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
CMake Error at C:/esp32/esp-idf/tools/cmake/kconfig.cmake:225 (message):
  Failed to run confgen.py
  (python;C:/esp32/esp-idf/tools/kconfig_new/confgen.py;--kconfig;C:/esp32/esp-idf/Kconfig;--sdkconfig-rename;C:/esp32/esp-idf/sdkconfig.rename;--config;F:/Programlar/Projeler/Eclipse_Proje/ESP32/KÃ¼tÃ¼phane/sdkconfig;--env-file;F:/Programlar/Projeler/Eclipse_Proje/ESP32/KÃ¼tÃ¼phane/build/config.env).
  Error 1
Call Stack (most recent call first):
  C:/esp32/esp-idf/tools/cmake/build.cmake:442 (__kconfig_generate_config)
  C:/esp32/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:6 (project)

[CMakeOutput.log](https://github.com/espressif/esp-idf/files/5532667/CMakeOutput.log)

dobairoland, 2020-11-12:
Hi @Thoria94. Are you able to build the project from command line instead of Eclipse? 

I suspect that this directory name is causing the issue: "KÃ¼tÃ¼phane". Please check that you haven't set it somewhere by mistake.

Thoria94, 2020-11-12:
Hello @dobairoland, thank you for quick answer. I solved this problem 5 mins ago by updating my git version 2.21 to 2.29 after that my issue is gone. I hope this works for anyone who needs.

georgik, 2021-04-16:
Let me add some info. It's recommended to use path for projects without special characters, because of encoding problems with some versions of Python and 3rd party tools.

Another option is to solve the problem is to enable "Use Unicode UTF-8 for worldwide language support" feature in the Windows ad described here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html

"System Administrator can enable the support via Control Panel - Change date, time, or number formats - Administrative tab - Change system locale - check the option “Beta: Use Unicode UTF-8 for worldwide language support” - Ok and reboot the computer."

Regarding integration with Eclipse integration we have an update: ESP-IDF Tools Windows Installer 2.6+ contains support for installation of preconfigured Eclipse with ESP plugin. You can download the newer version of the installer here:
https://dl.espressif.com/dl/esp-idf/

closed 2021-04-16 by tom-borcin