espressif/esp-idf #6111: Cannot install build dependencies (using install.bat)

MartyMcFlyInTheSky, 2020-11-12:
**Environment**

Development Kit: Node-MCU
Kit version: -
Module or chip used: ESP32-WROOM-32U
IDF version: v4.3-dev-1720-g494a124d9
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: Windows 10, 64bit
Power Supply: USB

**Previously done**

- Followed this guide: https://docs.espressif.com/projects/esp-idf/en/release-v3.0/get-started/index.html#get-started-get-esp-idf -> python, cmake, xtensia compiler and idf installed
- ... up until running install.bat. Then the error described below appears and the installation is (seemingly) cancelled

**Problem Description**

When I run install.bat from within [user]/esp/esp-idf the following displays in my console:

```
C:\Users\samue\esp\esp-idf>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.16.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Skipping ninja@1.10.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
Installing Python packages from C:\Users\samue\esp\esp-idf\requirements.txt
Processing c:\users\samue\esp\esp-idf\tools\kconfig_new\esp-windows-curses
Requirement already satisfied: setuptools>=21 in c:\users\samue\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\samue\esp\esp-idf\requirements.txt (line 4)) (49.6.0)
Collecting click>=5.0
  Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting future>=0.15.2
  Using cached future-0.18.2.tar.gz (829 kB)
Collecting cryptography>=2.1.4
  Using cached cryptography-3.2.1-cp36-abi3-win_amd64.whl (1.5 MB)
Collecting pyparsing<2.4.0,>=2.0.3
  Using cached pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Collecting pyelftools>=0.22
  Using cached pyelftools-0.27-py2.py3-none-any.whl (151 kB)
Collecting gdbgui==0.13.2.0
  Using cached gdbgui-0.13.2.0-py3-none-any.whl (878 kB)
Collecting pygdbmi<=0.9.0.2
  Using cached pygdbmi-0.9.0.2-py3-none-any.whl (16 kB)
Collecting reedsolo<=1.5.4,>=1.5.3
  Using cached reedsolo-1.5.4.tar.gz (271 kB)
Collecting bitstring>=3.1.6
  Using cached bitstring-3.1.7.tar.gz (195 kB)
Collecting kconfiglib==13.7.1
  Using cached kconfiglib-13.7.1-py2.py3-none-any.whl (145 kB)
Collecting windows-curses
  Using cached windows_curses-2.2.0-cp39-cp39-win_amd64.whl (88 kB)
Collecting six>=1.4.1
  Using cached six-1.15.0-py2.py3-none-any.whl (10 kB)
Collecting cffi!=1.11.3,>=1.8
  Using cached cffi-1.14.3-cp39-cp39-win_amd64.whl (179 kB)
Collecting gevent<2.0,>=1.2.2
  Using cached gevent-1.5.0.tar.gz (5.3 MB)
  Installing build dependencies ... error
[... 38 lines cut ...]
      Complete output (5 lines):
      running install
      running build
      running build_ext
      building 'greenlet' extension
      error: Microsoft Visual C++ 14.0 is required. Get it with "Build Tools for Visual Studio": https://visualstudio.microsoft.com/downloads/
      ----------------------------------------
  ERROR: Command errored out with exit status 1: 'C:\Users\samue\.espressif\python_env\idf4.3_py3.9_env\Scripts\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\samue\\AppData\\Local\\Temp\\pip-install-cwr9b6jh\\greenlet\\setup.py'"'"'; __file__='"'"'C:\\Users\\samue\\AppData\\Local\\Temp\\pip-install-cwr9b6jh\\greenlet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\samue\AppData\Local\Temp\pip-record-mv53gz39\install-record.txt' --single-version-externally-managed --prefix 'C:\Users\samue\AppData\Local\Temp\pip-build-env-w23b7lk0\overlay' --compile --install-headers 'C:\Users\samue\AppData\Local\Temp\pip-build-env-w23b7lk0\overlay\include\site\python3.9\greenlet' Check the logs for full command output.
  ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\samue\.espressif\python_env\idf4.3_py3.9_env\Scripts\python.exe' 'C:\Users\samue\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages\pip' install --ignore-installed --no-user --prefix 'C:\Users\samue\AppData\Local\Temp\pip-build-env-w23b7lk0\overlay' --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools >= 40.8.0' wheel 'Cython >= 0.29.14' 'cffi >= 1.12.3 ; platform_python_implementation == '"'"'CPython'"'"'' 'greenlet>=0.4.14 ; platform_python_implementation == '"'"'CPython'"'"'' Check the logs for full command output.
Traceback (most recent call last):
  File "C:\Users\samue\esp\esp-idf\tools\idf_tools.py", line 1589, in <module>
    main(sys.argv[1:])
  File "C:\Users\samue\esp\esp-idf\tools\idf_tools.py", line 1585, in main
    action_func(args)
  File "C:\Users\samue\esp\esp-idf\tools\idf_tools.py", line 1312, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "C:\Users\samue\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Users\\samue\\.espressif\\python_env\\idf4.3_py3.9_env\\Scripts\\python.exe', '-m', 'pip', 'install', '--no-warn-script-location', '-r', 'C:\\Users\\samue\\esp\\esp-idf\\requirements.txt']' returned non-zero exit status 1.
```

The part from `Installing build dependencies ... error` to `Check the logs for full command output.` is marked RED for error. It looks like it has difficulties building but I don't know how to solve it. CMake is installed and linked via PATH-variable.

I hope you can help me fix the problem. Thanks for your attention :)

dobairoland, 2020-11-12:
Hi @MartyMcFlyInTheSky. Unfortunately, there are no pre-built binaries for Python 3.9 of some of the dependencies, e.g. https://pypi.org/project/gevent/1.5.0/#files. That is why it tries to build them and looking for a visual c++ compiler.

Can you please use Python 3.8 until we find a solution for this?

MartyMcFlyInTheSky, 2020-11-12:
That worked like a charm, thank you @dobairoland !

georgik, 2020-11-30:
Package greenlet (https://pypi.org/project/greenlet/) has no pre-built binaries for Python 3.9.
It is necessary to install Microsoft Visual C++ Build Tools 2015, so that pip can build the package.
Suggested solution based on https://chocolatey.org/packages/vcbuildtools :
```
choco install vcbuildtools
refreshenv
pip install greenlet
```

bbeliveau, 2020-12-11:
Unfortunately @dobairoland it does not work for me with Python 3.8.2.
Still have issue. I will open a ticket.

georgik, 2020-12-11:
@bbeliveau Please open the ticket. According to my tests Python 3.8.6 should work fine with 4.3.
Please, provide also logs or stack trace.

georgik, 2021-01-11:
Update: To get installation working with Windows Python 3.9, please set the environment variable `PIP_EXTRA_INDEX_URL` to `https://dl.espressif.com/pypi` then launch the installation again.

This is workaround. The solution will be included in the future version of installer. It was necessary to build some Python Wheels which were not ready for Windows. The wheels are stored on URL mentioned above.

Alvin1Zhang, 2021-02-02:
Thanks for reporting and sorry fro slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/76acc8ddde0b2fab66f9918792f9f062040ee254, feel free to reopen. Thanks.

closed 2021-02-02 by Alvin1Zhang

georgik, 2021-02-02:
@bbeliveau Thank you for reporting the issue. The problem with Python is solved by new installers:
- Online installer (30 MB) : https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
- Offline installer - contains v4.2 and v4.1.1 (810 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.5.exe

The solution is based on Embedded Python 3.8.7 and prepared package of Python Wheels.