espressif/esp-idf #6112: FreeRTOS/coreMQTT Support

AshUK, 2020-11-12:
Currently, the MQTT component makes use of the esp-mqtt library supported and maintained by espressif. Although this is a feature-rich library it does lack the many of qualities of CoreMQTT such as the extensive unit test code coverage and code quality.

Now that FreeRTOS/coreMQTT has reached released will there be planned support or an option to switch out the current component?

FreeRTOS/coreMQTT#98

Alvin1Zhang, 2020-11-24:
Thanks for raising the feature request.

lhespress, 2020-12-02:
@AshUK 
Please replace **esp-idf/components/mqtt** and **esp-idf/examples/protocols/mqtt/ssl** to the attached files.

[FreeRTOS:coreMQTT.zip](https://github.com/espressif/esp-idf/files/5629126/FreeRTOS.coreMQTT.zip)

AshUK, 2020-12-02:
@lhespress Thanks very much for this port, I really was not expecting a response so soon.

I have tested and it appears to be working as expected. 

Free heap during a active MQTTS connection:
FreeRTOS=198068
ESP=192672

A saving of 5396 bytes seems promising.

KaeLL, 2020-12-02:
@AshUK will you submit a PR?

AshUK, 2020-12-03:
@KaeLL It wouldn't be my place to submit the PR as @lhespress has done all work.

@lhespress Do you know if there there is a plan to add this to the IDF ?

lhespress, 2020-12-04:
@AshUK 

I have submit a MR for reviewing.

lhespress, 2021-01-25:
@AshUK Hi, freertos coremqtt has been pushed on github, please check https://github.com/espressif/esp-freertos-coremqtt

closed 2021-01-25 by Alvin1Zhang