espressif/esp-idf #6114: Spurious MDNS collision detection

djphoenix, 2020-11-13:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.0.2
- Build System:         Make
- Compiler version: esp-2019r2
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

I have faced an issue with spurious MDNS TXT collision detection, that cause an infinite MDNS query/answer loop when router retransmits MDNS message to ESP module. E.g.

- ESP sends multicast response with TXT record
- ESP receives multicast response with TXT record (actually from itself)
- ESP validates response against service description ** *FAIL HERE* **
- ESP sends multicast request (probe) for its service name with ANY type
- ESP receives multicast request with its service name
- Looped to step 1

### Steps to reproduce

Register MDNS service with multiple TXT entries

### Point to code

https://github.com/espressif/esp-idf/blob/v4.0.2/components/mdns/mdns.c#L2108

It counts length of first record regardless to current linked-list foreach-loop record.
Also it counts `data_len` from 1 instead of 0.
Seems like `master` and `ESP8266_RTOS_SDK` have same issues.

Alvin1Zhang, 2020-11-16:
Thanks for reporting, we will look into.

lhespress, 2020-12-03:
@djphoenix 

It is much easier for us if you could provide small test code on top of  **esp-idf/examples/protocols/mdns**.

closed 2021-01-05 by espressif-bot

AndreyKarpov, 2021-01-20:
P.S. Continuation of the story: https://www.viva64.com/en/b/0790/ :)