espressif/esp-idf #6115: Missing support for xTaskGetIdleRunTimeCounter()

drws, 2020-11-13:
It seems the esp-idf does not support FreeRTOS's xTaskGetIdleRunTimeCounter() function. It would probably need some dual-core adaptation or just return the idle task stats for the core it runs on. Are there any other reasons this particular function support is missing?

Alternatively, the vTaskGetRunTimeStats() can provide the relevant information, but it populates a large buffer which needs to be parsed afterwards, which seems like a bulky solution to obtain a single value (or two).

Alvin1Zhang, 2020-11-16:
Thanks for raising this feature.

Dazza0, 2021-09-23:
@drws ESP-IDF updated to FreeRTOS 10 as of commit a3c90bf.  `xTaskGetIdleRunTimeCounter()` should be supported (defined as a macro to `ulTaskGetIdleRunTimeCounter()`.

closed 2021-09-23 by Dazza0