espressif/esp-idf #6116: RTC not keeping track of time - VBAT Pin

matheusmbgeng, 2020-11-13:
Hello guys, how are u?

I'm having some troubles using the internal RTC of ESP32. I'm used to develop with microcontrollers that has internal RTC and also a pin called VBAT Pin, that i'm able to wire a coin cell battery to this pin and the system still keeps track of time, even if it has no power supply.  Is there any VBAT pin in ESP32? I read on some other topic to wire the coin cell on GPIO36, but it didn't work, time reseted after power went off.

I know i can wire a battery to VCC but due to PCB and budget requirements, this is not possible.This is the same reason why i can't use DS3231 or something similar either. Also, i tought since ESP32 has an internal RTC, why to use an external module? Unless, there is no way to keep track of time while powered off.

I may or may not have internet to get the time from SNTP when the system restart, that's why i need RTC to keep counting. And time is a vital parameter for my system to execute properly.

I really hope ESP32 have a way to keep track of time with a VBAT pin, it would be weird to me that such a robust microcontroller doesn't have this feature like other common devices on the market.

felmue, 2020-11-14:
Hello @matheusmbgeng 

I have not yet come across an ESP32 with a VBAT pin. And GPIO36 is simply an analog input pin which to my knowledge cannot be used to power the RTC or anything else for that matter.

For the RTC in the ESP32 to be running the ESP32 needs to be powered, but it can be in light or deep sleep to conserve power.

However please note that with the internal oscillator the RTC is not very precise. You can improve that by adding an external 32kHz crystal or oscillator.

Check out section `2.4.2 RTC (optional)` in the [ESP32 Hardware Design Guidelines](https://www.espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf#page=10) about what clock sources are available for RTC.

Also if you haven't looked at the [ESP32 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf) document I recommend you do that. 

BTW: Searching though it with term `bat` yields exactly one hit which talks about when powering the complete chip by battery.

Thanks
Felix

igrr, 2020-11-16:
Hi @matheusmbgeng,
This issue tracker is for bug reports and feature requests for ESP-IDF project. For chip-related questions, such as the power supply scheme, please use either the esp32.com community forum or contact Espressif technical support at https://www.espressif.com/en/contact-us/technical-inquiries/hardware-issues.

closed 2020-11-16 by igrr