espressif/esp-idf #6118: Unable to get any BLE / Mesh examples to build with the exception of nimBLE bluephph.

mrSilkie, 2020-11-14:
I've tried using both 4.0.2 and 4.1 to build the following
https://github.com/espressif/esp-idf/tree/357a277603/examples/bluetooth/bluedroid/ble/gatt_server_service_table

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-ble-mesh/ble-mesh-index.html

In the docs, the first step is 
> idf.py menuconfig > Example Configuration > Board selection for ESP-BLE-MESH

but if my menuconfig, the example configuration is not even an option.

I don't believe that none of the project example builds are working as a result of incompatibility with IDF v 4.1 but am lost since I cant think of any other reasons.

>user@dev:~/esp/myprojects/Bluetooth Examples/gatt_server$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory '/home/user/esp/myprojects/Bluetooth Examples/gatt_server/build'
Executing "ninja all"...
[0/1] Re-running CMake...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
Loading defaults file /home/user/esp/myprojects/Bluetooth Examples/gatt_server/sdkconfig.defaults...
-- Adding linker script /home/user/esp/myprojects/Bluetooth Examples/gatt_server/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/user/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/user/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/user/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/user/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/user/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/user/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/user/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/user/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/user/esp/esp-idf/components/app_trace /home/user/esp/esp-idf/components/app_update /home/user/esp/esp-idf/components/asio /home/user/esp/esp-idf/components/bootloader /home/user/esp/esp-idf/components/bootloader_support /home/user/esp/esp-idf/components/bt /home/user/esp/esp-idf/components/cbor /home/user/esp/esp-idf/components/coap /home/user/esp/esp-idf/components/console /home/user/esp/esp-idf/components/cxx /home/user/esp/esp-idf/components/driver /home/user/esp/esp-idf/components/efuse /home/user/esp/esp-idf/components/esp-tls /home/user/esp/esp-idf/components/esp32 /home/user/esp/esp-idf/components/esp_adc_cal /home/user/esp/esp-idf/components/esp_common /home/user/esp/esp-idf/components/esp_eth /home/user/esp/esp-idf/components/esp_event /home/user/esp/esp-idf/components/esp_gdbstub /home/user/esp/esp-idf/components/esp_http_client /home/user/esp/esp-idf/components/esp_http_server /home/user/esp/esp-idf/components/esp_https_ota /home/user/esp/esp-idf/components/esp_https_server /home/user/esp/esp-idf/components/esp_local_ctrl /home/user/esp/esp-idf/components/esp_netif /home/user/esp/esp-idf/components/esp_ringbuf /home/user/esp/esp-idf/components/esp_rom /home/user/esp/esp-idf/components/esp_serial_slave_link /home/user/esp/esp-idf/components/esp_websocket_client /home/user/esp/esp-idf/components/esp_wifi /home/user/esp/esp-idf/components/espcoredump /home/user/esp/esp-idf/components/esptool_py /home/user/esp/esp-idf/components/expat /home/user/esp/esp-idf/components/fatfs /home/user/esp/esp-idf/components/freemodbus /home/user/esp/esp-idf/components/freertos /home/user/esp/esp-idf/components/heap /home/user/esp/esp-idf/components/idf_test /home/user/esp/esp-idf/components/jsmn /home/user/esp/esp-idf/components/json /home/user/esp/esp-idf/components/libsodium /home/user/esp/esp-idf/components/log /home/user/esp/esp-idf/components/lwip /home/user/esp/esp-idf/components/mbedtls /home/user/esp/esp-idf/components/mdns /home/user/esp/esp-idf/components/mqtt /home/user/esp/esp-idf/components/newlib /home/user/esp/esp-idf/components/nghttp /home/user/esp/esp-idf/components/nvs_flash /home/user/esp/esp-idf/components/openssl /home/user/esp/esp-idf/components/partition_table /home/user/esp/esp-idf/components/perfmon /home/user/esp/esp-idf/components/protobuf-c /home/user/esp/esp-idf/components/protocomm /home/user/esp/esp-idf/components/pthread /home/user/esp/esp-idf/components/sdmmc /home/user/esp/esp-idf/components/soc /home/user/esp/esp-idf/components/spi_flash /home/user/esp/esp-idf/components/spiffs /home/user/esp/esp-idf/components/tcp_transport /home/user/esp/esp-idf/components/tcpip_adapter /home/user/esp/esp-idf/components/ulp /home/user/esp/esp-idf/components/unity /home/user/esp/esp-idf/components/vfs /home/user/esp/esp-idf/components/wear_levelling /home/user/esp/esp-idf/components/wifi_provisioning /home/user/esp/esp-idf/components/wpa_supplicant /home/user/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/esp/myprojects/Bluetooth Examples/gatt_server/build
[21/1165] cd "/home/user/esp/myprojects/Bluetooth Examples/gatt_ser...******************************************************************

>Partition table binary generated. Contents:
*******************************************************************************
># Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
>[110/1165] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/__/__/port/esp32/esp_bignum.c.obj
/home/user/esp/esp-idf/components/mbedtls/port/esp32/esp_bignum.c:254:5: warning: no previous prototype for 'esp_mpi_mul_mpi_mod' [-Wmissing-prototypes]
 int esp_mpi_mul_mpi_mod(mbedtls_mpi *Z, const mbedtls_mpi *X, const mbedtls_mpi *Y, const mbedtls_mpi *M)
     ^~~~~~~~~~~~~~~~~~~
[359/1165] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v4.1
-- Building ESP-IDF components for target esp32
-- Adding linker script /home/user/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
[... 85 lines cut ...]
[73/91] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_flash_config_esp32.c.obj
[74/91] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/hal/sdio_slave_hal.c.obj
[... cut ...] /home/user/esp/esp-idf/components/esp32/ld -T esp32.peripherals.ld -L /home/user/esp/esp-idf/components/bootloader/subproject/main/ld/esp32 -T bootloader.ld -T bootloader.rom.ld && :
xtensa-esp32-elf-gcc: error: Examples/gatt_server/build/bootloader/bootloader.map: No such file or directory
ninja: build stopped: subcommand failed.
[464/1165] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/ipv6/nd6.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

chegewara, 2020-11-14:
```
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
```
`FAILED: bootloader.elf`
`xtensa-esp32-elf-gcc: error: Examples/gatt_server/build/bootloader/bootloader.map: No such file or directory`

Try to do math r dont use custom partition scheme.

mrSilkie, 2020-11-14:
HI! I managed to figure it out!

Such a silly error. I have myprojects > bluetooth examples > onoff_client

And what fixed it was the removal of whitespace in my folder structure to myprojects > bluetooth_examples > onoff_client

closed 2020-11-14 by mrSilkie