espressif/esp-idf #6119: Undefined reference to 'sysconf' and 'posix_memalign'

GiovanniCmpaner, 2020-11-15:
## Environment

- Environment type: [PlatformIO (Arduino)]
- Development Kit: [ESP32-S2-Saola-1M]
- Module or chip used: [ESP32-S2-WROOM]
- IDF version [idf-release/v4.2]
- Compiler version: [esp2020r2 8.2.0]
- Operating System: [Windows]
- Power Supply: [USB]

## Problem Description

When trying to compile the MVCE below inside PlataformIO, I get undefined references to `sysconf` and `posix_memalign`.
I need those functions because of a third party library ([tiny-dnn](https://github.com/tiny-dnn/tiny-dnn) especifically).

### Expected Behavior

Are those functions implemented?
What are the alternatives?

### Code to reproduce this issue

```cpp
#include <Arduino.h>
#include <cstdlib>
#include <thread>

void setup()
{
    void* ptr{ nullptr };
    posix_memalign( &ptr, 16, 1024 );

    long value = std::thread::hardware_concurrency();
}

void loop()
{

}
```

## Debug Logs

Error:
```
[...]

Building in release mode
Linking .pio\build\esp32doit-devkit-v1\firmware.elf
c:/users/giovanni/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\Giovanni\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32s2\lib\librtc.a(rtc.o)(.text.rtc_pad_gpio_wakeup+0xa9): could not decode instruction; possible configuration mismatch
c:/users/giovanni/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: c:/users/giovanni/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/lib\libstdc++.a(thread.o):(.literal._ZNSt6thread20hardware_concurrencyEv+0x0): undefined reference to `sysconf'
c:/users/giovanni/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: c:/users/giovanni/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/lib\libstdc++.a(thread.o): in function `std::thread::hardware_concurrency()':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s2-elf/src/gcc/libstdc++-v3/src/c++11/thread.cc:177: undefined reference to `sysconf'
c:/users/giovanni/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\src\main.cpp.o:(.literal._Z5setupv+0x0): undefined reference to `posix_memalign'
c:/users/giovanni/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\src\main.cpp.o: in function `setup()':
C:\Users\Giovanni\Desktop\auto2teste/src/main.cpp:8: undefined reference to `posix_memalign'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32doit-devkit-v1\firmware.elf] Error 1
===================================================================== [FAILED] Took 11.54 seconds =====================================================================
The terminal process "C:\Users\Giovanni\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.
```

## Other items

platformio.ini:
```ini
[env:esp32doit-devkit-v1]
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino

monitor_speed = 115200
board_build.speed = 921600
board_build.partitions = partitions_custom.csv
board_build.mcu = esp32s2

build_unflags = -std=gnu++11 -fno-rtti
build_flags = -std=gnu++14 -DCORE_DEBUG_LEVEL=5

[...]
```

Terminal output infos:
```
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.29e3b64
 - tool-esptoolpy 1.30000.200511 (3.0.0)
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32s2 1.80200.200827 (8.2.0)

[...]
```

aquibjamal, 2021-02-17:
Hi,
I am facing the same issue while working with [tiny-cnn](https://github.com/Xilinx/xilinx-tiny-cnn).

Let me know if there is any workaround for this.

With regards,
Aquib Jamal

GiovanniCmpaner, 2021-02-18:
I made some modifications: https://github.com/GiovanniCmpaner/tiny-dnn
Take a look at the commit history for more details.

Also, some additional unflags and flags are needed in the `platformio.ini` file:
```
build_unflags = -fno-rtti
build_flags = -mtext-section-literals
```

For some insight, because of the memory consumption of `tiny-dnn`, I switched over for `tensorflow-lite`.

aquibjamal, 2021-02-21:
Thank you for sharing the repository and pointing in the right direction.
I am able to compile the code as shown in your files.
I had already set the rtti flags in sdkconfig file.
I am not sure if it is connected to the same issue, but now I am getting linker error - 
"hello-world.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
region `dram0_0_seg' overflowed by 83856 bytes"

I am trying to use the IRAM_DATA_ATTR and IRAM_BSS_ATTR macros, for template class variables.
This results in the following error - "section attribute not allowed for 'm_weights'
   IRAM_BSS_ATTR ap_uint<SIMD>  m_weights[PE][TILES];"

Have you come across similar problems?

Many thanks,
Aquib

GiovanniCmpaner, 2021-02-21:
Probably there isnt enought memory in DRAM for your variable with IRAM macro, maybe you can use std::vector for allocating it in heap, at runtime.
But this erros specific to your project needs, try searching in google.
Good luck!

dzz10, 2021-08-12:
所以，这个问题是怎么回事，Undefined reference to 'sysconf'

closed 2021-12-15 by espressif-bot