espressif/esp-idf #6120: Comparison error in soc component

Silvanosky, 2020-11-15:
Hello,

In the version v4.1 there is an error here : 

https://github.com/espressif/esp-idf/blob/5d258af93c668d748a5d190137c1bf8ae58489b4/components/soc/esp32/include/soc/cpu.h#L77

When you enable -Werror=sign-compare it compares an unsigned with an int : 
```
../../../../esp/esp-idf/components/soc/esp32/include/soc/cpu.h: In function 'void cpu_configure_region_protection()':
../../../../esp/esp-idf/components/soc/esp32/include/soc/cpu.h:77:23: error: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Werror=sign-compare]
     for (int i = 0; i < sizeof(pages_to_protect)/sizeof(pages_to_protect[0]); ++i) {
                     ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors
```

Thanks

suda-morris, 2020-12-02:
Thanks for your feedback, we will fix it and add more tests to avoid such issues.

Alvin1Zhang, 2021-08-12:
Thanks for reporting and sorry for slow turnaround, fix is available https://github.com/espressif/esp-idf/commit/ec2817a28553fd13421f045782b26b5f54cce94a. Feel free to reopen.

closed 2021-08-12 by Alvin1Zhang