espressif/esp-idf #6121: Cannot add more than 6 endpoints to the BLE protocomm provisioning

ChromaMaster, 2020-11-16:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.3.2-323-gbf0220609
- Build System: Make
- Compiler version: 1.22.0-96-g2852398
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I'm trying to implement a provisioning system using the procomm method described [HERE](https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/provisioning/protocomm.html#transport-example-ble-with-security-0). I'm using the example provided [HERE](https://github.com/espressif/esp-idf/tree/release/v3.3/examples/provisioning/ble_prov). When I try to add more than 6 endpoints (BLE characteristics) the esp32 just enters in a panic loop rebooting itself.

### Expected Behavior

Don't panic, just add the 7th, 8th, 9th, ... endpoints correctly.

### Actual Behavior

The esp32 boots up, panics and restarts...

### Steps to reproduce

1. Copy the example from [HERE](https://github.com/espressif/esp-idf/tree/release/v3.3/examples/provisioning/ble_prov) (release/v3.3)
2. Add some endpoints to the [lookup table](https://github.com/espressif/esp-idf/blob/release/v3.3/examples/provisioning/ble_prov/main/app_prov.c#L61)
3. Compile & Flash

### Code to reproduce this issue

Use the ble_prov example from this repository (release/v3.3) and add some endpoints to the  like below:

```C
/* Endpoint UUIDs */
protocomm_ble_name_uuid_t nu_lookup_table[] = {
    {"prov-session", 0x0001},
    {"prov-config",  0x0002},
    {"proto-ver",    0x0003},
    {"prov-custom1", 0x0004},
    {"prov-custom2", 0x0005},
    {"prov-custom3", 0x0006},
    {"prov-custom4", 0x0007} // <--------- This is what makes everything to crash.
};
```

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c2e0  PS      : 0x00060e30  A0      : 0x8011a8c7  A1      : 0x3ffd8040  
A2      : 0x3ffd8058  A3      : 0x003d0050  A4      : 0x00000020  A5      : 0x3ffd8058  
A6      : 0x3ffd7d50  A7      : 0x00000002  A8      : 0x00000000  A9      : 0x3ffd7d20  
A10     : 0x3ffb664c  A11     : 0x3ffb65f4  A12     : 0x3ffd7f74  A13     : 0x00000000  
A14     : 0x0000000a  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x003d0050  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000001  

ELF file SHA256: 3cf1c1e736f1e595

Backtrace: 0x4000c2e0:0x3ffd8040 0x4011a8c4:0x3ffd8050 0x40115a8e:0x3ffd80b0 0x4011f181:0x3ffd80d0 0x4011ac62:0x3ffd8100 0x4008e63e:0x3ffd8140
0x4011a8c4: esp_ble_gap_start_advertising at /home/user/.espressif/esp-idf/components/bt/bluedroid/api/esp_gap_ble_api.c:428

0x40115a8e: gap_event_handler at /home/user/.espressif/esp-idf/components/protocomm/src/simple_ble/simple_ble.c:62

0x4011f181: btc_gap_ble_cb_to_app at /home/user/.espressif/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c:54
 (inlined by) btc_gap_ble_cb_handler at /home/user/.espressif/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c:898

0x4011ac62: btc_task at /home/user/.espressif/esp-idf/components/bt/common/btc/core/btc_task.c:141

0x4008e63e: vPortTaskWrapper at /home/user/.espressif/esp-idf/components/freertos/port.c:403
```

Alvin1Zhang, 2020-11-17:
Thanks for reporting, we will look into.

dhrishi, 2020-11-17:
Hi @ChromaMaster  This was reported in our QA tests and has been already fixed internally. The fix is not synced to Github yet. Now, we have also added a mention of this issue in the fix for v3.3 backport. So, you should get a notification here when it gets updated. Thanks!

mahavirj, 2020-11-24:
@ChromaMaster Fixed with https://github.com/espressif/esp-idf/commit/c8354f0231236c8b2da35af7cf65aac77300cfee. Please feel to reopen in case issue persist at your end.

closed 2020-11-24 by mahavirj

CSHaze, 2021-03-29:
This fix is in 4.2 right? I think I am seeing this exact error in 4.2