espressif/esp-idf #6122: socket receive timeout(SO_RCVTIMEO) does not work

nopnop2002, 2020-11-17:
# socket receive timeout(SO_RCVTIMEO) does not work

## Environment

	freeRTOS version:V8.2.0
	NEWLIB version:3.0.0
	lwIP version:2-1-3-0
	ESP-IDF version:v4.3-dev-1197-g8bc19ba89

## Problem Description
socket receive timeout does not work.

### Code to reproduce this issue

Change a part of examples/protocols/http_request/main/http_request_example_main.c as follows.

        /* Read HTTP response */
        do {
            bzero(recv_buf, sizeof(recv_buf));
            ESP_LOGI(TAG, "Start read now=%d", xTaskGetTickCount());
            r = read(s, recv_buf, sizeof(recv_buf)-1);
            ESP_LOGI(TAG, "End   read now=%d r=%d", xTaskGetTickCount(), r);
        #if 0
            for(int i = 0; i < r; i++) {
                putchar(recv_buf[i]);
            }
        #endif
        } while(r > 0);

## Debug Logs

	I (3113) example_connect: Connected to example_connect: sta
	I (3113) example_connect: - IPv4 address: 192.168.10.132
	I (3123) example: DNS lookup succeeded. IP=192.168.10.43
	I (3133) example: ... allocated socket
	I (3163) example: ... connected
	I (3163) example: REQUEST socket send success
	I (3163) example: ... set socket receiving timeout success
	I (3163) example: Start read now=260
	I (3173) example: End   read now=261 r=63
	I (3173) example: Start read now=261
	I (3173) example: End   read now=261 r=63
	I (3183) example: Start read now=262
	I (3183) example: End   read now=262 r=39
	I (3193) example: Start read now=263
	I (3193) example: End   read now=263 r=0 -----> ***** Not wait for 5 seconds
	I (3203) example: ... done reading from socket. Last read return=0 errno=128.

boarchuz, 2020-11-17:
It's likely the server at 192.168.10.43 is closing the connection once the response is complete.

If you remove the write section (so that there is no request for the server to respond to) does the read time out as expected?

nopnop2002, 2020-11-17:
I removed the write section.
socket receive timeout work fine.
And the return value of read () was -1.

	I (3113) example_connect: - IPv4 address: 192.168.10.132
	I (3123) example: DNS lookup succeeded. IP=192.168.10.43
	I (3133) example: ... allocated socket
	I (3143) example: ... connected
	I (3143) example: ... set socket receiving timeout success
	I (3143) example: Start read now=258
	I (8143) example: End   read now=758 r=-1
	I (8143) example: ... done reading from socket. Last read return=-1 errno=11.

>It's likely the server at 192.168.10.43 is closing the connection once the response is complete.

Thank you.

closed 2020-11-17 by nopnop2002