espressif/esp-idf #6123: Invalid len argument type in NVS C++ API?

hg, 2020-11-17:
## Environment

- IDF version: v4.3-dev-1720-g494a124d9
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

## Problem Description

Hi,

I omitted most of the template as it doesn't really apply to this issue.

I think the `NVSHandle::get_string` method uses the wrong type for argument `len`:

https://github.com/espressif/esp-idf/blob/5aa21584cfb4cfe4f2c7f6250151341e28141cf3/components/nvs_flash/include/nvs_handle.hpp#L152

If we look at the docs, it is described as the pointer to the variable which receives the length of the string actually read from flash memory:

https://github.com/espressif/esp-idf/blob/5aa21584cfb4cfe4f2c7f6250151341e28141cf3/components/nvs_flash/include/nvs_handle.hpp#L142-L144

And if we look at the C API, the corresponding function does receive a pointer:

https://github.com/espressif/esp-idf/blob/5aa21584cfb4cfe4f2c7f6250151341e28141cf3/components/nvs_flash/include/nvs.h#L346

It can be worked around by using `NVSHandle::get_item_size` and then passing its result to `NVSHandle::get_string`, but I guess docs for `NVSHandle::get_string` could be improved if it's too late to change its signature.

Thank you!

0xjakob, 2020-12-01:
@hg 
Thanks a lot for being attentive!
We will change the documentation. This is in fact a documentation mistake from copy-pasting the docs over from the C API.

0xjakob, 2021-02-20:
@hg A fix has been merged into our internal CI system. It should be pushed to the github master during the next few weeks.

closed 2021-03-12 by espressif-bot