espressif/esp-idf #6124: backtrace for abort / panic not available in coredump

mmadela, 2020-11-17:
## Environment

- Development Kit:      ESP32-Wrover-Kit or ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROVER-B or ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.1
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes Eclipse ESP-IDF Plugin
- Power Supply:         USB

## Problem Description

the CoreDump component is intended to locate problems in the sources, but it cannot provide a backtrace in case of panic / assert / abort

### Expected Behavior

Complete backtrace is available

### Actual Behavior

Just 2 frames are available
```
#0  0x4008429c in invoke_abort () at D:/esp-idf/esp-idf-v4.1/components/esp32/panic.c:159
#1  0x40084698 in abort () at D:/esp-idf/esp-idf-v4.1/components/esp32/panic.c:174
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
### Steps to reproduce

1. take an example (like examples / get-started / hello_world)
2. put an ``assert(0);`` somewhere, for example in the loop, line 45 of hello_world_main.c, see below too
3. configure coredump to flash in sdkconfig (or use attached config file)
4. run the code
5. read the dump via 
``\esp-idf-v4.1\components\espcoredump\espcoredump.py -p COM15 info_corefile D:\eclipse_cpp\workspace\hello_world\build\hello-world.elf``

### Code to reproduce this issue

```cpp
/* Hello World Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

#ifdef CONFIG_IDF_TARGET_ESP32
#define CHIP_NAME "ESP32"
#endif

#ifdef CONFIG_IDF_TARGET_ESP32S2BETA
#define CHIP_NAME "ESP32-S2 Beta"
#endif

void app_main(void)
{
    printf("Hello world!\n");

    /* Print chip information */
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("This is %s chip with %d CPU cores, WiFi%s%s, ",
            CHIP_NAME,
            chip_info.cores,
            (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
            (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    printf("silicon revision %d, ", chip_info.revision);

    printf("%dMB %s flash\n", spi_flash_get_chip_size() / (1024 * 1024),
            (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        assert(0);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}

```
## Other items if possible

- sdkconfig file (remove the txt extension)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5552499/sdkconfig.txt)

igrr, 2020-11-17:
@mmadela Thanks for raising this issue. The issue is likely related to the handling of `noreturn` functions. When the last instruction of a noreturn function is a call to another function, the return address falls outside of the function boundary. This seems to cause gdb to interrupt the backtrace. We are looking into this issue and will post an update here once there is progress.

maqian-cn, 2021-06-10:
i have a same problem. waiting for progress

mmadela, 2021-06-10:
in the meanwhile I realized, that a backtrace is present when coredump is downloaded from flash and parsed by the espcoredump.py

maqian-cn, 2021-06-21:
`		if ((3 * 60 * pdMS_TO_TICKS(1000)) <= abs(xTaskGetTickCount() - xPrintTimeLogTick))
		{
			xPrintTimeLogTick = xTaskGetTickCount();
			memset(time, 0, sizeof(time));
			Common_PrintTime(time, sizeof(time));
			ESP_LOGI(TAG, "Time: %s", time);
			ESP_LOGI(TAG, "Minimum Free memory: %d bytes", esp_get_minimum_free_heap_size());
			ESP_LOGI(TAG, "Free memory: %d bytes", esp_get_free_heap_size());
			assert(0);
                }
`

in APP code, add "assert(0)" for testing coredump to flash.

coredump info below:

> esptool.py v2.8
Serial port COM3
Connecting........_
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:db:7b:f8
Uploading stub...
Running stub...
Stub running...
15928 (100 %)
15928 (100 %)
Read 15928 bytes at 0x3f0000 in 1.5 seconds (85.3 kbit/s)...
Hard resetting via RTS pin...

===============================================================
==================== ESP32 CORE DUMP START ====================

================== CURRENT THREAD REGISTERS ===================
pc             0x4008808b       0x4008808b <invoke_abort+27>
lbeg           0x400014fd       1073747197
lend           0x4000150d       1073747213
lcount         0xfffffffd       4294967293
sar            0x4      4
ps             0x60b20  396064
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x40088364       1074299748
a1             0x3ffcb560       1073526112
a2             0x80040004       -2147221500
a3             0x2f     47
a4             0xc      12
a5             0xffffffff       -1
a6             0x0      0
a7             0xfffffffe       -2
a8             0x0      0
a9             0x3ffcb540       1073526080
a10            0x0      0
a11            0x3ffcb4f1       1073526001
a12            0x3ffcb49f       1073525919
a13            0x31     49
[... 83 lines cut ...]
.coredump.tasks.data 0x3ffdfc88 0x16c RW
.coredump.tasks.data 0x3ffe10b0 0x1b0 RW
.coredump.tasks.data 0x3ffcd464 0x16c RW
.coredump.tasks.data 0x3ffcd290 0x1c0 RW
.coredump.tasks.data 0x3ffd7f88 0x16c RW
.coredump.tasks.data 0x3ffd7cd0 0x2a4 RW
.coredump.tasks.data 0x3ffafd34 0x16c RW
.coredump.tasks.data 0x3ffc6710 0x188 RW
.coredump.tasks.data 0x3ffaf728 0x16c RW
.coredump.tasks.data 0x3ffaf580 0x194 RW
.coredump.tasks.data 0x3ffcec1c 0x16c RW
.coredump.tasks.data 0x3ffcea50 0x1b8 RW
.coredump.tasks.data 0x3ffafb50 0x16c RW
.coredump.tasks.data 0x3ffc6100 0x188 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!

> ==================== CURRENT THREAD STACK =====================
#0  0x4008808b in invoke_abort () at D:/esp-idf/components/esp32/panic.c:156
#1  0x40088364 in abort () at D:/esp-idf/components/esp32/panic.c:171
**Backtrace stopped: previous frame identical to this frame (corrupt stack?)**

why Backtrace had not saved into flash? @mmadela @igrr

mmadela, 2021-06-21:
Backtrace is saved to flash only if this is configured so, see sdkconfig

jkunkee, 2021-06-21:
Core dumps store the backtraces of `CONFIG_ESP32_CORE_DUMP_MAX_TASKS_NUM` threads. The core dump analysis that @maqian-cn shared contains two backtraces, so I'm not sure what their point of confusion is.

It may help at this point to provide another concrete and clear example of what the issue is: during core dump analysis, stack traces on threads that have called `abort()` or `panic()` stop after the second stack frame. I am using ESP-IDF v4.2.1.

I ran into this this week: my ESP32 project ([lc-esp32](https://github.com/jkunkee/lc-esp32)) has some long-term stability issues, so I have enabled some of the IDF and FreeRTOS heap debugging features--notably in this case the "panic on memory allocation failure" feature:

    CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS=y

I also configured it to save core dumps:

```
CONFIG_ESP32_ENABLE_COREDUMP_TO_FLASH=y
CONFIG_ESP32_COREDUMP_DATA_FORMAT_ELF=y
CONFIG_ESP32_COREDUMP_CHECKSUM_CRC32=y
CONFIG_ESP32_ENABLE_COREDUMP=y
CONFIG_ESP32_CORE_DUMP_MAX_TASKS_NUM=16
CONFIG_ESP32_CORE_DUMP_STACK_SIZE=0
```

It looks like I can't attach my binary or core dump here, but here's my [sdkconfig](https://github.com/espressif/esp-idf/files/6690044/sdkconfig.txt).

When I invoke `espcoredump.py` from the ESP-IDF Terminal in VS Code:

    python $env:IDF_PATH\components\espcoredump\espcoredump.py info_corefile --core .\core.bin --core-format raw .\build\lc-esp32.elf

I get this [full dump analysis](https://github.com/espressif/esp-idf/files/6690043/info.txt) is attached. Note that several threads have full backtraces to `vPortTaskWrapper`, like this:

```
==================== THREAD 2 (TCB: 0x3ffbdb1c, name: 'IDLE1') =====================
#0  0x400edfcd in esp_pm_impl_waiti () at C:/esp/esp-idf/components/esp32/pm_esp32.c:484
#1  0x400d5a5d in esp_vApplicationIdleHook () at C:/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63
#2  0x4008b9ea in prvIdleTask (pvParameters=0x0) at C:/esp/esp-idf/components/freertos/tasks.c:3386
#3  0x4008ac9d in vPortTaskWrapper (pxCode=0x4008b9d0 <prvIdleTask>, pvParameters=0x0) at C:/esp/esp-idf/components/freertos/xtensa/port.c:143
```

The faulting thread, however, is truncated:

```
==================== THREAD 1 (TCB: 0x3ffcdad8, name: 'httpd') =====================
#0  0x4008a38b in panic_abort (details=0x3f402b88 \"Memory allocation failed\") at C:/esp/esp-idf/components/esp_system/panic.c:341
#1  0x4008ac88 in esp_system_abort (details=0x3f402b88 \"Memory allocation failed\") at C:/esp/esp-idf/components/esp_system/system_api.c:106
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

I'm lucky that the `details` buffer is useful here; I have some other data suggesting a memory leak, and I'll be pursuing that next. For other `panic()` or `abort()` calls, I've just been guessing based on the thread name and using printf-style debugging.

Hopefully @igrr will be able to chime in with news sometime soon.

maqian-cn, 2021-06-22:
It is the configuration info in sdkconfig file below. 
`
CONFIG_ESP32_ENABLE_COREDUMP_TO_FLASH=y
# CONFIG_ESP32_ENABLE_COREDUMP_TO_UART is not set
# CONFIG_ESP32_ENABLE_COREDUMP_TO_NONE is not set
CONFIG_ESP32_ENABLE_COREDUMP=y
CONFIG_ESP32_CORE_DUMP_MAX_TASKS_NUM=64
`

I have configured the coredump to flash option. and the coredump info is that has been read from esp32 by “python espcoredump.py -p COM3 info_corefile -o 4128768 .\build\ec600s-esp32.elf”

I am using ESP-IDF v3.3.4.

@jkunkee @mmadela

mmadela, 2021-06-22:
please verify, if your partitions.csv contains line
`coredump, data, coredump,,         64K`

maqian-cn, 2021-06-22:
It is the partition table info below.

```
# Espressif ESP32 New Partition Table,,,,,
# Name,Type,SubType,Offset,Size,Flags
nvs,data,nvs,,16K,
otadata,data,ota,,8K,
phy_init,data,phy,,4K,
#ota_0,app,ota_0,,1536K,
#ota_1,app,ota_1,,1536K,
ota_0,app,ota_0,,1984K,
ota_1,app,ota_1,,1984K,
coredump,data,coredump,,64K
#custom,data,fat,,896K
```

@mmadela

mmadela, 2021-06-22:
try to read the crash data
`python.exe \esp-idf\esp-idf-v4.1\components/esptool_py/esptool/esptool.py -p COM19 -b 460800 read_flash 0x14000 65536 coredump.raw`
adjust port and replace address 0x14000 with valid one for your case (I think it is listed out during compilation)

then parse the file
`esp-idf\esp-idf-v4.2\components\espcoredump\espcoredump.py info_corefile path_to_your_poject.elf --core coredump.raw --core-format raw`

maqian-cn, 2021-06-25:
debug code below:
```
		if ((3 * 60 * pdMS_TO_TICKS(1000)) <= abs(xTaskGetTickCount() - xPrintTimeLogTick))
		{
			xPrintTimeLogTick = xTaskGetTickCount();
			memset(time, 0, sizeof(time));
			Common_PrintTime(time, sizeof(time));
			ESP_LOGI(TAG, "Time: %s", time);
			ESP_LOGI(TAG, "Minimum Free memory: %d bytes", esp_get_minimum_free_heap_size());
			ESP_LOGI(TAG, "Free memory: %d bytes", esp_get_free_heap_size());
			assert(0);
#if (E_SYSCONFIG_ONENET_DATASENDT_STATISTICS)
			Ctrl_OneNetDataSendStatistics(pCtrlObjHandle, time);
#endif
#if (E_SYSCONFIG_TASK_STATISTICS)
			Ctrl_AllTaskStatistics();
#endif
		}
```

According to your @mmadela reply

```
D:\module-esp32>python esptool.py -p COM3 -b 460800 read_flash 0x3f0000 65536 coredump.raw
esptool.py v2.8
Serial port COM3
Connecting........
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:2b:96:a1:c3:d0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
65536 (100 %)
65536 (100 %)
Read 65536 bytes at 0x3f0000 in 1.8 seconds (292.5 kbit/s)...
```

```
D:\module-esp32>python espcoredump.py info_corefile .\build\module-esp32.elf --core coredump.raw --core-format raw
espcoredump.py v0.3-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

================== CURRENT THREAD REGISTERS ===================
pc             0x4008779f       0x4008779f <invoke_abort+27>
lbeg           0x400014fd       1073747197
lend           0x4000150d       1073747213
lcount         0xfffffffd       4294967293
sar            0x4      4
ps             0x60720  395040
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
[... 99 lines cut ...]
.coredump.tasks.data 0x3ffccce0 0x2a8 RW
.coredump.tasks.data 0x3ffafb28 0x16c RW
.coredump.tasks.data 0x3ffaf980 0x194 RW
.coredump.tasks.data 0x3ffc3524 0x16c RW
.coredump.tasks.data 0x3ffc3360 0x1b0 RW
.coredump.tasks.data 0x3ffd69dc 0x16c RW
.coredump.tasks.data 0x3ffd67b0 0x218 RW
.coredump.tasks.data 0x3ffd5440 0x16c RW
.coredump.tasks.data 0x3ffd5150 0x2dc RW
.coredump.tasks.data 0x3ffbb014 0x16c RW
.coredump.tasks.data 0x3ffbae80 0x180 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

> ==================== CURRENT THREAD STACK =====================
#0  0x4008779f in invoke_abort () at D:/esp-idf/components/esp32/panic.c:156
#1  0x40087a78 in abort () at D:/esp-idf/components/esp32/panic.c:171
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

it is still “Backtrace stopped” yet.

@mmadela

mmadela, 2021-06-25:
may be it lays on IDF version, you are using 3.3.4, I am 4.1.1

igrr, 2022-02-11:
Unfortunately the fix for the backtraces on `abort` call which we introduced in esp-2021r2 release has caused a significant regression (https://github.com/espressif/esp-idf/issues/8065, https://github.com/espressif/esp-idf/issues/8342, https://github.com/espressif/openocd-esp32/issues/210). We will revert it for now and release esp-2021r2-patch3 release. I'm reopening this issue and will update it when a new fix is available. I'm sorry for the inconvenience!

KaeLL, 2022-06-06:
@igrr Do you have a recommendation for a way to break the app like `abort` but that will show correctly on the core dump while the fix isn't merged?

ztefanjo, 2022-06-09:
@igrr: We are also experiencing this issue on v4.4.1 and would really appreciate a fix. Is there any type of work-around that we can do to get meaningful core-dumps in case of panic?

ztefanjo, 2022-06-16:
@igrr, @KaeLL : I can confirm that this issue is related to the `noreturn` functions. As a workaround, we have patched away all "noreturns" from functions related to panic/abort and now we get more meaningful coredumps.
As the optimization benefit is very small for these kind of functions, perhaps it should be considered to apply this on e.g. v4.4.x until the real problem is solved?
[Remove-noreturn-on-panic-to-improve-CoreDump - Copy.patch.txt](https://github.com/espressif/esp-idf/files/8916121/Remove-noreturn-on-panic-to-improve-CoreDump.-.Copy.patch.txt)

igrr, 2022-06-16:
Hi @ztefanjo, i'm sorry I have missed your previous comment, I was on leave. Here is the patch you can apply to release/v4.4 to prevent the issue for noreturn functions.
[0001-system-add-workaround-for-interrupted-backtraces-on-.patch.txt](https://github.com/espressif/esp-idf/files/8916218/0001-system-add-workaround-for-interrupted-backtraces-on-.patch.txt)

We had this change in review a while ago, but it wasn't accepted because the issue may still occur for "noreturn" functions in application code. Given that we had to revert the fix in GDB, I'll put this patch for review once again.

@KaeLL for your question, you can trigger some exception (like `*((int*)1) = 0;`) or use the `__builtin_trap();` compiler built-in.

danfleck, 2022-08-04:
@igrr When using the patch I get this error (this is on a newly checked out version of v4.4 after applying the patch).

```
In file included from esp/esp-idf-v4.4/components/esp_system/ubsan.c:33:
esp/esp-idf-v4.4/components/esp_system/ubsan.c: In function '__ubsan_default_handler':
esp/esp-idf-v4.4/components/esp_system/ubsan.c:166:52: error: 'buf' undeclared (first use in this function)
     PROTECT_NORETURN_BACKTRACE_1(esp_system_abort, buf);
```

closed 2022-09-02 by espressif-bot

igrr, 2022-09-02:
We have updated GDB used in the `master` branch to fix this issue. The updates to release branches are on progress.

KaeLL, 2022-09-02:
Will there be backports? Will it make the cut for 5.0?

Lapshin, 2022-09-02:
> Will there be backports? Will it make the cut for 5.0?

@KaeLL, it's already avaliabe in 5.0 branch (06cdc0ee4f6a254f4c682e4652e345506d194e3e)

Also, it will be patched for v4.4 soon

AAJAY5, 2022-09-15:
# How to get backtrace from `esp_core_dump_summary_t `?
- I have enabled coredump to `flash` and it is working fine. Now, i get  the `coredump_summary` using this functions.
- I want to save this summary to sdcard.  I want to use this summary to work with [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder).
- is that possible? and How?

```c
if (esp_core_dump_image_get(&out_addr, &out_size) == ESP_OK) {
        esp_core_dump_summary_t coredump_summary;
        ESP_ERROR_CHECK(esp_core_dump_get_summary(&coredump_summary));
        // save this coredump
    }
```

johnboiles, 2022-11-08:
Patching esp-idf with this change fixed the backtrace for me in 4.4.2. Looking forward to an official fix in the stable release though!

```diff
diff --git a/components/newlib/abort.c b/components/newlib/abort.c
index 38ae5b73c4..daf8d601d6 100644
--- a/components/newlib/abort.c
+++ b/components/newlib/abort.c
@@ -19,7 +19,7 @@
 #include "soc/soc_caps.h"
 #include "hal/cpu_hal.h"
 
-void __attribute__((noreturn)) abort(void)
+void abort(void)
 {
     #define ERR_STR1  "abort() was called at PC 0x"
     #define ERR_STR2  " on core "
```

hitecSmartHome, 2024-04-15:
> CONFIG_ESP32_CORE_DUMP_STACK_SIZE

Have you found a solution?

hitecSmartHome, 2024-04-15:
I did this

```cpp
void Sys::saveCoreDump(){
    size_t size = 0;
    size_t address = 0;
    char* coreDump;
    if( !alloc(coreDump, 640) ){ return; }
    if (esp_core_dump_image_get(&address, &size) != ESP_OK) { free(coreDump); return; }
    const esp_partition_t* pt = NULL;
    pt = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_COREDUMP, "coredump");
    if (pt == NULL) { free(coreDump); return; }
    uint8_t bf[256];
    int16_t toRead;
    db.remove(CORE_DUMP_PATH);
    for (int16_t i = 0; i < (size / 256) + 1; i++) {
        strcpy(coreDump, "");
        toRead = (size - i * 256) > 256 ? 256 : (size - i * 256);
        esp_err_t er = esp_partition_read(pt, i * 256, bf, toRead);
        if (er != ESP_OK) {
            break;
        }
        for (int16_t j = 0; j < 256; j++) {
            char str_tmp[3];
            sprintf(str_tmp, "%02x", bf[j]);
            strcat(coreDump, str_tmp);
        }
        db.append(CORE_DUMP_PATH, coreDump);
    }
    esp_core_dump_image_erase();
    free(coreDump);
}
```

Where db.append is just a wrapper for LittleFs append.
I got the folllowing downloadable file:

```bin
646c0000000100000000000000000000000000007f454c4601010100000000000000000004005e00010000000000000034000000000000000000000034002000260028000000000004000000f40400000000000000000000c02a0000c02a0000060000000000000001000000b42f000058fbfd3f58fbfd3f640100006401000006000000000000000100000018310000a0f6fd3fa0f6fd3fb0040000b0040000060000000000000001000000c8350000d4ecfb3fd4ecfb3f64010000640100000600000000000000010000002c37000040ebfb3f40ebfb3f8001000080010000060000000000000001000000ac38000040f4fb3f40f4fb3f6401000064010000060000000000000001000000103a0000b0f2fb3fb0f2fb3f8001000080010000060000000000000001000000903b000018dffb3f18dffb3f6401000064010000060000000000000001000000f43c000070dbfc3f70dbfc3f8001000080010000060000000000000001000000743e000098a6fd3f98a6fd3f6401000064010000060000000000000001000000d83f000090a4fd3f90a4fd3f0002000000020000060000000000000001000000d84100006cb1fd3f6cb1fd3f64010000640100000600000000000000010000003c430000a0d4fe3fa0d4fe3f8001000080010000060000000000000001000000bc44000058acfc3f58acfc3f640100006401000006000000000000000100000020460000b0aafc3fb0aafc3fa0010000a0010000060000000000000001000000c0470000fcdcfc3ffcdcfc3f64010000640100000600000000000000010000002449000030effc3f30effc3fe0010000e0010000060000000000000001000000044b0000acfdfb3facfdfb3f6401000064010000060000000000000001000000684c000020fcfb3f20fcfb3f8001000080010000060000000000000001000000e84d0000b464fd3fb464fd3f64010000640100000600000000000000010000004c4f0000a061fd3fa061fd3f00030000000300000600000000000000010000004c52000064defc3f64defc3f6401000064010000060000000000000001000000b053000020f8fc3f20f8fc3ff0010000f0010000060000000000000001000000a0550000ccdffc3fccdffc3f6401000064010000060000000000000001000000045700007008fd3f7008fd3fb0010000b0010000060000000000000001000000b4580000f4fdfd3ff4fdfd3f6401000064010000060000000000000001000000185a0000003bfc3f003bfc3fd0010000d0010000060000000000000001000000e85b000054fafa3f54fafa3f64010000640100000600000000000000010000004c5d0000c0a5fb3fc0a5fb3fa0010000a0010000060000000000000001000000ec5e0000ccf7fa3fccf7fa3f64010000640100000600000000000000010000005060000020f6fa3f20f6fa3fa0010000a0010000060000000000000001000000f061000048d4fb3f48d4fb3f640100006401000006000000000000000100000054630000a0d2fb3fa0d2fb3fa0010000a0010000060000000000000001000000f4640000c012fd3fc012fd3f6401000064010000060000000000000001000000586600001011fd3f1011fd3fa0010000a0010000060000000000000001000000f867000004b0fd3f04b0fd3f64010000640100000600000000000000010000005c69000030aefd3f30aefd3fc0010000c00100000600000000000000040000001c6b0000000000000000000014010000140100000600000000000000080000004c02000001000000434f52450045535000000000000000000000000000000000000000000000000058fbfd3f00000000000000000000000000000000000000000000000000000000000000000000000000000000000000007c3108402303060049c300406bc30040000000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000c09a088060f7fd3fa0f7fd3fb1f7fd3f1000000090dffd3f09000000070000000000000001000000b1f7fd3fb1f7fd3f03000000f857fc3f98e87a00ffff3f0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000080000004c02000001000000434f524500455350000000000000000000000000000000000000000000000000d4ecfb3f0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000fe742640200d0600000000000000000000000000160000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...] 00000000000000000000000000000000000000000000000000000000000000000000598e2779000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000c00000094000000a502000045585452415f494e464f002a58fbfd3fe80000001d000000ee00000000000000c200000000000000c300000000000000c400000000000000c500000000000000c600000000000000b100000058d31c40b200000000000000b3000000
```

I don't know how to decode this. I'm trying the espcoredump.py script for now to get a meaningfull output.