espressif/esp-idf #6125: Inicializing PPPos example and getting errors

felipebayres, 2020-11-17:
I'm trying to run the PPPoS example in a TTGO T-Call (Esp-32 with a SIM800L embbed) but it keeps getting these errors as soon as esp32 starts:

```
E (968) esp-modem: esp_modem_dte_send_cmd(221): process command timeout
E (968) dce_service: esp_modem_dce_sync(48): send command failed
E (968) sim800: sim800_init(460): sync failed
```
I tested the hardware with another project so I assume that it is working, but these errors appear to be some kind of communication problem. If someone have found any way out of that , I would be very thankful to know.

## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): 
- IDF version: v4.2
- Build System:         [idf.py]
- Operating System:     [Windows]
- Power Supply:         [USB]

felipebayres, 2020-11-17:
For some reason , It appear that the pins for RX and TX are switched in the datasheet,I searched some forums and i couple of guys  changed the pins to RX=26 e TX=27 and the connection with SIM800L is established. But I'm still facing these errors:

```
E (22736) esp-modem: esp_modem_dte_send_cmd(229): process command timeout
E (22736) sim800: sim800_set_working_mode(307): send command failed
E (22736) esp-modem: esp_modem_dte_change_mode(319): set new working mode:0 failed
E (22746) esp-modem: esp_modem_stop_ppp(499): enter command mode failed 
```

david-cermak, 2020-11-20:
Hi @felipebayres 

Thanks for the report! Yes we are currently having trouble with exiting the data mode, which seems to be your issue as well. Could you please try to apply the patch posted [here](https://github.com/espressif/esp-idf/issues/6024#issuecomment-729568712) and check if it helps? The patch is against master, but should apply to `v4.2` as well.

felipebayres, 2020-11-25:
Hey , @david-cermak , thank you for the attention. I ended up solving the problems by searching in this issue(#3506), the commit made corrected the errors. I had to initiate the sim800l correctly too , because it had some problems with it in the code that we first used.

Alvin1Zhang, 2021-03-01:
Fixed in https://github.com/espressif/esp-idf/commit/6270de58adc6af804b18ed3143085f0ea3330bce, feel free to reopen.

closed 2021-03-01 by Alvin1Zhang

iamnarendrans, 2024-05-21:
@david-cermak Hey, I saw the issue(#3506), but i will not be able to debug the error. How did your corrected that error?