espressif/esp-idf #6127: How can I change the value of a BLE characteristic in the code? ‎

Zeni241, 2020-11-17:
I am following sample of “gatt_server_service_table” at https://github.com/espressif/esp-‎idf/tree/release/v4.1/examples/bluetooth/bluedroid/ble/gatt_server_service_table .‎
My question is how can I change the value of characteristic in the code? I mean if I want to convey ‎changes in temperature value to client, how will I change the characteristic value, so that it could be ‎notified when notification has been enabled by client?‎

xiewenxiang, 2020-11-18:
Hi ,

You can use this API:
```c
/**
 * @brief           This function is called to set the attribute value by the application
 *
 * @param[in]       attr_handle: the attribute handle which to be set
 * @param[in]       length: the value length
 * @param[in]       value: the pointer to the attribute value
 *
 * @return
 *                  - ESP_OK : success
 *                  - other  : failed
 *
 */
esp_err_t esp_ble_gatts_set_attr_value(uint16_t attr_handle, uint16_t length, const uint8_t *value);
```

Alvin1Zhang, 2020-11-18:
Thanks for reporting, for How questions, please help report on our forum at https://esp32.com, thanks.

closed 2020-11-18 by Alvin1Zhang