espressif/esp-idf #6128: ADC1 on sensor_vp (IO36) not working as expected

cardtap, 2020-11-18:
Hi all,
I am trying to analogRead GPIO 36 on an ESP32-WROOM-32UE module. GPIO is also known as sensor_vp, on ADC1.
For some reason, every reading gives me a value of 4095 with my current setup.
I initialize the GPIO with pinMode(36, INPUT); then read with int reading = analogRead(36);

Reading the voltage of the pin with my Fluke gives me 0.39v, so the reading should be roughly 495 at default settings.
My connections are as follows:
ESP32 sensor_vp <--> bus
Ground (ESP32) <--> 47k ohm <--> bus
8.4v (same gnd ref) <--> 330k ohm <--> bus
12v (same gnd ref) <--> 330k ohm <--> bus

Can anyone think of a reason this wouldn't work properly? When I load the bus down to ground with a 220 ohm resistor, it reads low, but with the current setup it doesn't read anything other than 4095.

Thanks for any ideas y'all might have.

I'm posting this in the esp-idf issues because I believe it might be some sort of idf bug, since not many people use sensor_vp. I do not use the hall sensors, and WiFi isn't turned on until well after the adc readings start. Is there any other functions that would affect this?

cardtap, 2020-12-04:
Just following up, has anyone had a similar issue, or found a fix for this? Thanks

istokm, 2020-12-08:
You'd probably get more answers on the [arduino-esp32](https://github.com/espressif/arduino-esp32) repo.. but there is one thing, you said it reads "low" when pulled down with a 220R. Could it be that `pinMode(36, INPUT)` is meant only for digital reads? Meaning that anything above the threshold voltage (~1.3V if I remember correctly) the value would be 1 - which the arduino library might return as max value during a analog read (4095 is max analog value).
Can't test it right now, but I looked for `analogRead` examples and none of them use the `pinMode` function.

cardtap, 2020-12-08:
> You'd probably get more answers on the [arduino-esp32](https://github.com/espressif/arduino-esp32) repo.. but there is one thing, you said it reads "low" when pulled down with a 220R. Could it be that `pinMode(36, INPUT)` is meant only for digital reads? Meaning that anything above the threshold voltage (~1.3V if I remember correctly) the value would be 1 - which the arduino library might return as max value during a analog read (4095 is max analog value).
> Can't test it right now, but I looked for `analogRead` examples and none of them use the `pinMode` function.
> 
> _edit: I guess your fluke actually gives you < 1.3V... so that would be under the threshold, but It also has a ~10M pulldown resistor for measuring voltage, so it may be skewing the results with your high resistance pullups. Still, give it a try, I couldn't find an `analogRead` example using the `pinMode` function_

Hi Istokm, thank you for the reply!

Those are some good ideas, and I will try without pinMode (although in the Arduino environment, pinMode is usually used to clear previous pin settings if setting as input), as well as I'll try some lower (1k and 10k) resistors instead of the 330k and 47k that I currently am using. They're 0402 so that'll be fun! 
As far as other tests I've tried, I have done 
int val = adc1_get_raw(ADC1_CHANNEL_0);
But this still doesn't function. I guess my main issue is figuring out if the problem is hardware or software related. I'll keep testing things...

Thanks for your time

cardtap, 2021-05-24:
@istokm I wanted to let you know we solved the issue. It turned out to be surprisingly obvious, but also very difficult to find. On our device, we have an RFID reader that interfaces using Weigand protocol. Apparently Weigand is active low, pulled up to 5v inside the RFID reader when idle. This caused some other ESP32 GPIO pins to be pulled to 5V which is terrible obviously, and caused some of these other pins to exhibit weird and non-obvious behaviors!

closed 2021-05-24 by cardtap

istokm, 2021-05-25:
That is a peculiar problem indeed, I'm glad you managed to solve it. I've seen internal voltage/current leakage on the ESP32 myself, so I'm not that surprised tbh... I don't have the microelectronics skills to tell you why it happens though.