espressif/esp-idf #6129: MQTT Connection Failed after changing Azure IoT Hub Certificate

eduhash-kimyk, 2020-11-18:
## Environment

(Now we're investigating current development environment with Device manufacturer and we'll update detailed info later)

I (309) cpu_start: ESP-IDF:          v4.0-dev-1443-g39f090a4f-dirty
I (316) cpu_start: Single core mode
I (320) heap_init: Initializing. RAM available for dynamic allocation:
I (327) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (333) heap_init: At 3FFCA040 len 00015FC0 (87 KiB): DRAM
I (339) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (346) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (352) heap_init: At 4008AC94 len 0001536C (84 KiB): IRAM
I (358) cpu_start: Pro cpu start user code
I (376) spi_flash: detected chip: generic
I (376) spi_flash: flash io: dio
I (377) cpu_start: Starting scheduler on PRO CPU.

## Problem Description

Hi all, we're using MQTT connection with MS Azure IoT Hub from last year.
From last week, our firmware doesn't connect to IoT Hub with CORRUPT HEAP dump.
1. We heard that IoT Hub's server certificate was changed last week like below link :
   https://techcommunity.microsoft.com/t5/internet-of-things/azure-iot-tls-changes-are-coming-and-why-you-should-care/ba-p/1658456
2. Before changing IoT Hub's certificate, there's no pb to connect with firmware & IoT Hub.
3. MS guaranteed that our certificate file (refer to below link) is no problem to use
   https://cacerts.digicert.com/BaltimoreCyberTrustRoot.crt.pem
4. When we lookup device log with debug mode, we found that CORRUPT HEAP occurred in 'ServerHello' step.
   ( refer to Debug Logs )
5. Our connection info describes in 'Code to reproduce this issue' Section

We tried to find similar issues on here, but we cannot find suitable solution.
Could you help me for our issue?

### Expected Behavior
MQTT Connected with MS Azure IoT Hub without any errors

### Actual Behavior
Connection failed with CORRUPT HEAP
(Refer to Device Log)

### Code to reproduce this issue
static void mqtt_app_start(void)
{

    const esp_mqtt_client_config_t mqtt_cfg = {
      .uri = mqtt_url,
      .username = username,
      .client_id  = clientid,
      .password = (char *)sasTokens,
      .cert_pem = (const char *)iot_openit_pem_start,
    };   

    ESP_LOGI(TAG, "[APP] Free memory: %d bytes", esp_get_free_heap_size());
     client = esp_mqtt_client_init(&mqtt_cfg);
    esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, client);
    esp_mqtt_client_start(client); 
};

## Debug Logs

I (11644) eth_example: Ethernet Got IP Address
I (11644) eth_example: ~~~~~~~~~~~
I (11644) eth_example: ETHIP:192.168.XX.XX
I (11654) eth_example: ETHMASK:255.255.255.0
I (11654) eth_example: ETHGW:192.168.XX.XX
I (11664) eth_example: ~~~~~~~~~~~
I (11664) MQTTS: The current date/time in  is: Mon Nov 16 15:12:37 2020
start_sas
 key > ...
decodedBytes(hex)[32] >>> (hidden)
[... 214 lines cut ...]

I (13264) mbedtls: ssl_tls.c:2721 in_left: 0, nb_want: 5

I (13264) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (13274) mbedtls: ssl_tls.c:2742 <= fetch input

D (13274) mbedtls: ssl_tls.c:4056 input record: msgtype = 22, version = [3:3], msglen = 3844

I (13284) mbedtls: ssl_tls.c:2536 => fetch input

I (13294) mbedtls: ssl_tls.c:2697 in_left: 5, nb_want: 3849

I (13304) mbedtls: ssl_tls.c:2721 in_left: 5, nb_want: 3849

I (13304) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned 1435 (-0xfffffa65)

I (13314) mbedtls: ssl_tls.c:2721 in_left: 1440, nb_want: 3849

I (13324) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned 1440 (-0xfffffa60)

CORRUPT HEAP: multi_heap.c:432 detected at 0x3fff0a98
abort() was called at PC 0x40085337 on core 0

ELF file SHA256: 2744357b4897769bc...

david-cermak, 2020-11-18:
Hi @eduhash-kimyk 
Could you please turn on the `CONFIG_HEAP_CORRUPTION_DETECTION` config to see more details on the corrupt heap as described here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#finding-heap-corruption?

eduhash-kimyk, 2020-11-18:
Hi @david-cermak, nice to meet you.
I’ll ask to our device engineer to turning on config and recapturing log. I’m sorry but it may takes a time due to timezone gap with your location.

eduhash-kimyk, 2020-11-20:
Hi @david-cermak ,

We're trying to follow your guide but our development environment doesn't work 'Menu config'.
(FYI, we doesn't use Heap tracing mode until now so nobody experienced how to turn on.)
So, our engineer trying to change Heap tracing option using 'sdkconfig.defaults' and it changed 'sdkconfig.h' like below :

<sdkconfig.h>
#define CONFIG_HEAP_POISONING_COMPREHENSIVE 1
#define CONFIG_HEAP_TRACING_STANDALONE 1
#define CONFIG_HEAP_TRACING 1
#define CONFIG_HEAP_TRACING_STACK_DEPTH 2
#define CONFIG_HEAP_TASK_TRACKING 1
#define CONFIG_LOG_DEFAULT_LEVEL_INFO 1
#define CONFIG_LOG_DEFAULT_LEVEL 3

But our firmware still doesn't work Heap tracing until now.

I'm sorry but could you let me know how to change config value of CONFIG_HEAP_CORRUPTION_DETECTION directly in a certain file considering our situation, please?

david-cermak, 2020-11-20:
To enable basic heap trancing, you can start with these options in your `skdconfig`:
```
CONFIG_HEAP_POISONING_COMPREHENSIVE=y
CONFIG_HEAP_TRACING_STANDALONE=y
CONFIG_HEAP_TRACING=y
CONFIG_HEAP_TRACING_STACK_DEPTH=2
CONFIG_HEAP_TASK_TRACKING=y
```
Which is pretty much the same as what you've mentioned above (but not sure if your options wouldn't get overwritten if set directly in `sdkconfig.h`)

In general, heap corruption issues are very difficult to track down and usually just turning the heap trancing on wouldn't reveal the source of the issue, but could give some pointers. It's also good to periodically call `heap_caps_check_integrity_all()` or set breakpoints to memory location where the corruption happens (using JTAG or with `esp_set_watchpoint()`). All this is described in the documentation I've linked above.

Also from the code snippet you've posted, I can see it sets the `.cert_pem = (const char *)iot_openit_pem_start,` to your server certificate. Where is the certificate located? (in flash, added via makefile or as `const ptr` in the code, allocated on heap, or as a static/global variable)?

eduhash-kimyk, 2020-11-20:
Thanks for your help, David.

I’ll transfer to our engineer to follow your guide and give information to you about certificate location in our device. FYI, you can find link in ‘Problem Description’ which certificate file we use.

I’m not sure but what I remember is... certificate key value is copied from .pem file and paste to firmware source code as assigning string value to a certain variable.

eduhash-kimyk, 2020-11-23:
Hi David,

1. We tried to apply 'skdconfig' following your guide on friday, but still we cannot capture heap tracing log until now.
   We'll find more and get back to you with how we tried.

2. Also, I'll explain how to apply certificate file in our device firmware you asked.

   1) Put our certificate file(PEM format) into folder 'project/main/'
   2) Assign certificate file in 'component.mk' like below :
       [image]

   3) Read certificate file in 'project/main/Mqtt_.c' like below :
       [image]

3. FYI, I'll show where ESP-IDF is located in our environment : 
   [image]

If you have any question, please let me know.
Thanks for your great help.

eduhash-kimyk, 2020-11-27:
Hi David,

We failed to turn on Heap Tracing Mode so we decided to upgrade ESP-IDF module to 4.1 (from 4.0.0).
After upgrading, no more issue occurred until now so we'll apply new version of the firmware to our real devices until next sunday.

For my understanding, do you know which changes are related to solve this issue?

david-cermak, 2020-11-27:
> do you know which changes are related to solve this issue?

TBH, I expected the heap debugging to show the root cause somewhere in the application code, as the workflow described above is such a *very very common* usecase, that majority of customers use. It is possible though that the issue might have been in the IDF, too, due to the version you used `v4.0-dev-1443-g39f090a4f-dirty`. Please note that it's always better to stick with releases, e.g. there's already a `v4.0.2` release tag, or we recommend to use release branches.

eduhash-kimyk, 2020-11-30:
Hi David,

We've done rebuilding our firmware using release 4.1 with some minor bug fixes and applied to our devices on weekend.
After that, no more error was occurred on MQTT connection to Azure IoT Hub.
Still we're finding real root cause but we identified that retaining latest version of ESP-IDF is best way to solve unexpected problem as you mentioned.

Thanks for your good coaching. I can close this issue.

closed 2020-11-30 by eduhash-kimyk