espressif/esp-idf #6132: #include gives no such file or directory

veelox321, 2020-11-18:
Hi, my project tree look like this

**TOP LEVEL FOLDER:**
- main folder
- component folder
- CMakeLists.txt with :
 
```
    cmake_minimum_required(VERSION 3.5)
    set(SUPPORTED_TARGETS esp32)
    include($ENV{IDF_PATH}/tools/cmake/project.cmake)
    project(FirmwareRTOS)  
```

**MAIN FOLDER :**
- main.cpp
- main.h
- component.mk (empty)
- CMakeLists.txt :

`    idf_component_register(SRCS "main.cpp" INCLUDE_DIRS ".")`

**COMPONENT FOLDER :**
- Sensor folder
- STC3115_Drivers folder

**STC3115_DRIVERS FOLDER :**
- stc3115_Battery.h
- stc3115_Driver.cpp /.h
- stc3115_I2C.cpp /.h
- component.mk (empty)
- CMakeLists.txt 

`    idf_component_register(SRCS "stc3115_Driver.cpp" "stc3115_I2C.cpp" INCLUDE_DIRS ".")`

**SENSOR FOLDER :**
- sensor.cpp / .h
- sensorInit.cpp / .h
- sensorRoutine.cpp / .h
- component.mk (empty)
- CMakeLists.txt :

  `  idf_component_register(SRCS "sensor.cpp" "sensorInit.cpp" "sensorRoutine.cpp" INCLUDE_DIRS ".")`

Inside sensorInit.h there is

    `  #include "stc3115_Driver.h"  `

i get the following error :

    `No such file or directory`

I've tried adding an include folder in every component folder and into the previous component.mk i've added :

```
    COMPONENT_SRCDIRS := .
    COMPONENT_ADD_INCLUDEDIRS := include/
```

No success

igrr, 2020-11-18:
Can you try the following?
```cmake
idf_component_register(SRCS "stc3115_Driver.cpp" "stc3115_I2C.cpp"
                       INCLUDE_DIRS "."
                       REQUIRES STC3115_Drivers)
```

(or `PRIV_REQUIRES` instead of `REQUIRES` if `sensorInit.h` is not a public interface of `Sensor` component)

See https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/build-system.html#component-requirements for more details.

veelox321, 2020-11-19:
This solution work, thanks you for your help. 
Regarding include from esp idf component, which relative file i'm suppose to be in.
am i starting in ../esp-idf/components/any file/include? Am i calling #include <driver/gpio.h>?

igrr, 2020-11-19:
Unfortunately this will depend on the component, and we don't have a way yet to show the canonical include name in the documentation. Suggest checking esp-idf examples of the component you intend to use, for now.

closed 2020-11-20 by veelox321