espressif/esp-idf #6134: MODBUS timeouts cause errors in data or no data

mbuckaway, 2020-11-19:
- Development Kit:      ESP32-DevKitC clone (also used ESP32S2 SAOLA)
- Kit version: v4 (hardware isn't relavent - both ESP32 and ESP32S2 boards have same issues)
- Module or chip used:   ESP32-D0WDQ6 (revision 1) (or ESP32S2)
- IDF version: v4.3-dev-1720-g494a124d9 and  v4.2-beta1-227-gf0e87c933
- Build System:  idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux or macOS (used both)
- Using an IDE?: Visual Studio code, but only as editor. Build is by command line.
- Power Supply:         USB or external 5V (doesn't matter)

## Problem Description
The application is a data logger for a EPever Tracer Solar Charge controller. The charge controller has a modbus port running at 115200bps. I'm reading 7 input regs and one discrete reg from their set of 30+. This is done on 60 second cycle. The application uses WIFI to get an internet connection and MQTT to sent the data to Thingspeak.

The modbus logger is based on the modbus master and MQTT examples from the IDF.

### Expected Behavior

Data is read from all registers every time without error.

### Actual Behavior

Data is read most of the time. Randomly, sometimes after a few minutes, and sometimes after an hour, an error reading one of the registers occurs:
MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).^

Sometimes, that error is followed by:
MB_PORT_COMMON: vMBMasterRunResRelease(172): vMBMasterRunResRelease: resource release failure.

...after which, the modbus code stops working and the ESP32 requires a reset. Once the ESP32 reboots, it will continue working properly again.

I have gone through many iterations trying to resolve the issue. I've referred to issue #5986 to see if that would help. The issue seems to be thread related. If I remove WIFI and MQTT, I can run the modbus for hours without issue. Originally, the problem presented itself by reads after the error being shifted - registers would be read in wrong order - off by one. When I used the retry logic from #5968, it would stop returning data. All return values would be zero. Then, I tried IDF v4.3 (master), and the shifted data (off by one) problem returned.

What I have done:
- refactored the pointer math in the example and simplified the code (thinking the off by one problem was pointer math related) - no effect
- reduced the number of register reads to 8 (used to get 13) - no effect
- increased the delay between register read requests - no effect
- added a retry when a read register fails (as in #5986) - fixed the problem, until a vMBMasterRunResRelease error occured, and then data to stop being returned (all zeros)
- added a delay to the retry on error - no effect - same problem with or without the delay
- reviewed the patch from #5986 - and hacked the freemodbus Kconfig to allow a modbus thread priority up to 23 and set that to 23. Current Kconfig limits it to 10. - this just lengthened the amount of time before the error appeared and the modbus code stopped working.
- switched from IDF v4.2 (release/v4.2 branch) to v4.3 (master branch) - the off by one issue returned. Any error caused a shift in the return values from the modbus.

This problem has me baffled. It appears to be thread related because if I stop the MQTT thread or shutdown WIFI, it works. I have config options in my Kconfig to disable both as part of my testing.

It would appear the only fix is to either shut down the modbus on error and restart it, or restart the device in code. As it is a data logger, it's not mission critical, but the loss of reads it getting frustrating. This device is set to be deployed with the charge controller and solar charging system in the field, so manual intervention will require a 20min drive to work on the device. Currently, it sit on my workbench while it's in development.

This issue is simular but not the same as #5986 as I am not updating the OTA.

### Code to reproduce this issue

You can find my code at [ESP32 EPSolar](https://github.com/mbuckaway/esp32-epsolarmodbus)

## Debug Logs

Debug logs are attached.
error10.log - IDF v4.3
error9.log - IDF v4.2
error8.log - IDF v4.2 (no delay between reg reads)

[error8.log](https://github.com/espressif/esp-idf/files/5563967/error8.log)
[error9.log](https://github.com/espressif/esp-idf/files/5563968/error9.log)
[error10.log](https://github.com/espressif/esp-idf/files/5563969/error10.log)

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5563974/sdkconfig.txt) - last sdkconfig used under IDF v4.3. I've blanked out the passwords in the file.

alisitsyn, 2020-11-19:
Hello @mbuckaway,

This issue is known and happens when higher priority tasks with WiFi and flash r/w access exist in your application and Modbus is not configured correctly. When the response time configured incorrectly (CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND < actual slave response), the timeout is expired, the current transaction treated as completed by Modbus FSM, but  the response to previous transaction is received. This causes desincronization of FSM and cause errors. Your maximum response time is set to 200ms, that is less then actual response time as I can guess. According to this I would recommend these steps to impove processing:
1. Apply the patch for v4.1 from #5986.
2. Change the CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND to the maximum slave response time (your current value = 200ms).
3. Set CONFIG_MB_TIMER_PORT_ENABLED =n in the Kconfig, this can extend the processing time when MB serial task suspended periodically by other high priority tasks (your current value = y).
4. Increase CONFIG_FMB_PORT_TASK_PRIO = ~your maximum tasks priority.
5. Set CONFIG_FMB_TIMER_ISR_IN_IRAM =y, to allow interrupts placement into IRAM.
6. Use retries for Modbus transaction.

Try the above recomendations and if this does not work, please enable the log verbosity in Kconfig to Debug to enable modbus debug messages and send me extended log of Modbus. This should help to avoid issues. To fix the state machine from misalingnment try to use patch below from #5738:
0002-bugfix-workaround-to-clear-Modbus-stack-misalignment.patch.log

Please let me know the results.

Thanks.

mbuckaway, 2020-11-19:
Thanks for the quick response.

I've tried to apply the patch from #5986, but I am using v4.2 and it does not apply cleanly - and there are changing outside the freemodbus component. For the moment, I ignored the patch.

I had already implemented the retry mechanism from $5986, without any effect. The first few retries would work, but whenever the vMBMasterRunResRelease happened, the code stopped working (off by one error or no data at all).

So, I did the following:
* set CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND=400 and CONFIG_MB_MASTER_TIMEOUT_MS_RESPOND=400 (two items?)
* set CONFIG_MB_TIMER_PORT_ENABLED=n and CONFIG_FMB_TIMER_PORT_ENABLED=n (don't know why there are two)
* set CONFIG_FMB_SERIAL_TASK_PRIO=23 (CONFIG_FMB_PORT_TASK_PRIO doesn't exist) - my highest task priority is 9 otherwise.
* set CONFIG_FMB_TIMER_ISR_IN_IRAM=y is the default

These had no effect.

Then, I applied the 0002-bugfix-workaround-to-clear-Modbus-stack-misalignment.patch.log patch from  #5738. It required some hacking as it did go into v4.2 cleanly, and that seems to work. The errors still occur including the vMBMasterRunResRelease error, but the code is able to recover. The code has been running for some hours.

I'm going to leave the code running and hopefully it will continue to work correctly....at which point, I may create a patch against v4.2.

alisitsyn, 2020-11-20:
I have to create the same patch for v4.2. The patch increase the stability of Modbus when it works with higher priority tasks. The 'misalingnment' patch allows to recover from errors. You do not need to change the CONFIG_MB_TIMER_PORT_ENABLED if the patch is not appied. 
> set CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND=400 and CONFIG_MB_MASTER_TIMEOUT_MS_RESPOND=400 (two items?) set CONFIG_MB_TIMER_PORT_ENABLED=n and CONFIG_FMB_TIMER_PORT_ENABLED=n (don't know why there are two)

I don't know why you have two items in your config. Probably because you applied first patch incorrectly.

mbuckaway, 2020-11-20:
As for the duplicate items, I did:
* on a clean checkout of v4.2, went to the examples/protocols/modbus/serial/mb_master directory
* checked that there were no incorrect (duplicate) items in sdkconfig.default (it correctly sets up CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND, etc)
* ran idf.py menuconfig, and hit save (no changes to anything)
* checked sdkconfig for MB_MASTER_TIMEOUT_MS_RESPOND (searched on this keyword), and found CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND and CONFIG_MB_MASTER_TIMEOUT_MS_RESPOND. The CONFIG_MB_MASTER_TIMEOUT_MS_RESPOND is outside the Freemodbus section. There appears to be several duplicate MB and FMB items.

I also did the same one the master (v4.3-dev-1720-g494a124d9) branch and got the same result.

While I have something working, it would be useful to have changes incorporated into a main release or at least the master branch. I could go back to v4.1, but I switching between the ESP32 and ESP32S2 depending on which board I use.

alisitsyn, 2020-11-30:
@mbuckaway,

Sorry for very late response. The CONFIG_MB_MASTER_TIMEOUT_MS_RESPOND is from old revisions of Modbus (v4.0, v4.1 for example). It was renamed to CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND  later. However when you checkout different branches and compile the example you do not clean the sdkconfig file. I think this is the reason. This should work in spite of this.

The fixes of cause will be in master. Unfortunately they will be delayed. If you need urgent update, I can create patch for you against v4.2.

mbuckaway, 2020-11-30:
Thanks for the update.

I manually patched v4.2 myself and it is current running and working in the field. Having a v4.2 patch would be useful, but I won't be able to test it for a while as I only have the one charge controller. I can wait until master is updated unless, for some reason, my device stops working. It would be nice to know when the update is apply, though.

alisitsyn, 2020-12-01:
@mbuckaway,

I would propose you to test your device with adopted patch for now. Once you have any issues just let me know here or open new issue. I will let you know about merge of related MR.

mbuckaway, 2020-12-01:
If you can provide the patch against v.4.2, I can arrange to build and test the device.

alisitsyn, 2020-12-02:
@mbuckaway,
The patch for v4.2 is below:
[backport_latest_fixes_v42.patch.log](https://github.com/espressif/esp-idf/files/5629482/backport_latest_fixes_v42.patch.log)
Please test it and let me know if you have any issues. 
Thanks.

mbuckaway, 2020-12-02:
@alisitsyn 

Thanks. Will install and test. It will take some time because the device is currently on a solar charger out in a forest and it has no OTA yet.

alisitsyn, 2020-12-18:
@mbuckway,

Please take a look to these patches: https://github.com/espressif/esp-idf/issues/5986
Please let me know when you have any update related to this.

mbuckaway, 2020-12-18:
Thanks for the extra info. I intend to look into updating the firmware for the solar charge control monitor before the new year.

mbuckaway, 2020-12-31:
To test out the new code, I setup a new project based on my other modbus project. This project reads the temperature and humidity values from a modbus temperature sensor. I checked out a fresh copy of IDF v4.2 and applied the back port patch. I cleaned the old sdkconfig I had from the old project, and setup the code to run. After about a day of polling the device on 60 second intervals, it read correctly.

This was not always the case. In the past, I had no luck trying to test the ESP32 modbus support with this modbus sensor. Of course, this sensor runs at 9600 baud and only has two values to read but it seems to work.

Now that I have some idea the code will function, the next step will be to update the solar charge monitor.

alisitsyn, 2021-01-06:
@mbuckaway, 
Thank you for update. Let me know if you have further information about your project or have any issues with Modbus.

ThermIQ, 2021-03-04:
I can confirm this is also happening happening in v4.3-beta1

**Setup:**
uart2 @115200, 
polling ~40 registers in 30s interval
prio 10
Running for several hours when i get the following Error from MBR:

```
E (35622299) MB_CONTROLLER_MASTER: mbc_master_get_parameter(83): Master get parameter failure, error=(0x108) ESP_ERR_INVALID_RESPONSE). 
E (35622309) MBR: Characteristic #15 (REG_SENSOR_RHS) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
```

After that all readings are of by one.

Yuva86, 2021-03-05:
@alisitsyn 
I use v4.2 IDF and trying to playaround with Modbus master example. I replaced the master.c file with the one from "simple modbus example" available in https://esp32.com/viewtopic.php?t=15121 . Unfortunately I get the timeout error everytime. I tried applying the backport patch(backport_latest_fixes_v42.patch) attached in this thread and the issue still exists. I dont have any other tasks running in the application. Before I build my own application, I wanted to ensure if the modbus works but stuck in the modbus timeout error. And we use gpios(RX-IO34, TX-IO4, RTS-IO2) for uart communications which are then connected to slaves via ST3485Eb. Any suggestions, please.

alisitsyn, 2021-03-05:
@Yuva86,
The err = 0x107 (ESP_ERR_TIMEOUT), means no response from slave after configured by CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND timeout. The v4.2 can work correctly with the "simple modbus example" or with default examples. The issue is usually related to your infrastructure setup but not the stack error. In order to find the reason for issue please follow the items below even if you checked it before. It is better to start with the standard examples first.

1. Ensure your board connections with ST3485Eb are correct [rs485 connection](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#interface-connection-options).
2. Check the connection using [official RS485 check example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo_rs485). 
3. Once the RS485 are functional and you able to receive echo. Follow with setup your slave and master examples as described in [modbus master-slave](https://github.com/espressif/esp-idf/tree/master/examples/protocols/modbus/serial). I suggest you to set the CONFIG_MB_MASTER_TIMEOUT_MS_RESPOND=500 in the kconfig menu in mb_master example. Compile and run these examples. Please follow the examples documentation during setup.
4. Set the log verbosity to Debug in kconfig menu. 
[image]
5. Please save the mb_master and mb_slave communication log and share it here.

Thank you.

alisitsyn, 2021-03-05:
@ThermIQ,

> After that all readings are of by one.

Does this mean that all other following reading are correct? 
The error `ESP_ERR_INVALID_RESPONSE ` means that incorrect or fragmented response is received from slave and may be happened if your application uses other tasks with priority higher than Mod bus which do intensive work or the tasks which call the flash read write functions which disable CPU cache and can cause interrupt handler latency when the handler is in Flash memory. Please try to follow recommendations above to configure `CONFIG_FMB_SERIAL_TASK_PRIO`, `CONFIG_FMB_TIMER_ISR_IN_IRAM`, `CONFIG_MB_MASTER_TIMEOUT_MS_RESPOND`, `CONFIG_MB_TIMER_PORT_ENABLED`.
Unfortunately this can happen and your application should be configured accordingly to avoid issues. The errors can happen sometime and it is allowed to retry request in this case. The debug log can explain what happened on the stack side in case of error.

ThermIQ, 2021-03-05:
@ESP_Alisitsy,
Thanks for explanation. I will check again that settings are correct.

All sub-sequent reads, to all addresses and forever, are of by one after the above error until reset and restart. I do polling ~40 registers in 30s interval so there's a silent gap

Would you recommend to restart the modbus master when read fails? What is an appropriate restart sequence?
`E (35622309) MBR: Characteristic #15 (REG_SENSOR_RHS) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).`

Would it make more sense to reset the MASTER state machine when:
`E (35622299) MB_CONTROLLER_MASTER: mbc_master_get_parameter(83): Master get parameter failure, error=(0x108) ESP_ERR_INVALID_RESPONSE). `

alisitsyn, 2021-03-05:
@ThermIQ,

Thank you for extended information. Yes, you are right, the state machine has to be reset in this case to prevent misalignment. Please see the: https://github.com/espressif/esp-idf/issues/5738 and try to apply `0002-bugfix-workaround-to-clear-Modbus-stack-misalignment.patch.log`. 
Sorry for inconvenience and delay with the merge of the fixes. Let me know the results. 
Thanks.

ThermIQ, 2021-03-05:
The `0002-bugfix-workaround-to-clear-Modbus-stack-misalignment.patch.log` instructions are a bit to complex/messy for me to follow. What is your plan for next push? Will that also include the fix for memory region check mentioned [here ](https://esp32.com/viewtopic.php?f=13&t=14119&start=30)

I'm happy to try it out as soon there is a proper push, I have a good working setup, continously logging data but not yet in commercial deployment so there's room for flexibility.

ThermIQ, 2021-03-05:
Will change try to change `CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND=200` to 400 which is new (?) upper limit of timout

alisitsyn, 2021-03-05:
> What is your plan for next push? Will that also include the fix for memory region check mentioned here 

Unfortunately I can not give you expectations on when and what will be pushed to github.

The increased response time should help. Initially it was too small in the kconfig but was changed long time ago. If the (kconfig esponse time) >= maximum slave response time the above issue does not happen. The issue happens when timeout event is triggered but response from slave is received.

ThermIQ, 2021-03-08:
changing to `CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND=400` did NOT resolve the problem. It does occur less frequently.  It does seem to coincide with another task, with lower prio, is printing to uart using ESP_LOGx

While waiting for the next release I've implemented a hard reset if not getting the expected value out of a "static" register. Brute-force but kinda works in lab

alisitsyn, 2021-03-11:
@TermIQ,

Sorry for the delays with answer.

1. Try to increase the CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND to higher value (change the maximum in kconfig). 
2. CONFIG_FMB_SERIAL_TASK_PRIO could help to avoid issues. 
3. The other way is to start Modbus tasks on other code.
The issue happens because your other tasks do not allow to process the modbus data during the response time counted by timer. The logging increases the processing time even more. The fix above I mentioned will allow skipping later misalignment but does not solve the issue with processing.

Could you give me some little bit more information: sdkconfig file and what kind of tasks you are using and what priorities are set.
Your solution is just temporary. I will check the code and prepare the patch to fix the issue in your particular case while the issue is not fixed in master.

ThermIQ, 2021-03-18:
I lowered the amount of trace and debug which improves situation. I'm down to a reboot  ebery second day wich is ok for internal development.

I have a quite low load on both cores with a total of maybee 12 tasks., The modbus task is reading data every 15s, providing data to an occasional reporting task over mqtt locked to one core, some background calculation on the other.

alisitsyn, 2021-03-19:
@ThermIQ,

Thank you for update. Please also take a look to [this issue](https://github.com/espressif/esp-idf/issues/6700) and patch there. It allows to change affinity for Modbus tasks. I would recommend to start Modbus tasks and WiFi related tasks on different CPUs.

mbuckaway, 2021-04-20:
@alisitsyn,

Any updates on this issue? I ask because the code I deployed to the field (with the various patches against 4.2) worked fine for three months, although, lately, it won't run for more than a day without the data sequence problem. I'm about to spend time to update the code to use AWS IoT Core and ethernet (rather than WIFI). I assume managing the ethernet interface is as intensive as WIFI. Also thinking of starting with v4.3 as it progresses. Assigning modbus to it's own CPU is a useful hint.

alisitsyn, 2021-04-20:
@mbuckaway, @ThermIQ,

I remember about the problem and thinking about cardinal solution for it with current requirements. I do have current workaround for it but it won't be merged for now. I am going to return to this topic  as soon as complete other things. Understand your complains and going to help you soon. If it is possible please send your additional logs which would explain more details of issue.

mbuckaway, 2022-02-22:
@alisitsyn 

Any updates on this issue? Has the "appropriate" patches been migrated into newer branches? I've been running a few ESP32 devices talking to MODBUS solar controllers for some time, and I've not experienced any issues with the last series of patches. While the timeout error might still exist, the patches and the retries in the code seems make the system reliable. Currently looking to migrate to release V4.4.

alisitsyn, 2022-02-23:
@mbuckaway,

>  I've been running a few ESP32 devices talking to MODBUS solar controllers for some time, and I've not experienced any issues with the last series of patches.

Did you try the patch below:
[0001_modbus_fix_idfgh5107_v43.patch.log](https://github.com/espressif/esp-idf/files/8059152/0001_modbus_fix_idfgh5107_v43.patch.log)

Most of the patches are backported to release branches except this one. This patch allows avoiding issues when high-priority tasks terminate the Modbus tasks but increases the response processing time which may cause response timeout failure.

AshUK, 2022-04-13:
@alisitsyn thanks, so far looking good in testing. The v4.4 patch didn't apply correctly to `port_tcp_slave.c` so had to manually correct it to the following:

```
diff --git a/components/freemodbus/tcp_slave/port/port_tcp_slave.c b/components/freemodbus/tcp_slave/port/port_tcp_slave.c
--- a/components/freemodbus/tcp_slave/port/port_tcp_slave.c
+++ b/components/freemodbus/tcp_slave/port/port_tcp_slave.c
@@ -133,7 +133,8 @@ xMBTCPPortInit( USHORT usTCPPort )
         ESP_LOGE(TAG, "TCP client info allocation failure.");
         return FALSE;
     }
-    for(int idx = 0; idx < MB_TCP_PORT_MAX_CONN; xConfig.pxMbClientInfo[idx] = NULL, idx++);
+
+    for(int idx = 0; idx <= MB_TCP_PORT_MAX_CONN; xConfig.pxMbClientInfo[idx] = NULL, idx++);
 
     xConfig.xRespQueueHandle = xMBTCPPortRespQueueCreate();
     if (!xConfig.xRespQueueHandle) {
@@ -636,8 +637,6 @@ static void vMBTCPPortServerTask(void *pvParameters)
 void
 vMBTCPPortClose( )
 {
-    // Release resources for the event queue.
-
     // Try to exit the task gracefully, so select could release its internal callbacks
     // that were allocated on the stack of the task we're going to delete
     xShutdownSemaphore = xSemaphoreCreateBinary();
@@ -667,6 +666,7 @@ vMBTCPPortDisable( void )
             xConfig.pxMbClientInfo[i] = NULL;
         }
     }
+    free(xConfig.pxMbClientInfo);
     close(xListenSock);
     xListenSock = -1;
     vMBTCPPortRespQueueDelete(xConfig.xRespQueueHandle);

```

alisitsyn, 2022-04-13:
@AshUK,

Unfortunately, some pending backports of Modbus still merged in random order and this makes the patches obsolete often. The `freemodbus` component is also obsolete and will be removed soon from esp-idf. The new component is located [here](https://components.espressif.com/component/espressif/esp-modbus). The new updates will be available in this component which can work with ESP-IDF version 4.1 up to 5.0.

AshUK, 2022-04-13:
@alisitsyn I will look to use v1.0.0 of esp-modbus then. Does it include the aforementioned patches and fixes above ? https://github.com/espressif/esp-modbus

alisitsyn, 2022-04-13:
The v1.0.0 includes both of them but still does not include the latest ones. It will be merged a bit later. In spite of this, it is more reliable.

MarkEvens, 2022-06-30:
What is a patch for v4.4? I have the same issue.

thiker5, 2022-12-03:
> The v1.0.0 includes most of them but still does not include the latest ones. Some of the fixes from the patch above are not included. It will be merged a bit later. In spite of this, it is more reliable.

Thanks for the updates, is their a timeline when the dependency will have all the fixes in the patch?

alisitsyn, 2024-01-24:
The freemodbus component in esp-idf v4.4 will not be updated. The current version of that component corresponds to the version esp-modbus v1.0.7. All described issues are fixed in the latest esp-modbus component which is compatible with all version of esp-idf. The old freemodbus component needs to be excluded as described in the [documentation](https://docs.espressif.com/projects/esp-idf/en/v4.4.6/esp32/api-reference/protocols/modbus.html). It is recommended to use it in your software.

Feel free to reopen the issue if you have any comments.

closed 2024-01-24 by alisitsyn