espressif/esp-idf #6137: ESP32-S2 RTC-GPIO hold for ULP-FSM is not woking after first time.

pavlos77, 2020-11-19:
Hello, I want to blink a LED , when the ESP is in deep sleep mode ( CPU is off ) from ULP-FSM. I set the appropriate registers ( See below), i can hold the LED ON with no problem and then turn it OFF, but i cannot turn it ON again after the first time.I have read and printed the register and i can see that it set properly.Is it a bug or i am missing something?

Turn OFF LED
```
I_WR_REG_BIT(RTC_CNTL_PAD_HOLD_REG, rtc_io_number_get(LED_GPIO), 0)
I_WR_REG_BIT(RTC_GPIO_OUT_W1TC_REG, (uint8_t)(RTC_GPIO_OUT_DATA_W1TC_S + rtc_io_number_get(LED_GPIO)), 1),
I_WR_REG_BIT(RTC_CNTL_PAD_HOLD_REG, rtc_io_number_get(LED_GPIO), 1)
```

Turn ON LED
```
I_WR_REG_BIT(RTC_CNTL_PAD_HOLD_REG, rtc_io_number_get(LED_GPIO), 0)
I_WR_REG_BIT(RTC_GPIO_OUT_W1TS_REG, (uint8_t)(RTC_GPIO_OUT_DATA_W1TS_S + rtc_io_number_get(LED_GPIO)), 1),
I_WR_REG_BIT(RTC_CNTL_PAD_HOLD_REG, rtc_io_number_get(LED_GPIO), 1)
```

igrr, 2020-11-19:
@pavlos77 these snippets look correct! Could you show the complete code which exhibits the problem?

pavlos77, 2020-11-19:
Here is the complete code. 
```
#include "esp_sleep.h"
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "driver/adc.h"

#include "soc/rtc_cntl_reg.h"
#include "soc/rtc_io_reg.h"
#include "esp32s2/ulp.h"
#include "driver/rtc_io.h"

#define LED_GPIO GPIO_NUM_21
#define ARRAY_COUNT(x) (sizeof(x) / sizeof((x)[0]))
#define RTC_WORD_OFFSET(x) ((uint16_t)((uint32_t*)&(x) - RTC_SLOW_MEM))

RTC_DATA_ATTR uint32_t ulp_led_state;

enum labels {
        LED_OFF,
        LED_ON,
        END,
};

void init_ulp()
{
        const ulp_insn_t program[] = {
        I_MOVI(R3, RTC_WORD_OFFSET(ulp_led_state)),
        I_LD(R0, R3, 0),
        M_BL(LED_ON, 1),

        M_LABEL(LED_OFF),
        I_WR_REG_BIT(RTC_CNTL_PAD_HOLD_REG, rtc_io_number_get(LED_GPIO), 0),
        I_WR_REG_BIT(RTC_GPIO_OUT_W1TC_REG, (uint8_t)(RTC_GPIO_OUT_DATA_W1TC_S + rtc_io_number_get(LED_GPIO)), 1),
        I_WR_REG_BIT(RTC_CNTL_PAD_HOLD_REG, rtc_io_number_get(LED_GPIO), 1),
        I_MOVI(R0,0),
        I_ST(R0, R3, 0),
        M_BX(END),

        M_LABEL(LED_ON),
        I_WR_REG_BIT(RTC_CNTL_PAD_HOLD_REG, rtc_io_number_get(LED_GPIO), 0),
        I_WR_REG_BIT(RTC_GPIO_OUT_W1TS_REG, (uint8_t)(RTC_GPIO_OUT_DATA_W1TS_S + rtc_io_number_get(LED_GPIO)), 1),
        I_WR_REG_BIT(RTC_CNTL_PAD_HOLD_REG, rtc_io_number_get(LED_GPIO), 1),
        I_MOVI(R0,1),
        I_ST(R0, R3, 0),

        M_LABEL(END),
        I_HALT(),
    };

    ulp_led_state = 0;

    ESP_ERROR_CHECK(rtc_gpio_init(LED_GPIO));
    ESP_ERROR_CHECK(rtc_gpio_set_direction(LED_GPIO, RTC_GPIO_MODE_OUTPUT_ONLY));

    size_t program_size = ARRAY_COUNT(program);
    ESP_ERROR_CHECK(ulp_process_macros_and_load(0, program, &program_size));
    ESP_ERROR_CHECK(ulp_set_wakeup_period(0, 2000 * 1000));
    ESP_ERROR_CHECK(ulp_run(0));
}

void app_main(void)
{
        init_ulp();
        esp_deep_sleep_start();
}

```

boarchuz, 2020-11-19:
@pavlos77 This seems to be based loosely on https://github.com/boarchuz/s2-ulp-fsm-demo/blob/master/main/main.c but you will need the changes here - https://github.com/boarchuz/esp-idf/blob/s2-ulp-fsm/components/ulp/include/esp32s2/ulp.h - as there are different and new instructions for S2, and the IDF definitions haven't been updated yet. That's incomplete, but it's a start.

@igrr is 'legacy' macro support planned for S2 (and future)?

pavlos77, 2020-11-20:
@boarchuz Yes, i have used this code as reference, because the ULP wasn't working . I have already used the files "include/esp32s2/ulp.h" and "ulp.c" to my working directory.

Also, i test it with pure assembly and same result.

boarchuz, 2020-11-20:
You might need to wait until ULP-FSM is supported. Follow status here: 
https://esp32.com/viewtopic.php?t=14532

pavlos77, 2020-11-20:
I cannot find any support for ULP RISCV on branch "release/v4.2".Is it currently supported or it'll be developed in the future?

esp-wzh, 2022-04-18:
@pavlos77 According to your description, ULP is working fine before deepsleep, so I don't think the problem is with ULP. In your case, I set the log leve in the menuconfig to debug mode and observed that RTC_PERIPH was powered down in deepsleep, so this should be the cause of your problem. One solution is to call `esp_sleep_enable_gpio_wakeup()` before sleep to keep RTC_PERIPH powered in deepsleep.

jack0c, 2022-04-27:
@esp-wzh Do we have a option just keep ULP in deepsleep since `esp_sleep_enable_gpio_wakeup` is somehow different.
Do you run the code on your board to verify?

esp-wzh, 2022-04-27:
@jack0c Yeah，`esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);` is bettter!

jack0c, 2022-05-17:
@pavlos77 Are you still working on this?

jack0c, 2022-06-15:
@pavlos77 any updates?

jack0c, 2022-07-07:
Will close the issue since no update. Feel free to reopen it if necessary.

closed 2022-07-07 by jack0c