espressif/esp-idf #6138: fopen returns 0 when fileending bigger than 4 chars

LoPromise, 2020-11-19:
Basic Example:
void app_main(void) {
        ESP_LOGI(TAG, "Opening file");
	FILE *f = fopen(MOUNT_POINT"/hello.txtt", "w");
	if (f == NULL) {
		ESP_LOGE(TAG, "Failed to open file for writing");
		return;
	}
}

fopen seams to fail with 4wire SD-Setup when used with file endings longer than 4chars.

Alvin1Zhang, 2020-11-20:
@LoPromise Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

LoPromise, 2020-11-20:
Thanks for coming back so quickly, got a bit late yesterday... so here some more details

Current Version of esp-idf used: 0289d1cc81c210b719f28c65f113c45f9afd2c7b
Hardware: DevKitC(or reproduction... amazon...) + SD-Card in sdmmc-Mode
IDF Version: v4.3-dev-1521-g0289d1cc8
Build System: cmake --build . -- -v or idf.py shows up with both
Operating System: Linux
IDE: Eclipse or without
PowerSupply: USB

Steps to reproduce:
1. load example for sd card
2. modify line 151 to have FileEnding with 4 chars

Can't Provide debuglogs right now since i don't have a setup for it currently... I got a STMLink v2 lying around. link me instructions i guess
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5572411/sdkconfig.txt)
[sd_card.elf.txt](https://github.com/espressif/esp-idf/files/5572415/sd_card.elf.txt)

negativekelvin, 2020-11-23:
https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/components/fatfs/Kconfig#L86

igrr, 2020-11-23:
@LoPromise as suggested by @negativekelvin above, please enable CONFIG_FATFS_LONG_FILENAMES 
 option in menuconfig: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-fatfs-long-filenames

LoPromise, 2020-11-23:
you were right using longer file names fixed it. 
Might I suggest adding a note in documentation for this? because "/sdcard/test.flac" does not seam like a "long" Filename to me xD. I mean you will be able to find it now...
**Can be closed after this.**

igrr, 2020-11-24:
The term "long filenames" doesn't indicate the relative length of the filename, it refers to file names which are not [8.3 filenames](https://en.wikipedia.org/wiki/8.3_filename). Good point about the docs, will add a mention of this configuration option.

closed 2021-08-24 by espressif-bot