espressif/esp-idf #6139: Unexpect strerror() output

AxelLin, 2020-11-20:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v4.3-dev-1901-g178b122c145c
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

On esp32:

I notice the issue because I try to convert 113 errno to human readable string by
strerror(113), and I got "Software caused connection abort".
But in lwIP, the errno 113 is #define  EHOSTUNREACH   113  /* No route to host */.
So I would expect strerror(113) return "No route to host"

### Expected Behavior
expect strerror(113) return "No route to host"

### Actual Behavior
expect strerror(113) return "Software caused connection abort"

### Code to reproduce this issue

```c

{
        int i;
        for (i = 1; i < 128; i++)
        printf("%d %s\r\n", i, strerror(i));
}

```
Run above code snip on ESP32, I got:

1 Not owner
2 No such file or directory
3 No such process
4 Interrupted system call
5 I/O error
6 No such device or address
7 Arg list too long
8 Exec format error
9 Bad file number
10 No children
11 No more processes
12 Not enough space
13 Permission denied
14 Bad address
15 
16 Device or resource busy
17 File exists
18 Cross-device link
19 No such device
20 Not a directory
21 Is a directory
22 Invalid argument
23 Too many open files in system
24 File descriptor value too large
25 Not a character device
26 Text file busy
27 File too large
28 No space left on device
29 Illegal seek
30 Read-only file system
31 Too many links
32 Broken pipe
33 Mathematics argument out of domain of function
34 Result too large
35 No message of desired type
36 Identifier removed
37 
38 
39 
40 
41 
42 
43 
44 
45 Deadlock
46 No lock
47 
48 
49 
50 
51 
52 
53 
54 
55 
56 
57 
58 
59 
60 Not a stream
61 No data
62 Stream ioctl timeout
63 No stream resources
64 
65 
66 
67 Virtual circuit is gone
68 
69 
70 
71 Protocol error
72 
73 
74 Multihop attempted
75 
76 
77 Bad message
78 
79 
80 
81 
82 
83 
84 
85 
86 
87 
88 Function not implemented
89 
90 Directory not empty
91 File or path name too long
92 Too many symbolic links
93 
94 
95 Operation not supported on socket
96 
97 
98 
99 
100 
101 
102 
103 
104 Connection reset by peer
105 No buffer space available
106 Address family not supported by protocol family
107 Protocol wrong type for socket
108 Socket operation on non-socket
109 Protocol not available
110 
111 Connection refused
112 Address already in use
113 Software caused connection abort
114 Network is unreachable
115 Network interface is not configured
116 Connection timed out
117 Host is down
118 Host is unreachable
119 Connection already in progress
120 Socket already connected
121 Destination address required
122 Message too long
123 Unknown protocol
124 
125 Address not available
126 Connection aborted by network
127 Socket is already connected

igrr, 2020-11-24:
@AxelLin LwIP provides its own errno definitions only if LWIP_PROVIDE_ERRNO macro is defined. This shouldn't be the case in ESP-IDF, errno values should come from newlib. The value used for EHOSTUNREACH in newlib is different, but it is consistent with strerror output:

```c++
printf("%s\n", strerror(EHOSTUNREACH));
```
prints "Host is unreachable".

If you are seeing that EHOSTUNREACH is expanded to value 113 by the preprocessor somewhere, this is unexpected. Please give more details if this is the case.

AxelLin, 2020-11-24:
Use examples/protocols/sockets/tcp_client snip code as example.

I only made below change in the example code:
```
-            ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno);
+            ESP_LOGE(TAG, "Socket unable to connect: errno %d %s", errno, strerror(errno));
```

I make the example application to connect an unreachable IP:

I (3564) esp_netif_handlers: example_connect: sta ip: 192.168.1.102, mask: 255.255.255.0, gw: 192.168.1.1
I (3564) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.1.102
I (3574) example_connect: Connected to example_connect: sta
I (3574) example_connect: - IPv4 address: 192.168.1.102
I (3584) example: Socket created, connecting to 192.168.20.188:3333
W (3594) wifi:<ba-add>idx:1 (ifx:0, d8:47:32:7c:34:ba), tid:0, ssn:0, winSize:64
E (12844) example: Socket unable to connect: errno 113 Software caused connection abort

I would expect the strerror(errno) shows "No route to host" in this case.
So I'm confused, and would like to check the errno and the corresponding error strings.
Did I miss something?

AxelLin, 2020-11-24:
Looks like it will return errno 118 (Host is unreachable) only when the netif is down.

closed 2020-11-24 by AxelLin

AxelLin, 2020-12-30:
Just for reference, https://github.com/espressif/newlib-esp32/blob/master/newlib/libc/include/sys/errno.h

jcecconi, 2024-08-15:
Hi, did you solve this issue? I'm having the same problem!