espressif/esp-idf #6140: How do I process ESP32-S2 random PSRAM error or Image hash failed

coollofty, 2020-11-20:
The module is esp32-s2-wroom 2MB PSRAM

The partitions.csv
```
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, spiffs,  , 0x80000,
```
idf.py size
```
Total sizes:
 DRAM .data size:   13732 bytes
 DRAM .bss  size:   22928 bytes
Used static DRAM:       0 bytes (      0 available, nan% used)
Used static IRAM:       0 bytes (      0 available, nan% used)
Used stat D/IRAM:  125583 bytes ( 267633 available, 31.9% used)
      Flash code:  490183 bytes
    Flash rodata:  121008 bytes
Total image size:~ 736774 bytes (.bin may be padded larger)
```

The current overall problem is: it is easy to restart indefinitely, or suddenly restart during use, and then it will restart indefinitely. The three esp32-s2s in hand all have the same situation. The two logs that often appear during infinite restart are as follows:
```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x1914
load:0x40050000,len:0x18a8
load:0x40054000,len:0x210c
entry 0x40050314
I (48) boot: ESP-IDF v4.2-dev-1983-gccdfbb865-dirty 2nd stage bootloader
I (48) boot: compile time 16:26:31
I (48) boot: chip revision: 0
I (52) qio_mode: Enabling default flash chip QIO
I (57) boot.esp32s2: SPI Speed      : 80MHz
I (62) boot.esp32s2: SPI Mode       : QIO
I (67) boot.esp32s2: SPI Flash Size : 4MB
I (72) boot: Enabling RNG early entropy source...
I (77) boot: Partition Table:
I (81) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (95) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (103) boot:  2 factory          factory app      00 00 00010000 00100000
I (110) boot:  3 storage          Unknown data     01 82 00110000 00080000
I (118) boot: End of partition table
I (122) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x1d8b0 (121008) map
I (153) esp_image: segment 1: paddr=0x0002d8d8 vaddr=0x3ffc9b60 size=0x02740 ( 10048) load
I (156) esp_image: segment 2: paddr=0x00030020 vaddr=0x40080020 size=0x77ac8 (490184) map
0x40080020: _stext at ??:?

I (248) esp_image: segment 3: paddr=0x000a7af0 vaddr=0x3ffcc2a0 size=0x00e64 (  3684) load
I (249) esp_image: segment 4: paddr=0x000a895c vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at E:/esp32/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (255) esp_image: segment 5: paddr=0x000a8d68 vaddr=0x40024404 size=0x15758 ( 87896) load
I (285) esp_image: segment 6: paddr=0x000be4c8 vaddr=0x40070000 size=0x0001c (    28) load
I (296) boot: Loaded app from partition at offset 0x10000
I (297) boot: Disabling RNG early entropy source...
I (297) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (304) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
E (312) psram: PSRAM ID read error: 0x00ffff7f
E (317) cpu_start: Failed to init external RAM!
```

The other one cannot be given here because I did not save the log. But the reason for the error is probably written: calculate image hash 0xAA (0xBB) AA!=BB, but I am not using OTA related functions.

Once the above infinite restart occurs, it is usually completely disconnected from the power for a few seconds, including unplugging the battery, and then plugging in the power to start, and it will work normally.

I don’t even know where to find this kind of problem, and where it is......

chegewara, 2020-11-20:
Do you have connected something to pin IO26? Its is psram cs on wrover module (wroom is no psram). 
Do you have stable power supply, maybe USB cable is bad quality.

coollofty, 2020-11-20:
1. No, I know IO26 is used by PSRAM, so it is left floating
2. You mean there may be a problem with the power supply of the USB port? But the working current of this module is only a few hundred mA. Can the USB port still not be able to provide it?

chegewara, 2020-11-20:
USB port is fine, but sometimes quality of USB cable is not. And if its battery powered then make sure wires are not too long, i had situation when old esp32 battery powered with 30-ish cm wires caused issue to run.

coollofty, 2020-11-20:
> USB port is fine, but sometimes quality of USB cable is not. And if its battery powered then make sure wires are not too long, i had situation when old esp32 battery powered with 30-ish cm wires caused issue to run.

Thanks for your answer, I will try :)

Alvin1Zhang, 2020-12-01:
@chegewara Thanks for the notes. @coollofty has your issue resolved? Thanks.

coollofty, 2020-12-02:
> @chegewara Thanks for the notes. @coollofty has your issue resolved? Thanks.

Not sure, more testing .....

Alvin1Zhang, 2021-02-07:
@coollofty Thanks for reporting, would you please help share if any updates for this issue? Thanks.

coollofty, 2021-02-19:
After a period of testing, I can basically confirm that the problem is caused by the power supply:)

closed 2021-02-19 by coollofty