espressif/esp-idf #6142: mbedtls_ssl_setup result in an error.

nopnop2002, 2020-11-21:
## Environment

	freeRTOS version:V10.2.1
	NEWLIB version:3.0.0
	lwIP version:2-1-3-0
	ESP-IDF version:v4.3-dev-1901-g178b122c1-dirty

## Problem Description
mbedtls_ssl_setup result in an error.

```
I (143932) smtp_example: Seeding the random number generator
I (143932) smtp_example: Loading the CA root certificate...
I (143932) smtp_example: Setting hostname for TLS session...
I (143932) smtp_example: Setting up the SSL/TLS structure...
E (143942) smtp_example: mbedtls_ssl_setup returned -0x7f00
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c46c  PS      : 0x00060930  A0      : 0x800d6a6a  A1      : 0x3ffc78f0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000114d  A5      : 0x00000000
A6      : 0x3fff9d01  A7      : 0x00000114  A8      : 0x800d5758  A9      : 0x3ffc78c0
A10     : 0x3ffc7c9c  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffc7790
A14     : 0x00000047  A15     : 0x3ffc7790  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000113
```

### Steps to reproduce

1. step1

Modify [this](https://github.com/espressif/esp-idf/blob/master/examples/protocols/smtp_client/main/smtp_client_example_main.c#L252) file and send the email repeatedly.

```
static void smtp_client_task(void *pvParameters)
{
    char *buf = NULL;
    unsigned char base64_buffer[128];
    int ret, len;
    size_t base64_len;

    mbedtls_entropy_context entropy;
    mbedtls_ctr_drbg_context ctr_drbg;
    mbedtls_ssl_context ssl;
    mbedtls_x509_crt cacert;
    mbedtls_ssl_config conf;
    mbedtls_net_context server_fd;

  for(int loop=0;loop<20;loop++) { // ------------------> Added
    mbedtls_ssl_init(&ssl);
    mbedtls_x509_crt_init(&cacert);
    mbedtls_ctr_drbg_init(&ctr_drbg);
    ESP_LOGI(TAG, "Seeding the random number generator");

    No change

    putchar('\n'); /* Just a new line */
    if (buf) {
        free(buf);
    }
    
  // ------------- Start adding ---------------------
    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now %d.\n", loop);
  }
  // ------------- End adding ---------------------
    vTaskDelete(NULL);
}

```

2. step2
Build and run.

Alvin1Zhang, 2020-11-23:
Thanks or reporting, we will look into.

vikramdattu, 2020-11-30:
Hi @nopnop2002 

You should loop from [this](https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/examples/protocols/smtp_client/main/smtp_client_example_main.c#L324) line instead.

Some of the code you put in the loop is initialization code.

Please find modified example for your reference: [here.](https://github.com/espressif/tensorflow/blob/bde4f48eaa211ab94a12aa16bb6c67d6808b08d9/tensorflow/lite/micro/examples/doorbell_camera/esp/components/smtp_client/smtp_client.c#L535)

This modifies send_email function in a way that, there are checks around initialization stuff.

nopnop2002, 2020-12-01:
@vikramdattu thank you.

The problem resolved.

closed 2020-12-01 by nopnop2002