espressif/esp-idf #6143: Memory leak

cas1no, 2020-11-21:
https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/components/esp_https_server/src/https_server.c#L162
Memory is allocated, but never freed when error occurs.

Alvin1Zhang, 2020-11-23:
@cas1no Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

negativekelvin, 2020-11-23:
@Alvin1Zhang that is not necessary since the problematic code is linked

Alvin1Zhang, 2020-11-25:
@negativekelvin Thanks for the notes. @cas1no Thanks for reporting, we have a fix under internal reviewing, and the issue will be closed automatically once the issue is available on GitHub. Thanks.

closed 2020-12-13 by espressif-bot