espressif/esp-idf #6144: Trying to incorperate my own sensor script into the BLE Mesh sensor server example. Can't build from my project folder OR the idf example folder.

mrSilkie, 2020-11-21:
I have IDF v4.1 installed

```
# get_idf
# cd ~/esp/esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_sensor_model/sensor_server
# idf.py build
```

results in many familiar lines...

```
>gdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.1-dirty\" -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj   -c ../main/main.c
../main/main.c:23:10: fatal error: ble_mesh_example_init.h: No such file or directory
 #include "ble_mesh_example_init.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[1168/1231] Building C object esp-idf/...blake2b/ref/blake2b-compress-ref.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
user@dev:~/esp/esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_sensor_model/sensor_server$ 
```

Any help will be much appreciated!:+1: 

I am also trying to incorporate the mesh example into a folder in my own directory so any pointers to what changes i need to make to my CMakeLists is appreciated. I have already tried the following

```
>idf_component_register(SRCS "smartbucket.c"
                    INCLUDE_DIRS "~/esp/esp-idf/components/bt/esp_ble_mesh/api/models/include"
                    INCLUDE_DIRS "~/esp/esp-idf/components/bt/esp_ble_mesh/api/")
```

with no success

chegewara, 2020-11-21:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/common_components/example_init

Campou, 2020-11-23:
Hi @mrSilkie 

The `ble_mesh_example_init.h` is included in the [common_components](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/common_components). When trying to include the header file, the extra component needs to be added into the `CMakeLists.txt` or `Makefile`.

Like this: https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/examples/bluetooth/esp_ble_mesh/ble_mesh_sensor_model/sensor_server/CMakeLists.txt#L5

@chegewara  Thanks for the update.

Thanks.

mrSilkie, 2020-11-25:
> Hi @mrSilkie
>
> The `ble_mesh_example_init.h` is included in the [common_components](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/common_components). When trying to [...]

Great reply. I'm still a bit stuck. My CMakeLists.txt looks like;

```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/bluetooth/esp_ble_mesh/common_components/example_init
    EXTRA_COMPONENT_DIRS ~/esp/esp-idf-lib/components/
    EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/components/bt/esp_ble_mesh/api
    EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/bluetooth/esp_ble_mesh/common_components/example_init)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(smartbucket)
```

Which fixed the ble_mesh_example_init.h not found error but the following line produces a second BLE related error. 

```
../main/smartbucket.c:27:10: fatal error: esp_ble_mesh_defs.h: No such file or directory
 #include "esp_ble_mesh_defs.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[856/992] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_generichash/blake2b/ref/blake2b-compress-ref.c.obj
ninja: build stopped: subcomma
```

I really appreciate your first reply but I feel like I'm still a bit useless here. How am i meant to know to include /common_components/example_init to resolve **ble_mesh_example_init.h**?

I am asking as I now want to resolve **esp_ble_mesh_defs.h** and have already included  `EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/components/bt/esp_ble_mesh/api`

InfiniteYuan, 2021-09-26:
Please list the structure of project.

And, try to add private requirements components. [Reference](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-requirements).