espressif/esp-idf #6145: Blufi error on release v4.1 with git pull --rec update

filzek, 2020-11-22:
Blufi is completed break in latest release v4.1 with updates pulled in git pull --rec.

Its simple hangs and doesnt reply to any blufi command.

Very eratic error in a stable release version.

Urgent feedback needed.

xiewenxiang, 2020-11-23:
Could you describe in detail what kind of problems you have encountered.

filzek, 2020-11-23:
same as latest report, some parts update in ble but the error on past ble problem on packet as of IDFGH-3893, so, the problem still happen.

I will close this as it is redundant.

filzek, 2020-11-23:
as reported in sep 24, 

@hb-kaertech @filzek @jimwhitfield

If it is urgent, can you make such modification temporarily? We are doing further analysis.

components/bt/host/bluedroid/btc/profile/esp/blufi/blufi_prf.c

esp_ble_get_cur_sendable_packets_num(blufi_env.conn_id) -> esp_ble_get_sendable_packets_num()

this fix temporary the problem.

closed 2020-11-23 by filzek