espressif/esp-idf #6146: esp_http_client_read causes buffer overread

istokm, 2020-11-22:
## Environment

- IDF version:
    v4.3-dev-1721-g9bcf8ad19
- Build System:
    idf.py
- Compiler version:
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

## Problem Description

Context: This is for OTA. I'm calling an endpoint that returns a version string (that looks like: "0.1.3"), after which I wanna call another endpoint which is supposed to return the OTA file - so instead of `_perform` I call `_open`, while reusing the same `esp_http_client` handle (so just calling `esp_http_client_set_url` before the `_open` call) - but I kept getting `VALIDATE` errors that magically went away after I removed the `_perform` call.

Calling `_read` after a `_perform` call, even if separated by `esp_http_client_close` or `esp_http_client_flush_response` returns a length that is too big (by exactly the same number of bytes, as the data length I received from the `_perform` call) but without appending or prefixing the previous data. This essentially leads to a overread in my code, because even though the content length is, for example 1245, the `_read` call returns 1252 (as the `_perform` call received data with the content length of 7).

It's really hard to explain, but I hope I did a good enough job. If need, then on request I could create a MWE code that would use Pipedream to simulate our internal servers, but this would possibly take me hours so I hope my explanation is good enough :) 
I also cannot make sense of the http_client code enough to write a proper fix...

## Debug Logs

Short, but I hope descriptive, internally, the `_read` ("web_services: [read()]") function correctly determines the content length, but for some reason, returns a larger number ("rx length"), by exactly the number of bytes that was received in the previous `_perform` call

The code used in this demonstration is as follows:
`esp_http_client_perform`
`esp_http_client_close`
`esp_http_client_open`
`esp_http_client_read`
`esp_http_client_read`

The URL in this demonstration is the same across all requests.
```
D (3227) HTTP_CLIENT: set post file length = 30
D (3227) HTTP_CLIENT: Begin connect to: <redacted>
D (3257) TRANS_TCP: [sock=54] Connecting to server. IP: <redacted>, Port: <redacted>
D (3267) HTTP_CLIENT: Write header[5]: POST /api/sensors/app-version?api-version=2.0 HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: <redacted>
Client: <redacted>
Content-Type: application/json
Content-Length: 30

D (3327) HTTP_CLIENT: on_message_begin
D (3337) HTTP_CLIENT: HEADER=Cache-Control:no-cache
D (3337) HTTP_CLIENT: HEADER=Pragma:no-cache
D (3347) HTTP_CLIENT: HEADER=Content-Type:application/json; charset=utf-8
D (3367) HTTP_CLIENT: HEADER=Expires:-1
D (3377) HTTP_CLIENT: HEADER=X-Powered-By:ASP.NET
D (3397) HTTP_CLIENT: HEADER=Content-Length:7
D (3407) HTTP_CLIENT: http_on_headers_complete, status=200, offset=201, nread=201
D (3417) HTTP_CLIENT: http_on_body 7
V (3417) web_services: [eventHandler]: Rx: data {
        data: "0.1.4"
        length: 7
}
D (3427) HTTP_CLIENT: http_on_message_complete, parser=3ffca34c
D (3437) HTTP_CLIENT: content_length = 7
D (3437) web_services: [eventHandler]: HTTP session ended
V (3447) web_services: [eventHandler]: Status code: 200
V (3447) web_services: [close()]
D (3447) web_services: [eventHandler]: Disconnected
D (3577) HTTP_CLIENT: Begin connect to: <redacted>
D (3587) TRANS_TCP: [sock=54] Connecting to server. IP: <redacted>, Port: <redacted>
D (3597) HTTP_CLIENT: Write header[5]: POST /api/sensors/app-version?api-version=2.0 HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: <redacted>
Client: <redacted>
Content-Type: application/json
Content-Length: 30

D (7117) HTTP_CLIENT: on_message_begin
D (7117) HTTP_CLIENT: HEADER=Cache-Control:no-cache
D (7117) HTTP_CLIENT: HEADER=Pragma:no-cache
D (7137) HTTP_CLIENT: HEADER=Content-Type:application/json; charset=utf-8
D (7147) HTTP_CLIENT: HEADER=Expires:-1
D (7157) HTTP_CLIENT: HEADER=X-Powered-By:ASP.NET
D (7187) HTTP_CLIENT: HEADER=Content-Length:7
D (7197) HTTP_CLIENT: http_on_headers_complete, status=200, offset=201, nread=201
D (7197) HTTP_CLIENT: http_on_body 7
V (7207) web_services: [eventHandler]: Rx: data {
        data: "0.1.4"
        length: 7
}
D (7217) HTTP_CLIENT: http_on_message_complete, parser=3ffca34c
D (7217) HTTP_CLIENT: content_length = 7
V (7227) web_services: [open]: Status code: 200
V (7227) web_services: [read()]
D (7227) HTTP_CLIENT: is_data_remain=0, is_chunked=0, content_length=7
D (7237) web_services: [read]: rx length: 14
V (7237) web_services: [read]: {
        data: "0.1.4"�
                      ��-_�
        length: 14
}
V (7247) web_services: [read()]
D (7257) HTTP_CLIENT: is_data_remain=0, is_chunked=0, content_length=7
D (7257) web_services: [read]: rx length: 0
I (7267) web_services: [read]: Data received
V (7267) web_services: [read]: {
        data:
        length: 0
}
```

Alvin1Zhang, 2020-11-23:
Thanks for reporting and the detailed issue summary, we will look into.

shubhamkulkarni97, 2020-11-27:
Hi @istokm,

From the description you mean performing first request using esp_http_client_perform and then with esp_http_client_read, content length for both the requests add up.

I tried to reproduce this issue but content length looks correct at my end.

Can you share a code snippet to reproduce this issue?

istokm, 2020-11-30:
You need to close and reopen the connection, then read. Just reading after a perform works as expected. I'll try to share the code asap, but I'm using a very large web service library I wrote over last few months, so I'll have to write that from scratch.

istokm, 2020-12-03:
I've been taking for granted that one calls `esp_http_client_fetch_headers` after `open` - this is what actually causes the mismatch between the returned length and buffer contents - otherwise the content lengths still add up, but the buffer does so too (which I think is still unintended, especially with `close` & `flush_response` calls in between the requests).
I've made a self-contained MWE example of this issue, feel free to use the pipedream URL as well - I'll disable it when this issue closes.
https://github.com/istokm/esp-idf-http-buffer-overread

Only thing you need to change is the `EXAMPLE_ESP_WIFI_SSID` & `EXAMPLE_ESP_WIFI_PASS` to connect to your wifi
All other details about the behaviour of various function calls can be found in the code itself
https://github.com/istokm/esp-idf-http-buffer-overread/blob/3165359ac5f3694372e26dd56b155db2a9e0877d/main/main.cpp#L165-L178

Here is the output I receive when I run this code on my ESPs with the latest ESP-IDF version: https://gist.github.com/istokm/26afb735a23281059ff98cd94bd2085b

Let me know if there is anything else you need to replicate this, I don't have much free time to work on this, so I'm trying to spend it well :)

Note: as the fix for #6163 hasn't been merged into master as of writing this, you'll have to apply the `tls.diff` patch that I've included in the repo itself - otherwise the esp_http_close calls will always return -1 with `https` URLs.

shubhamkulkarni97, 2020-12-09:
Hi @istokm, Thanks for providing the code, I could reproduce this issue.
We are working on the fix and will keep you posted.

closed 2021-01-06 by espressif-bot