espressif/esp-idf #6147: mDNS does not return a response after 10 minutes.

nopnop2002, 2020-11-23:
# Environment

	freeRTOS version:V10.2.1
	NEWLIB version:3.0.0
	lwIP version:2-1-3-0
	ESP-IDF version:v4.3-dev-1901-g178b122c1-dirty

# Problem Description

mDNS does not return a response after 10 minutes.

## Steps to reproduce

1. Step1
build [this](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mdns) example and run.   
mdns hostname set to: [esp32-mdns-05C634]   

	I (0) cpu_start: Starting scheduler on APP CPU.
	I (605) system_api: Base MAC address is not set
	I (605) system_api: read default base MAC address from EFUSE
	I (615) mdns-test: mdns hostname set to: [esp32-mdns-05C634]
	I (615) mdns-test: EXAMPLE_MDNS_INSTANCE=ESP32 with mDNS

2.Step2
Run the ping command in the terminal.

	$ ping esp32-mdns-05C634.local
	PING esp32-mdns-05C634.local (192.168.10.132) 56(84) bytes of data.
	64 bytes from 192.168.10.132 (192.168.10.132): icmp_seq=1 ttl=255 time=40.7 ms
	64 bytes from 192.168.10.132 (192.168.10.132): icmp_seq=2 ttl=255 time=55.3 ms
	64 bytes from 192.168.10.132 (192.168.10.132): icmp_seq=3 ttl=255 time=77.8 ms
	^C
	--- esp32-mdns-05C634.local ping statistics ---
	3 packets transmitted, 3 received, 0% packet loss, time 2002ms
	rtt min/avg/max/mdev = 40.747/57.984/77.807/15.240 ms

3.Step3
Stop ping.

4.Step4
After 10 minutes, execute the ping command in the terminal.

	$ ping esp32-mdns-05C634.local
	ping: esp32-mdns-05C634.local: Name or service not known

Alvin1Zhang, 2020-11-23:
Thanks for reporting, we will look into.

david-cermak, 2020-11-24:
@nopnop2002  I can reproduce this issue under Ubuntu 18.04, but still think this is related to hostname resolution in `avahi` (since it works correctly on Mac which is assumed to be something like a mDNS standard). The esp32 mdns responder replies correctly, can see the answers in the packet capture. It seems like the ping to local domain relies on `avahi-resolve` which in turn doesn't correctly accept answers. Interestingly the `avahi-browse` does resolve the names correctly...

Could you please try to run:
```
avahi-browse -a -v -r
```

and post the output here and check if you can ping the hostname after that?

nopnop2002, 2020-11-24:
@david-cermak 
I tried.
There is [esp32-mdns-05C634.local].

	$ avahi-browse -a -v -r
	Server version: avahi 0.7; Host name: Compaq-Pro.local
	E Ifce Prot Name                                          Type                 Domain
	+   eno1 IPv4 landisk-HDL-AA1 [34:76:c5:71:e0:6d]           Workstation          local
	+   eno1 IPv4 landisk-HDL-AA1(smb)                          Microsoft Windows Network local
	+   eno1 IPv4 landisk-HDL-AA1                               Web Site             local
	+   eno1 IPv4 ESP32-WebServer                               Web Site             local
	=   eno1 IPv4 landisk-HDL-AA1 [34:76:c5:71:e0:6d]           Workstation          local
	   hostname = [landisk-HDL-AA1.local]
	   address = [192.168.10.112]
	   port = [9]
	   txt = []
	=   eno1 IPv4 landisk-HDL-AA1(smb)                          Microsoft Windows Network local
	   hostname = [landisk-HDL-AA1.local]
	   address = [192.168.10.112]
	   port = [445]
	   txt = []
	=   eno1 IPv4 landisk-HDL-AA1                               Web Site             local
	   hostname = [landisk-HDL-AA1.local]
	   address = [192.168.10.112]
	   port = [80]
	   txt = []
	=   eno1 IPv4 ESP32-WebServer                               Web Site             local
	   hostname = [esp32-mdns-05C634.local]
	   address = [192.168.10.132]
	   port = [80]
	   txt = ["board=esp32" "u=admin" "p=password" "path=/foobar"]
	: Cache exhausted
	: All for now
	^CGot SIGINT, quitting.

Is there a solution?

david-cermak, 2020-11-30:
@nopnop2002 seen some MS local networks in your listing, couldn't you experience some standard issues described e.g. here? http://www.lowlevelmanager.com/2011/09/fix-linux-dns-issues-with-local.html

> Is there a solution?

I would suggest working this around as outlined, so pinging the address resolved with `avahi`.
As said above, I can reproduce the issue and seen the queries sent by avahi, as well as, replies by ESP32, so I might eventually dig deep into it to see what's wrong, but that will take some some. Furthermore, as mentioned above, very unlikely is this issue related to the ESP32 mDNS library, since it works correctly on Mac and the mDNS component **pass** the Apple's *Bonjour Conformance Test* (which does not seem to be the case for `avahi`).

nopnop2002, 2020-12-01:
@david-cermak 

>very unlikely is this issue related to the ESP32 mDNS library.

I'm thinking about closing this issue.

nopnop2002, 2020-12-01:
>very unlikely is this issue related to the ESP32 mDNS library.

1.build [this](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mdns) example and run.
mdns hostname set to: [esp32-mdns-05C634]

	I (0) cpu_start: Starting scheduler on APP CPU.
	I (601) system_api: Base MAC address is not set
	I (601) system_api: read default base MAC address from EFUSE
	I (601) mdns-test: mdns hostname set to: [esp32-mdns-05C634]

2.Run the ping command in the terminal.

	$ ping esp32-mdns-05C634.local
	PING esp32-mdns-05C634.local (192.168.10.132) 56(84) bytes of data.
	64 bytes from 192.168.10.132 (192.168.10.132): icmp_seq=1 ttl=255 time=57.5 ms
	64 bytes from 192.168.10.132 (192.168.10.132): icmp_seq=2 ttl=255 time=70.9 ms
	64 bytes from 192.168.10.132 (192.168.10.132): icmp_seq=3 ttl=255 time=99.4 ms
	64 bytes from 192.168.10.132 (192.168.10.132): icmp_seq=4 ttl=255 time=18.5 ms
	^C
	--- esp32-mdns-05C634.local ping statistics ---
	4 packets transmitted, 4 received, 0% packet loss, time 3004ms
	rtt min/avg/max/mdev = 18.509/61.610/99.429/29.120 ms

3.I ran the following program on another ESP32.
An error will occur.

[image]

	static void query_mdns_host(const char * host_name)
	{
	        ESP_LOGI(TAG, "Query A: %s.local", host_name);

	        struct esp_ip4_addr addr;
	        addr.addr = 0;

	        esp_err_t err = mdns_query_a(host_name, 2000,  &addr);
	        if(err){
	                if(err == ESP_ERR_NOT_FOUND){
	                        ESP_LOGW(TAG, "%s: Host was not found!", esp_err_to_name(err));
	                        return;
	                }
	                ESP_LOGE(TAG, "Query Failed: %s", esp_err_to_name(err));
	                return;
	        }

	        ESP_LOGI(TAG, "Query A: %s.local resolved to: " IPSTR, host_name, IP2STR(&addr));
	}

	void app_main()
	{
	        //Initialize NVS
	        esp_err_t ret = nvs_flash_init();
	        if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
	                ESP_ERROR_CHECK(nvs_flash_erase());
	                ret = nvs_flash_init();
	        }
	        ESP_ERROR_CHECK(ret);

	        ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
	        if (wifi_init_sta() == ESP_OK) {
	                ESP_LOGI(TAG, "Connection success");
	        } else {
	                ESP_LOGE(TAG, "Connection failed");
	                while(1) { vTaskDelay(1); }
	        }
	        ESP_LOGI(TAG, "BIT2=0x%x", BIT2);
	        ESP_LOGI(TAG, "BIT4=0x%x", BIT4);

	        //initialize mDNS
	        ESP_ERROR_CHECK( mdns_init() );
	        query_mdns_host("esp32-mdns-05C634");

	}

I am very confused.

david-cermak, 2020-12-01:
@nopnop2002 suggest closing this issue and raise another related to the problems you describe in your latest post. We generally try to track a single issue in one item with good description and the title of the issue.

> I am very confused.

Please, understand that failing to ping the ESP32 name on your platform while resolving it in `avahi` does point to your platform. Whereas failing to resolve the same name with ESP32 resolver does not. (My previous note about this not likely to be a *IDF mDNS problem* is of course related to this issue titled **mDNS does not respond after 10 minutes**)

Also, before submitting a new issue, please make sure the problem is reproducible and repeatable. Note that the `mdns_query_a()` has a timeout argument (set to 2 seconds here) and the BCT for example provides greater timeouts, even `avahi` resolver has about 5 seconds default timeout. Plus, if you're running multiple ESP32-board with the same hostname, then the unique name assignment might take place mangling the hostname on some devices.

nopnop2002, 2020-12-01:
@david-cermak 

Thank you for your serious answer.

closed 2020-12-01 by nopnop2002