espressif/esp-idf #6148: How to give OTA update support in ESP-IDF mesh

MarkEvens, 2020-11-23:
**How to upgrade firmware of all connected device in WiFI-MESH network **

I am working on project which required OTA update support in mesh network. But i can't find anything useful.
Is there any way to do my job.

ESP-IDF version is 4.0.1

EspHuifeng, 2020-11-23:
We recommend you use ESP-MDF to develop the ESP-Mesh application. There's the OTA functionality that you need. Please see: https://github.com/espressif/esp-mdf

MarkEvens, 2020-11-23:
esp-mdf is not good idea for my project it is time taking process for me to transfer code to mdf and i don't have much idea about esp-idf. any thing in esp-idf 4.0.1 is great any resources or link.

MarkEvens, 2020-11-25:
I have made mesh network using esp-idf and want to give ota firmware update to any connected device.

for this i use idea that first root node call https server for get firmware and received packet send to connected node. so that this node store firmware in ota partition and after reboot new firmware start executing.

But for this idea i need way or idea to stream data in mesh network between root to node.

So there is any good way or idea for it ?????????

MarkEvens, 2020-11-27:
Anything ??

MarkEvens, 2020-12-11:
any update in V4.2 about this?

EspHuifeng, 2020-12-11:
You can refer to the mupgrade component in MDF, which is to send the firmware to other MESH nodes.1024 bytes per transmission.
https://github.com/espressif/esp-mdf/blob/1c76187faaf16d40a574fc7c0c98122e93dee746/components/mupgrade/mupgrade_root.c#L331

MarkEvens, 2020-12-12:
@EspHuifeng  I tried your suggestion and facing problem in mesh receiving i am send 1400 packets but node received only half of packet

MarkEvens, 2020-12-12:
i am using below function for sending
```
    esp_err_t err;
    mesh_data_t data;
    data.data = (uint8_t *)dataOTA;
    data.size = sizeOTA;
    data.proto = MESH_PROTO_BIN;
    data.tos = MESH_TOS_P2P;
    printf("\n routing table size : %d", s_route_table_size);
    for (int i = 0; i < s_route_table_size; i++)
    {
        err = esp_mesh_send(&s_route_table[i], &data, MESH_DATA_P2P, NULL, 0);
    }
```

MarkEvens, 2020-12-14:
Receiving half data issue is found. I am calling esp_mesh_recv in two different task so data dividing between them and it solved.

MarkEvens, 2020-12-14:
How to know at receiver side that firmware completely received ?

MarkEvens, 2020-12-15:
@EspHuifeng  as per your suggested way i am succeed in ota update in mesh network.

closed 2020-12-15 by MarkEvens

pmartinz, 2021-01-26:
@MarkEvens is it possible for you to share how you implemented it?

HughStanway, 2023-08-14:
Has anyone got any advice on implementing this?

tonivss, 2023-08-21:
@MarkEvens I'm very interested on how you implemented it. Could you share some code?

MarkEvens, 2023-08-23:
@tonivss  are you able to make communication between node and root?

tonivss, 2023-08-23:
Yes I am. I tried another way and it works fine. However I am very interested in the way you do it. Are you still using the approach you mentioned here https://github.com/espressif/esp-idf/issues/6148#issuecomment-733569989?