espressif/esp-idf #6149: Cannot mount spiffs after OTA update and rollback

ZweiEuro, 2020-11-23:
## Environment

- Development Kit:     ESP32 Devkit V1 -Doit (I think, it's soldered i can't look at it anymore) [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/modules-and-boards.html#esp32-wroom-32)
- Kit version -
- Module or chip used:  ESP32-WROOM-32

- IDF version (run ``git describe --tags`` to find it): framework-espidf 3.40001.200521 (4.0.1)  (PIO)
- Build System:         platformio
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

- Operating System:   Linux

- Using an IDE?:  Vscode platformio
- Power Supply:         USB

## Problem Description

Wrote OTA example native into my program, tried it out, succeeded. 
But then i couldn't upload locally anymore since it kept using the OTA'd version.

Parttool just straight up refuses to start (yes I have followed the instructions for installation trice and thoroughly) it keeps saying some py files are missing.
Running:
```
git clone git@github.com:espressif/esp-idf.git
cd esp-idf
git checkout release/v4.0
./install.sh
. ./export.sh

```
following everything from otatool at
https://github.com/espressif/esp-idf/tree/master/examples/system/ota/otatool

```
cd components/app_update/
./otatool.py
```
Result:
Traceback (most recent call last):
  File "./otatool.py", line 29, in <module>
    from parttool import PartitionName, PartitionType, ParttoolTarget, PARTITION_TABLE_OFFSET
ModuleNotFoundError: No module named 'parttool'

Which of course causes https://github.com/espressif/esp-idf/tree/master/examples/system/ota/otatool to fail as well.
With the same error.

So now i am stuck with a chip i can't change since the version on the ota doesn't have the ota yet.

So i tried using esptool to reset the whole thing with erase_flash
and then re-> uploading with a partition table and the firmware.

Now it boots from the new firmware but produces this for spiffs:
      esp_err_t ret = esp_vfs_spiffs_register(&spiffs_conf);
returns ESP_FAIL

and now i am stuck with a dead chip.

did OTA overwrite something ciritcal?

igrr, 2020-11-24:
Hi @ZweiEuro, after re-uploading the partition table and new application, have you also uploaded the SPIFFS partition binary? Or is your intention to start from an empty SPIFFS partition?

ZweiEuro, 2020-11-24:
@igrr  I wanted to start with an all new spiffs since erase_flash just erases everything. 
My fault though was in
esp_vgs_spiffs_conf_t   .format_if_mount_failed 
was false, after changing it to true it works again. 
(Sorry that issue must've sounded serious )

I am still kinda bummed that I can't use the otatool to free the otadata but still, I now have a way to reset the whole chip if something fails

igrr, 2020-11-24:
@ZweiEuro thanks for confirming that the SPIFFS part works. I've reopened the issue so that we can track fixing the otatool failure.

dobairoland, 2020-11-24:
Hi @ZweiEuro. 

```
cd components/app_update/
./otatool.py
```
At this point, do you have `$IDF_PATH` set? (i.e. `echo $IDF_PATH` shows a correct path)

Does the following command work for you (and points to the right IDF_PATH)?
`python -c 'import os; print(os.path.expandvars(os.path.join("$IDF_PATH", "components")))'`

ZweiEuro, 2020-11-24:
@dobairoland 
yes, IDF Path shows correctly where i have installed the esp-idf ( in my case `/home/zweieuro/Documents/tools/esp-idf`)

`python -c 'import os; print(os.path.expandvars(os.path.join("$IDF_PATH", "components")))'`
returns
`/home/zweieuro/Documents/tools/esp-idf/components`

dobairoland, 2020-11-24:
@ZweiEuro The esptool issue is probably caused by not initializing the submodules. `git submodule update --init --recursive` should help.

ZweiEuro, 2020-11-26:
@dobairoland after running your command and retrying 
`./otatool.py -q --port /dev/ttyUSB0 erase_otadata`
I still get:
Traceback (most recent call last):
  File "./otatool.py", line 29, in <module>
    from parttool import PartitionName, PartitionType, ParttoolTarget, PARTITION_TABLE_OFFSET
ModuleNotFoundError: No module named 'parttool'

During handling of the above exception, another exception occurred:

```
Traceback (most recent call last):
  File "./otatool.py", line 35, in <module>
    from parttool import PartitionName, PartitionType, ParttoolTarget, PARTITION_TABLE_OFFSET
ModuleNotFoundError: No module named 'parttool'
```

dobairoland, 2020-11-27:
@ZweiEuro I'm sorry that I cannot give you a straight solution but this is new to me as well.

Can I ask you to try to run `idf.py python-clean` and see if it helps?

I can run your command successfully for IDFv4.0 you are using.

ZweiEuro, 2020-12-23:
I no longer have access to the system that produced this error; Sorry but I can't debug this further (something else completley broke)
Should I leave this issue open or close it ?
@dobairoland

dobairoland, 2020-12-23:
Thank you @ZweiEuro for the feedback. Yes, I'm closing this issue because I cannot reproduce it. Feel free to create a new one if you encounter this again.

closed 2020-12-23 by dobairoland