espressif/esp-idf #6155: PSRAM init on master causes FreeRTS queue assert; works in release/4.0

lastquestion, 2020-11-25:
### Environment

- Development Kit: ESP32-CAM AI-thinker clone
- Module or chip used: ESP32
- IDF version (run ``git describe --tags`` to find it): master, commit 178b122c145c19e94ac896197a3a4a9d379cd618
- Build System: idf.py
- Power Supply: 5V external

## Problem Description

I checked out master 178b122c145c19e94ac896197a3a4a9d379cd618 yesterday, and copied `wifi/station` example and then attempted to init PSRAM. Failed during boot, causing bootloop, turned on verbose logging, got this:

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6912
ho 0 tail 12 room 4
load:0x40078000,len:14332
load:0x40080400,len:3672
entry 0x40080678
I (30) boot: ESP-IDF v4.3-dev-1901-g178b122c1 2nd stage bootloader
I (31) boot: compile time 19:49:58
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1ac30 (109616) map
I (157) esp_image: segment 1: paddr=0x0002ac58 vaddr=0x3ffb0000 size=0x04034 ( 16436) load
I (164) esp_image: segment 2: paddr=0x0002ec94 vaddr=0x40080000 size=0x00404 (  1028) load
I (165) esp_image: segment 3: paddr=0x0002f0a0 vaddr=0x40080404 size=0x00f78 (  3960) load
I (174) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x741e4 (475620) map
I (362) esp_image: segment 5: paddr=0x000a420c vaddr=0x4008137c size=0x159e4 ( 88548) load
I (414) boot: Loaded app from partition at offset 0x10000
I (414) boot: Disabling RNG early entropy source...
D (426) efuse: coding scheme 0
/Users/lin/esp-idf/components/freertos/queue.c:1441 (xQueueSemaphoreTake)- assert failed!
Re-enable cpu cache.

abort() was called at PC 0x4008d6bf on core 0
Backtrace:0x4008998f:0x3ffe39c0 0x4008a185:0x3ffe39e0 0x40090ffa:0x3ffe3a00 0x4008d6bf:0x3ffe3a70 0x4008f9be:0x3ffe3ab0 0x401440a9:0x3ffe3ad0 0x4008f955:0x3ffe3b00 0x400d812d:0x3ffe3b50 0x400d8029:0x3ffe3bb0 0x400d7fe1:0x3ffe3bd0 0x40083930:0x3ffe3c00 0x400d50e9:0x3ffe3c30 0x4008149a:0x3ffe3c50 0x400791d0:0x3ffe3c80 |<-CORRUPTED

ELF file SHA256: 4e859f056b97d68e

Rebooting...
```

I played around with many options assuming it was some hardware quirk; the board i have is a ESP32-CAM AI-THINKER clone board. 

Nothing worked, in frustration tested arduino IDE, it worked; now arduino esp uses something like 3.3 release, so I decided to revert back to older builds. I went to release/4.0 branch, now it works. I didn't test the 4.x branches.

LMK if you want more information.

lastquestion, 2020-11-25:
Note, with PSRAM init disabled in `menuconfig`, it will boot past this and work. In 4.0, with PSRAM enabled, I get 4MB printed out, so the board actually does have 4MB and it actually does work.

mahavirj, 2021-03-23:
@lastquestion 

Sorry for delayed reply.

This looks to be similar to https://github.com/espressif/esp-idf/issues/6660. We will close once fix appears on github

mahavirj, 2021-04-05:
Fixed with https://github.com/espressif/esp-idf/commit/c05fde2cbfe3e45febf07d01f9daf238e74d664d

closed 2021-04-05 by mahavirj