espressif/esp-idf #6156: gatt_server_service_table character not find

xukaijie111, 2020-11-25:
I use gatt_server_serivce_table example. the phone can find the Ble device and connect it successfully.but the character number is error.
  the code define IDX_CHAR_A IDX_CHAR_B IDX_CHAR_C ,total 3 characters.but the phone only see the IDX_CHAR_A character.
  i not edit any code about the gatt_server_serivce_table example

xiewenxiang, 2020-11-25:
I have test the problem you mentioned,  all the three characteristics can be discovery.

Alvin1Zhang, 2020-11-25:
@xukaijie111 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? More information would help us debug further. Thanks.

Alvin1Zhang, 2020-12-04:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates or details. Thanks.

closed 2020-12-04 by Alvin1Zhang