espressif/esp-idf #6157: esp_http_client truncates reponse headers

Clickau, 2020-11-25:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    v4.0.2
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    8.4.0
- Operating System:     Linux
- Using an IDE?: VSCode ESP-IDF extension
- Power Supply:         USB

## Problem Description

I encountered this issue when trying to download an asset from a github release. It redirects to a long AWS URL, which gets truncated when reading the location header. I made a python server to test this, and when the server responds with a long header, it gets cut down to around 422 characters. Interestingly, if I add another header "test: test" before the long one, the response is shortened to 410 characters. But if I add the test header after the long one, its length remains 422. When I add up the length of all the headers before and including the long one, I get something close to 512, so what I'm thinking is happening is that the client reads the response in 512 bytes chunks, and when the last header does not fit inside, for whatever reason, it doesn't get completed on the next pass.

### Expected Behavior 

The headers are fully read, or if there is a hard-coded limit, at least it should be configurable via menuconfig.

### Actual Behavior

The header is truncated without any explanation

### Code to reproduce this issue

```c
    esp_http_client_config_t config = {
        .url = "http://192.168.0.100",
        .event_handler = _http_event_handler,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    esp_err_t err = esp_http_client_perform(client);

    if (err == ESP_OK) {
        ESP_LOGI(TAG, "HTTPS Status = %d, content_length = %d",
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
    } else {
        ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(client);

```

Python server:
```python
from http.server import HTTPServer, BaseHTTPRequestHandler
import ssl

string = '01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789'

class Handler(BaseHTTPRequestHandler):
    def do_GET(self):
        self.send_response(200)
        # self.send_header('test', 'test')
        self.send_header('location', string)
        # self.send_header('test2', 'test2')
        self.end_headers()

httpd = HTTPServer(('0.0.0.0', 80), Handler)

httpd.serve_forever()
```

## Debug Logs

```
I (3690) HTTP_CLIENT: HTTP_EVENT_ON_CONNECTED
I (3700) HTTP_CLIENT: HTTP_EVENT_HEADER_SENT
I (3710) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Server, value=BaseHTTP/0.6 Python/3.8.5
I (3710) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Date, value=Wed, 25 Nov 2020 07:01:38 GMT
I (3710) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=test, value=test
I (3720) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=location, value=01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890
I (3760) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=test2, value=test2
I (3770) HTTP_CLIENT: HTTP_EVENT_ON_FINISH
I (3770) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (3780) HTTP_CLIENT: HTTPS Status = 200, content_length = -1
I (3780) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
```

Alvin1Zhang, 2020-11-25:
Thanks for reporting, we will look into.

chegewara, 2020-11-25:
In menuconfig you have setting how long headers can be. By default it is 512 IIRC.

Clickau, 2020-11-25:
The only setting I can find in menuconfig is for the HTTP server, not for the client. They're right under the client settings, so I also thought they are for the client initially, but they don't help. Correct me if I'm wrong.

pdemianczuk07, 2020-11-25:
You can set in config structure 
 esp_http_client_config_t config = {
        .url = "http://192.168.0.100",
        .event_handler = _http_event_handler,
         .buffer_size = XXZ,   //size in Bytes
         .buffer_size_rx = YYZ,
    };

buffer is set by default on  #define DEFAULT_HTTP_BUF_SIZE (512)

Clickau, 2020-11-25:
Setting buffer_size and buffer_size_tx to 2048 in config works fine. But I think this is still a pretty big issue. Even if you think that having long headers is an uncommon scenario, this issue also occurs when there are a lot of headers of normal size.
```
I (3966) HTTP_CLIENT: HTTP_EVENT_ON_CONNECTED
I (3976) HTTP_CLIENT: HTTP_EVENT_HEADER_SENT
I (3986) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Server, value=BaseHTTP/0.6 Python/3.8.5
I (3986) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Date, value=Wed, 25 Nov 2020 11:47:57 GMT
I (3996) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=aa, value=0123456789012345678901234567890123456789
I (3996) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=bb, value=0123456789012345678901234567890123456789
I (4006) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=cc, value=0123456789012345678901234567890123456789
I (4016) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=dd, value=0123456789012345678901234567890123456789
I (4026) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=ee, value=0123456789012345678901234567890123456789
I (4036) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=ff, value=0123456789012345678901234567890123456789
I (4046) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=gg, value=0123456789012345678901234567890123456789
I (4056) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=hh, value=0123456789012345678901234567890123456789
I (4066) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=ii, value=0123456789012345678901234567890123456789
I (4076) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=jj, value=01234
I (4086) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=kk, value=0123456789012345678901234567890123456789
I (4096) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=ll, value=0123456789012345678901234567890123456789
I (4106) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=mm, value=0123456789012345678901234567890123456789
I (4116) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=nn, value=0123456789012345678901234567890123456789
I (4126) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=oo, value=0123456789012345678901234567890123456789
I (4136) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=pp, value=0123456789012345678901234567890123456789
I (4146) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=qq, value=0123456789012345678901234567890123456789
I (4156) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=rr, value=0123456789012345678901234567890123456789
I (4166) HTTP_CLIENT: HTTP_EVENT_ON_FINISH
I (4166) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
```
The value of header jj got truncated without any warning. If you consider that making this work properly, i.e. fixing incomplete headers on the next read, is not worth the trouble, the client should at least ignore all headers that do not fit into the first buffer and display a warning, to make debugging this easier. Also, this limitation should be documented properly, as I did not find any reference of it in the docs.
Thank you for the help.

pdemianczuk07, 2020-11-25:
Hi, oh..that is according to headers not payload,

I found the struct about connection type which one field is max_store_header_size
typedef struct {
    char                         *url;
    char                         *scheme;
    char                         *host;
    int                          port;
    char                         *username;
    char                         *password;
    char                         *path;
    char                         *query;
    char                         *cert_pem;
    esp_http_client_method_t     method;
    esp_http_client_auth_type_t  auth_type;
    esp_http_client_transport_t  transport_type;
    int                          max_store_header_size;
} connection_info_t;
 
becasue of obj client is pointer to obj which struct is (and one of field is connection_info)
struct esp_http_client {
    int                         redirect_counter;
    int                         max_redirection_count;
    int                         max_authorization_retries;
    int                         process_again;
    struct http_parser          *parser;
    struct http_parser_settings *parser_settings;
    esp_transport_list_handle_t     transport_list;
    esp_transport_handle_t          transport;
    esp_http_data_t                 *request;
    esp_http_data_t                 *response;
    void                        *user_data;
    esp_http_auth_data_t        *auth_data;
    char                        *post_data;
    char                        *location;
    char                        *auth_header;
    char                        *current_header_key;
    char                        *current_header_value;
    int                         post_len;
    connection_info_t           connection_info;
    bool                        is_chunk_complete;
    esp_http_state_t            state;
    http_event_handle_cb        event_handler;
    int                         timeout_ms;
    int                         buffer_size_rx;
    int                         buffer_size_tx;
    bool                        disable_auto_redirect;
    esp_http_client_event_t     event;
    int                         data_written_index;
    int                         data_write_left;
    bool                        first_line_prepared;
    int                         header_index;
    bool                        is_async;
};

so, probablly you should test change one field 
like client->connection_info.max_store_headr_size = 1024; //or check >512B

before  line : esp_err_t err = esp_http_client_perform(client);

Clickau, 2020-11-25:
I can't change that value inside my app because the struct esp_http_client is defined in a .c file and I can't access its members outside of that file. But I tried logging the value of client->connection_info.max_store_header_size inside esp_http_client_perform and it's 0. Setting it to a higher value doesn't help.

shubhamkulkarni97, 2020-11-26:
Hi @Clickau,

Setting `buffer_size` in esp_http_client_config_t should resolve this issue.
It is a bit difficult to display a warning message in this case, because [esp_transport_read](https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/components/esp_http_client/esp_http_client.c#L1053) only reads data according to specified buffer length. There is no way to get length of actual header.

Still we will check if we can add workaround to make debugging easier.

shubhamkulkarni97, 2021-01-05:
@Clickau, I didn't find out any way to display warning message in this scenario.

Is your issue resolved?

yuanjianmin, 2021-02-18:
hi @Clickau, As you mentioned above, ‘The only problem is with the Location header, which crashes the ESP if I use http_utils_append_string instead of http_utils_assign_string’, does this problem still exist in your[ PR](https://github.com/espressif/esp-idf/pull/6370)? Or how to reproduce, I did not reproduce the problem with your PR locally.

Clickau, 2021-02-18:
If I remember correctly, it was just a coincidence that the crash happened after the Location header. The problem was that the name of the header itself could get split (for example into "Loc" and "ation") and the program would try to log the header value, which was null. In the PR I use http_utils_append_string to avoid splitting for the header name as well, not just the value, and I've added another null check, so it shouldn't be a problem anymore.

yuanjianmin, 2021-02-19:
Got it. Thanks for your contribution @Clickau .

AxelLin, 2021-07-14:
@ESP-YJM 
This issue was reported on v4.0.2. Please backport the fix to required branches.

Alvin1Zhang, 2021-08-11:
Thanks for reporting.

- Fix on release/4.3 is https://github.com/espressif/esp-idf/commit/8f75f93c14095e43b37a4e0075176340df4a8c02;
- Fix on release/4.2 is https://github.com/espressif/esp-idf/commit/88b3e0d8d8af4cea5edf7c365f96a8fe9b5436b0;
- Fix on release/4.0 is https://github.com/espressif/esp-idf/commit/40cf3c5e9977ba3eba784223d00917d73a5e7510;
- Fix on release/3.3 is https://github.com/espressif/esp-idf/commit/8f0bcc570a4a2b1447b04a9e75992387e990fb76.

Feel free to reopen.

closed 2021-08-11 by Alvin1Zhang