espressif/esp-idf #6158: OTA on ESP32-IDF from AWS S3 bucket

ManjuMandya, 2020-11-25:
Hi,
  How do i do OTA on ESP32-IDF from AWS S3 bucket?
I tried simple_ota_example and modified the firmware URL in configuration as below

 esp_http_client_config_t config = {
       // .url = CONFIG_EXAMPLE_FIRMWARE_UPGRADE_URL,
        .url = "https://my-url-link-ap-south-1.amazonaws.com/firmware.bin" ,
        .cert_pem = (char *)server_cert_pem_start,
        .event_handler = _http_event_handler,
    };

but i get error as below:

E (5044) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (5044) esp-tls: Failed to verify peer certificate!
I (5054) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA

E (5064) esp-tls: Failed to open new connection
E (5064) TRANS_SSL: Failed to open a new connection
E (5074) HTTP_CLIENT: Connection failed, sock < 0
E (5074) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (5084) esp_https_ota: Failed to establish HTTP connection
  
  How to go about in solving the issue
  
Thanks

mahavirj, 2020-11-26:
@ManjuMandya 

Please see pointer below that should help:

https://github.com/espressif/esp-idf/issues/3021#issuecomment-460917118

Also for `how-to` questions please prefer using esp32.com forum

ManjuMandya, 2020-11-26:
@mahavirj  Thanks for the link. I am able to connect to AWS after creating certificate but iam getting access denied response for AWS. It looks like i have to send access token, which iam not sure how to do it from ESP32-IDF.
Let me know if you have any idea.

mahavirj, 2020-11-27:
@ManjuMandya 

You may use pre-signed URL method as mentioned in my earlier linked comment, see [here](https://docs.aws.amazon.com/AmazonCloudFront/latest/DeveloperGuide/private-content-signed-urls.html) for more details.

mahavirj, 2020-11-30:
@ManjuMandya Closing as there are no followup questions here. However, if you are still facing any issues from ESP-IDF side please feel free to re-open.

closed 2020-11-30 by mahavirj

ManjuMandya, 2020-11-30:
Thank you Mahavir. Our cloud team had not given read access, now we are able to download firmware from S3 bucket.