espressif/esp-idf #6159: esp_websocket_client_destroy corrupts heap

maikeriva, 2020-11-25:
## Environment

- Development Kit:  Custom board
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.0.2
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?:  Yes (vscode)
- Power Supply: Custom

## Problem Description
Calling `esp_websocket_client_destroy` to close and destroy a websocket connection corrupts heap.

### Expected Behavior
No heap corruption should occur

### Actual Behavior
Corruption occurs

### Steps to reproduce

0. Create a simple project which connects to WiFi, performs a websocket connection, then destroys such connection (code snippet coming soon). Ensure that the call to `esp_websocket_client_destroy` is placed between two `heap_caps_check_integrity_all` calls to detect the heap corruption.
1. In `idf.py menuconfig`, ensure that heap tracing capabilities are enabled.
2. Flash the project. 
3. Project will connect WiFi hotspot, connect to the websocket server, and then crash due to detected heap corruption when destroying the connection.

### Code to reproduce this issue

Coming soon (quite complex to isolate from current project) 

## Debug Logs

Coming soon

## Other items if possible

Coming soon
```
- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)
```

david-cermak, 2020-11-30:
Hi @maikeriva 

Thanks for the report, but please add the code for reproducing this.
I've quickly checked with the standard IDF example, no issue experienced. Please, also note that corrupting some of the ws client internal structures in some way (located in the heap too) might result in a similar behaviour.

Alvin1Zhang, 2021-08-12:
@maikeriva Thanks for reporting and sorry for slow turnaround, would you please help share if any updates for the issue? Thanks.

Alvin1Zhang, 2021-08-31:
@maikeriva Thanks for reporting, will close due to short of feedback, feel free to reopen with more details/updates.

closed 2021-08-31 by Alvin1Zhang