espressif/esp-idf #6161: /examples/system/network_tests crashes at start

nx518, 2020-11-25:
## Environment
- Development Kit:      ESP32-DevKitC
- Kit version:  v4
- Module or chip used:  ESP32-WROOM-32
- IDF version:   v4.2-rc-5-g511965b26
- Build System:   idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description
/examples/system/network_tests crashes at start.

### Expected Behavior
Running network_tests example.

### Actual Behavior
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

### Code to reproduce this issue
/examples/system/network_tests crashes
unmodified

## Debug Logs
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7060
load:0x40078000,len:13212
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (29) boot: ESP-IDF v4.2-rc-5-g511965b26 2nd stage bootloader
I (29) boot: compile time 12:26:18
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0c300 ( 49920) map
I (132) esp_image: segment 1: paddr=0x0001c328 vaddr=0x3ffb0000 size=0x020c0 (  8384) load
I (136) esp_image: segment 2: paddr=0x0001e3f0 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/mx/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (139) esp_image: segment 3: paddr=0x0001e7fc vaddr=0x40080404 size=0x0181c (  6172) load
I (150) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x33de0 (212448) map
0x400d0020: _stext at ??:?

I (237) esp_image: segment 5: paddr=0x00053e08 vaddr=0x40081c20 size=0x094d0 ( 38096) load
0x40081c20: esp_dport_access_stall_other_cpu_start at /home/mx/esp/esp-idf/components/esp32/dport_access.c:106

I (260) boot: Loaded app from partition at offset 0x10000
I (261) boot: Disabling RNG early entropy source...
I (261) cpu_start: Pro cpu up.
I (265) cpu_start: Application information:
I (269) cpu_start: Project name:     net_suite
I (275) cpu_start: App version:      v4.2-rc-5-g511965b26
I (281) cpu_start: Compile time:     Nov 25 2020 12:26:04
I (287) cpu_start: ELF file SHA256:  3e906bcf407c3daa...
I (293) cpu_start: ESP-IDF:          v4.2-rc-5-g511965b26
I (299) cpu_start: Starting app cpu, entry point is 0x400818a0
0x400818a0: call_start_cpu1 at /home/mx/esp/esp-idf/components/esp32/cpu_start.c:287

I (291) cpu_start: App cpu up.
I (309) heap_init: Initializing. RAM available for dynamic allocation:
I (316) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (322) heap_init: At 3FFB3830 len 0002C7D0 (177 KiB): DRAM
I (329) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (335) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (341) heap_init: At 4008B0F0 len 00014F10 (83 KiB): IRAM
I (348) cpu_start: Pro cpu start user code
I (366) spi_flash: detected chip: generic
I (367) spi_flash: flash io: dio
W (367) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (377) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060830  A0      : 0x800e8422  A1      : 0x3ffb8280  
A2      : 0x3ffb8570  A3      : 0x3ffb8d82  A4      : 0x0000002a  A5      : 0x3ffb8d68  
A6      : 0x3ffb7538  A7      : 0x00060023  A8      : 0x80103154  A9      : 0x0000002a  
A10     : 0x3f4030c0  A11     : 0x3ffb8d82  A12     : 0x0000002a  A13     : 0x3ffb8d68  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0xfffffffd:0x3ffb8280 |<-CORRUPTED

ELF file SHA256: 3e906bcf407c3daa

Rebooting...

```

Alvin1Zhang, 2020-11-26:
Thanks for reporting, we will look into.

david-cermak, 2020-11-27:
@nx518 This was fixed in https://github.com/espressif/esp-idf/commit/59471c9f3f6 but hasn't been backported to `v4.2` yet. You can cherry-pick that commit in the meantime. We will provide fixes to v4.2 release branch.

nx518, 2020-11-30:
@david-cermak Sorry, I missed to check it with master before reporting. 
Works fine with the fix.

I´m using `esp_netif_create_ip6_linklocal` and a Router Advertisement (ICMPv6 type 134) via injection (`esp_netif_receive`) to obtain an IPv6 address for this example. Without injection I get 
`E (2400) esp_netif_lwip: nd6 cb: failed to post IP_EVENT_GOT_IP6 (103)`.
Is there a way similiar to `esp_netif_set_ip4_addr` to obtain an IPv6 address without external injection?

david-cermak, 2020-11-30:
Unfortunately there's no such ipv6 API on esp-netif level. It should be possible to set this up directly in lwip, for example
``` c
struct netif * lwip_netif = esp_netif_get_netif_impl(esp_netif_t *esp_netif);
// Set in lwip_netif->ip6_addr[0] (Note, that index 0 indicate it's a local link address)
```

nx518, 2020-12-02:
Setting the IP via the lwip-netif worked, additionally had to set `ip6_addr_state` to `IP6_ADDR_PREFERRED`.

I also found out that `esp_netif_create_ip6_linklocal` does create a linklocal address, just the callback fails because I cannot setup the esp_`event_handler_register` for `IP_EVENT_GOT_IP6`.

So for now im good to go, thank you.

Alvin1Zhang, 2021-01-19:
Thanks for reporting, issue fixed by https://github.com/espressif/esp-idf/commit/bdc914bbf4d7b0302cb29129f12273b7d735b6c4, feel free to reopen.

closed 2021-01-19 by Alvin1Zhang