espressif/esp-idf #6162: [esp32 S2] Allow .bss segment placed in external memory

chegewara, 2020-11-25:
When i am trying to enable `Allow .bss segment placed in external memory` in menuconfig then code does not build with error:
```
/home/chegewara/esp/esp-idf/components/esp_system/port/cpu_start.c:395:13: error: '_ext_ram_bss_start' undeclared (first use in this function); did you mean '_rtc_bss_start'?
     memset(&_ext_ram_bss_start, 0, (&_ext_ram_bss_end - &_ext_ram_bss_start) * sizeof(_ext_ram_bss_start));
             ^~~~~~~~~~~~~~~~~~
             _rtc_bss_start
/home/chegewara/esp/esp-idf/components/esp_system/port/cpu_start.c:395:13: note: each undeclared identifier is reported only once for each function it appears in
/home/chegewara/esp/esp-idf/components/esp_system/port/cpu_start.c:395:38: error: '_ext_ram_bss_end' undeclared (first use in this function); did you mean '_rtc_bss_end'?
     memset(&_ext_ram_bss_start, 0, (&_ext_ram_bss_end - &_ext_ram_bss_start) * sizeof(_ext_ram_bss_start));
                                      ^~~~~~~~~~~~~~~~
                                      _rtc_bss_end
```

IDF: master commit 357a2776032299b8bc4044900a8f1d6950d7ce89
board: S2 wrover module
`SPI RAM access method` - all 3 options

projectgus, 2021-01-06:
Hi @chegewara,

Sorry no one got back to you about this earlier. This config option shouldn't actually be visible for ESP32-S2 as this feature has only been implemented for ESP32. Because ESP32-S2 can map PSRAM in different ways, it's not straightforward to map the variable addresses at build time.

Will update this ticket again once we have an ETA for supporting this feature on ESP32-S2.

sgnes, 2021-07-21:
@projectgus Any update for this issue, do you have any plan to support esp32-s2?

igrr, 2021-07-21:
@sgnes this feature will be supported on S2, the code is in review now. Should be available in the next couple weeks. This issue will be updated when the feature is merged.

pratikpanchal4, 2021-07-29:
Any update on timeline for this feature support on esp32-s2. As timeline was mentioned around couple of weeks, Is it still the same?

Alvin1Zhang, 2021-08-30:
Thanks for reporting and sorry for slow turnaround, the fix is available https://github.com/espressif/esp-idf/commit/b287e01f38f1026a619f180d7bbc1c2d24670f84, feel free to reopen.

closed 2021-08-30 by Alvin1Zhang