espressif/esp-idf #6163: Closing HTTPS sockets returns ESP_FAIL - due to two lwip_close calls

istokm, 2020-11-25:
## Environment

- IDF version:
    v4.3-dev-1721-g9bcf8ad19
- Compiler version:
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

## Problem Description

A call to `esp_http_client_close` - after opening a connection to a **HTTPS** server - results in `ESP_FAIL` & `errno` being set to 9, which is _Bad file number_.
I've tracked the issue to `lwip_close` being indirectly called twice in https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/components/esp-tls/esp_tls.c#L103-L116

Once through https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/components/esp-tls/esp_tls.c#L107 and again directly https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/components/esp-tls/esp_tls.c#L109

What is the correct course of action here? I'd suggest completely removing the call to `close` in `esp_tls_conn_destroy` (L109) as `_esp_tls_conn_delete` already calls the same function with the same parameter. Or, if it's there for the offchance that `tls->is_tls` is false, I'd change the if statement (L108) to `if (tls->sockfd >= 0 && !tls->is_tls)`.

## Debug Logs

```
// call to esp_http_client_close
D (36796) TRANS_SSL: [ssl_close(0x3ffca0a8)]
D (36806) esp-tls: [esp_tls_conn_destroy(0x3ffcb510)]

// call to _esp_tls_conn_delete(tls)
D (36806) esp-tls-mbedtls: [esp_mbedtls_conn_delete(0x3ffcb510)]:
sockets.c: [lwip_close(54)]
sockets.c: [lwip_close]: socket=0x3ffb4a54

D (36826) esp-tls: [esp_tls_conn_destroy] errno=0
D (36826) esp-tls: [esp_tls_conn_destroy] tls->sockfd=54

// call to close(tls->sockfd)
sockets.c: [lwip_close(54)]
sockets.c: [lwip_close]: socket=0x0

D (36846) esp-tls: [esp_tls_conn_destroy] errno=9
D (36846) esp-tls: [esp_tls_conn_destroy] ret=-1
```

Alvin1Zhang, 2020-11-26:
Thanks for reporting, we will look into.

AdityaHPatwardhan, 2020-12-02:
@istokm Thank you for reporting the issue and also suggesting the fix, the fix has already been merged internally. Should be updated on github in a couple of days.

closed 2020-12-13 by espressif-bot