espressif/esp-idf #6164: Timer and Wifi: (Cache disabled but cached memory region accessed)

iCarletto, 2020-11-25:
## Environment

- Development Kit: ESP32-PICO-Kit
- Kit version: v4.1
- Module or chip used: ESP32-PICO-D4
- IDF version: v4.3-dev-1561-g357a27760
- Build System: idf.py
- Compiler version: esp-2020r3-8.4.0
- Operating System: Windows
- Environment type: Plain Command Prompt
- Using an IDE?: Yes (Visual Studio Code)
- Power Supply: USB

## Problem Description

My board needs to start performing some background tasks immediately after power-up at a rate of 1 kHz. Subsequently, it should connect to WiFi for data logging. However, the following `Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)` is generated.

### Expected Behavior
The background task should start before connecting to WiFi.

### Actual Behavior
If the background task with a refresh rate < 10-15 ms is started before the WiFi, then the error is generated. If the task is started after the WiFi connection is established, then no error is generated.

### Steps to reproduce

1. Run the demo code as is to generate the error
2. Switch lines 179 with 180 to avoid the error

### Code to reproduce this issue

[test.zip](https://github.com/espressif/esp-idf/files/5597712/test.zip)

## Debug Logs
```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6844
load:0x40078000,len:14328
load:0x40080400,len:3660
0x40080400: _init at ??:?

entry 0x40080678
I (29) boot: ESP-IDF v4.3-dev-1561-g357a27760 2nd stage bootloader
I (29) boot: compile time 09:52:40
I (29) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x15c44 ( 89156) map
I (148) esp_image: segment 1: paddr=0x00025c6c vaddr=0x3ffb0000 size=0x03f38 ( 16184) load
I (155) esp_image: segment 2: paddr=0x00029bac vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (155) esp_image: segment 3: paddr=0x00029fb8 vaddr=0x40080404 size=0x06060 ( 24672) load
I (174) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x72538 (468280) map
0x400d0020: _stext at ??:?

I (353) esp_image: segment 5: paddr=0x000a2560 vaddr=0x40086464 size=0x0e7ec ( 59372) load
0x40086464: tx_pwctrl_cal at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:2556

I (390) boot: Loaded app from partition at offset 0x10000
I (391) boot: Disabling RNG early entropy source...
I (402) cpu_start: Pro cpu up.
I (402) cpu_start: Starting app cpu, entry point is 0x4008116c
0x4008116c: call_start_cpu1 at C:/esp-idf/components/esp_system/port/cpu_start.c:124

I (0) cpu_start: App cpu up.
I (416) cpu_start: Pro cpu start user code
I (416) cpu_start: cpu freq: 160000000
I (416) cpu_start: Application information:
I (421) cpu_start: Project name:     template-app
I (426) cpu_start: App version:      1
I (431) cpu_start: Compile time:     Nov 25 2020 09:52:13
I (437) cpu_start: ELF file SHA256:  4ea3a3ec3762764e...
I (443) cpu_start: ESP-IDF:          v4.3-dev-1561-g357a27760
I (449) heap_init: Initializing. RAM available for dynamic allocation:
I (456) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (463) heap_init: At 3FFB7DE0 len 00028220 (160 KiB): DRAM
I (469) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (475) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (482) heap_init: At 40094C50 len 0000B3B0 (44 KiB): IRAM
I (489) spi_flash: detected chip: gd
I (492) spi_flash: flash io: dio
W (496) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (510) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (520) Test: Init NVS Flash
Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x400f2288  PS      : 0x00060034  A0      : 0x400824b4  A1      : 0x3ffb07e0
0x400f2288: timer_set_alarm at C:/esp-idf/components/driver/timer.c:188

0x400824b4: _xt_lowint1 at C:/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

A2      : 0x00000000  A3      : 0xff000000  A4      : 0x15122500  A5      : 0x4008774c
0x4008774c: _frxt_int_enter at C:/esp-idf/components/freertos/xtensa/portasm.S:119

A6      : 0x00000020  A7      : 0xffffeffb  A8      : 0x800828d7  A9      : 0x00000001
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00060c20
A14     : 0x00000003  A15     : 0x00060c23  SAR     : 0x0000001e  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe

Backtrace:0x400f2285:0x3ffb07e0 0x400824b1:0x3ffb0800 0x4000bfed:0x3ffbb2e0 0x4008328f:0x3ffbb2f0 0x40083386:0x3ffbb310 0x40083596:0x3ffbb360 0x400d5f81:0x3ffbb390 0x400d6085:0x3ffbb3c0 0x400d615f:0x3ffbb3e0 0x400d61a3:0x3ffbb400 0x400d9863:0x3ffbb420 0x400d7aa6:0x3ffbb440 0x400d6496:0x3ffbb470 0x400d64b6:0x3ffbb490 0x400d5482:0x3ffbb4b0 0x400d567e:0x3ffbb5a0 0x40141d2d:0x3ffbb5c0 0x40089fd5:0x3ffbb5e0
0x400f2285: vPortExitCriticalSafe at C:/esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:268
 (inlined by) timer_set_alarm_value at C:/esp-idf/components/driver/timer.c:171
0x400824b1: _xt_lowint1 at C:/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105
0x4008328f: spi_flash_mmap_init at C:/esp-idf/components/spi_flash/flash_mmap.c:144
0x40083386: spi_flash_mmap_pages at C:/esp-idf/components/spi_flash/flash_mmap.c:211
0x40083596: spi_flash_mmap at C:/esp-idf/components/spi_flash/flash_mmap.c:183
0x400d5f81: load_partitions at C:/esp-idf/components/spi_flash/partition.c:165
0x400d6085: ensure_partitions_loaded at C:/esp-idf/components/spi_flash/partition.c:75 (discriminator 15)
0x400d615f: esp_partition_find at C:/esp-idf/components/spi_flash/partition.c:88
0x400d61a3: esp_partition_find_first at C:/esp-idf/components/spi_flash/partition.c:136
0x400d9863: nvs::partition_lookup::lookup_nvs_partition(char const*, nvs::NVSPartition**) at C:/esp-idf/components/nvs_flash/src/nvs_partition_lookup.cpp:14
0x400d7aa6: nvs::NVSPartitionManager::init_partition(char const*) at C:/esp-idf/components/nvs_flash/src/nvs_partition_manager.cpp:53
0x400d6496: nvs_flash_init_partition at C:/esp-idf/components/nvs_flash/src/nvs_api.cpp:125
0x400d64b6: nvs_flash_init at C:/esp-idf/components/nvs_flash/src/nvs_api.cpp:130
0x400d5482: init_network at c:\esp-codes\test\build/../main/main.c:140
0x400d567e: app_main at c:\esp-codes\test\build/../main/main.c:180
0x40141d2d: main_task at C:/esp-idf/components/freertos/xtensa/port.c:537 (discriminator 2)
0x40089fd5: vPortTaskWrapper at C:/esp-idf/components/freertos/xtensa/port.c:170
```

Alvin1Zhang, 2020-11-26:
Thanks for reporting, we will look into.

YouDONG-ESP, 2020-11-26:
@iCarletto Hi, timer_set_alarm() must be located in IRAM if you call it in an ISR, add IRAM_ATTR attribute to this function will fix the problem.

iCarletto, 2020-11-26:
@YouDONG-ESP `timer_set_alarm()` is already called within an ISR with attribute IRAM_ATTR, where else shall I add it? Can you please provide a modified snippet of the demo code I uploaded?

YouDONG-ESP, 2020-11-26:
@iCarletto, in file timer.c, line 187:
`esp_err_t timer_set_alarm(timer_group_t group_num, timer_idx_t timer_num, timer_alarm_t alarm_en)`
->
`IRAM_ATTR esp_err_t timer_set_alarm(timer_group_t group_num, timer_idx_t timer_num, timer_alarm_t alarm_en)`

iCarletto, 2020-11-26:
@YouDONG-ESP your solution requires the modification of an IDF source file, which I guess I'm not supposed to do. Is this a bug in the IDF or a problem with my code?

mahavirj, 2020-11-26:
@iCarletto 

You have registered interrupt with `ESP_INTR_FLAG_IRAM`, which means that you would like to have your interrupt enabled even if flash cache is disabled (e.g. scenario where flash programming is on-going). If that is the case, kindly refer to following pointers:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/intr_alloc.html#iram-safe-interrupt-handlers
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#iram-safe-interrupt-handlers

In summary, for this case, entire interrupt handler code must be IRAM and data DRAM resident. In sample code that you have shared, you are using APIs from `esp_timer` module which itself is not IRAM resident and hence the exception. 

Potential options:

1. If you do not want interrupt to be enabled when flash cache is disabled, please remove `ESP_INTR_FLAG_IRAM` flag and code should work without any exceptions
2. If you wish to have interrupt to be enabled when flash cache is disabled, then please refactor interrupt handler such that it obeys IRAM/DRAM principles mentioned in above documentation pointers.

Hope this helps.

Mahavir

iCarletto, 2020-11-26:
@mahavirj and @YouDONG-ESP thank you very much for your suggestions and explanations, everything makes sense now. I have to admit that I added that flag to be consistent with the attribute IRAM_ATTR of the ISR, but without really understanding its implications

closed 2020-11-26 by iCarletto