espressif/esp-idf #6165: esp_http_client_read returns -1 instead of 0 in specific case

xire-, 2020-11-25:
## Environment

- IDF version: v4.3-dev-1720-g494a124d9
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No

## Problem Description

```cpp
    // ... http client init and connection open

    char recv_buf[1000];
    int data_read;
    while (true) {
        data_read = esp_http_client_read(http_client, recv_buf, sizeof(recv_buf));
        ESP_LOGI(TAG, "read chuck of %d bytes", data_read);

        if (data_read == 0) {
            ESP_LOGI(TAG, "Download finished");
            break;
        } else if (data_read > 0) {
            // write chunk of data
        } else {
            ESP_LOGE(TAG, "Download failed");
        }
    }

    // ... cleanup code
```

`esp_http_client_read` returns 0 whenever there are no more bytes to read from the HTTP body and that means the download is finished.

If the length of the http body is a multiple of the size of the buffer used by `esp_http_client_read` (1000 in the example above),
the last call will return -1 instead of 0, making the whole process fail.

### Potential fix
Reverting the changes in commit https://github.com/espressif/esp-idf/commit/4c166085fb6c6ee67361a1e7af461887659e6184 fix the issue, 
but i don't understand what that commit fixed in the first place and there might be a better fix.

### Expected Behavior

```
I (10532) main_prog: HTTP status code: 200
I (10602) main_prog: read chuck of 1000 bytes
I (10612) main_prog: read chuck of 1000 bytes
D (10712) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128 
I (10712) main_prog: read chuck of 0 bytes
I (10712) main_prog: Download finished
```

### Actual Behavior

```
I (558542) main_prog: HTTP status code: 200
I (558622) main_prog: read chuck of 1000 bytes
I (558732) main_prog: read chuck of 1000 bytes
D (558752) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128 
I (558752) main_prog: read chuck of -1 bytes
E (558752) main_prog: Download failed
```

### Steps to reproduce

The easiest way to reproduce it is to take the `simple_ota_example` and append '\x00' to the ota binary until it's size is a multiple of the `esp_http_client_read` buffer size which is defined in `esp_https_ota.c` as `DEFAULT_OTA_BUF_SIZE` (289 on my system).

Let the program perform the update and it will fail every time at the end.

Padding one byte more instead always result in success.

Alvin1Zhang, 2020-11-26:
Thanks for reporting, we will look into.

shubhamkulkarni97, 2020-11-27:
@xire-,

I tried to reproduce this issue by performing OTA with file size of 814980, which is a multiple of 289(buffer size for esp_http_client_read). I didn't get any issue and file download is successful. Following are the logs:

```
I (4728) advanced_https_ota_example: Starting Advanced OTA example
I (4858) esp_https_ota: 289 is buffer size
I (4858) esp_https_ota: 814980 is content length
I (4858) esp_https_ota: Starting OTA...
I (4858) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
I (4868) advanced_https_ota_example: Running firmware version: v4.3-dev-1720-g494a124d96-dirty
I (29018) esp_https_ota: Connection closed
I (29018) advanced_https_ota_example: esp_https_ota_perform returned 0
I (29018) advanced_https_ota_example: Complete data received
```

Does the issue happen in specific condition or with a specific server?
Can you try running advanced_https_ota example and check if you observe same issue?

https://github.com/espressif/esp-idf/commit/4c166085fb6c6ee67361a1e7af461887659e6184 commit was added to fix issue when WiFi disconnects while performing OTA Update.

boarchuz, 2020-11-27:
I'm not sure a return value of 0 should be used as the 'response complete' indicator. In this case, the response is complete with that last 1000b read - entering _read again after it's already completed will return ESP_FAIL.

Perhaps make use of esp_http_client_is_complete_data_received?

shubhamkulkarni97, 2020-11-27:
Yes, return value must only be used to check if data is read or if there is some error in esp_http_client_read.

You should use esp_http_client_is_complete_data_received even when return value from esp_http_client_read is 0.

```
data_read = esp_http_client_read(client, recv_buf, sizeof(recv_buf));
if (data_read <= 0) {
    if (esp_http_client_is_complete_data_received(client)) {
        ESP_LOGI(TAG, "Download Finished");
    } else {
        ESP_LOGE(TAG, "Download Failed");
    }
else {
    // write chunk of data
}

shubhamkulkarni97, 2020-11-30:
@xire-, Any updates? Can we close this issue?

xire-, 2020-11-30:
@shubhamkulkarni97 regarding reproducing the issue i was using `openssl s_server -WWW -key ../cert/ca_key.pem -cert ../cert/ca_cert.pem -port 8070` that sends the content chunked.

Using `esp_http_client_is_complete_data_received` fixes the issue for me, thank you.

closed 2020-11-30 by xire-

mjgci, 2024-08-30:
I am having the same issue with this on Windows platform using the general IDF installers and the openssl commands over the included GIT binaries (including the paths as guided here https://github.com/espressif/esp-idf/tree/v5.2.2/examples/system/ota#run-https-server

I keep reading ESP_ERR_HTTP_EAGAIN (-28679) even after the last chunk is received successfully. 

@shubhamkulkarni97 I tried your suggestions but to no avail.

If I use a browser to download the same file (skipping the certificate issue), I can download the file without problems. 

I am using the "native_ota_example".