espressif/esp-idf #6166: Flash Encryption release mode bug

achraf-boussaada, 2020-11-25:
## Environment

- Development Kit:    None
- Kit version (DevKitC): None
- Module or chip used:  ESP32-WROOM-32E
- IDF version : ESP-IDF v4.1-520-gc3324a82a
- Build System:         idf.py
- Compiler version : (crosstool-NG esp-2020r2) 8.2.0
- Operating System:    Linux
- Using an IDE?: yes, visual code studio
- Power Supply:        USB

## Problem Description
I'm encountring a similar problem that's been mentioned [here](https://github.com/espressif/esp-idf/issues/5747#issuecomment-676131639) and a workaround for it was given [here](https://github.com/espressif/esp-idf/issues/5747#issuecomment-678256847)

To go further in details, I'm trying to enable Flash Encryption(release mode) and Secure Boot on an ESP32 chip. I followed the documentation to activate both features through `idf.py menuconfig`. Once I flashed the chip by `idf.py flash monitor` everything seemed fine and the flashing process was successfull except one detail. I got the following error along the logs:
```
E (675) flash_encrypt: Flash encryption settings error: app is configured for RELEASE but efuses are set for DEVELOPMENT
E (677) flash_encrypt: Mismatch found in security options in bootloader menuconfig and efuse settings. Device is not secure.
```
The solution linked above suggests to execute `espefuse.py write_protect_efuse FLASH_CRYPT_CNT` to write protect the efuse from being overwritten. It seemed to work with the setup I had, but since the chip and the ESP-IDF version changed the problem persists. 
Here is a part of my actual efuse table:
```
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 127 R/- (0x7f)
UART_DOWNLOAD_DIS      Disable UART download mode (ESP32 rev3 only)      = 0 R/- (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/- (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 1 R/W (0x1)
JTAG_DISABLE           Disable JTAG                                      = 1 R/W (0x1)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/- (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 1 R/- (0x1)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 1 R/- (0x1)
```
when executing `idf.py monitor` after burning the efuse in question I still get the same error message. Is there any additional steps to be done? or can the error message be ignored?

negativekelvin, 2020-11-25:
It's complaining about DISABLE_DL_ENCRYPT and since it is showing as write protected you can't disable it so treat the message as a warning

achraf-boussaada, 2020-11-26:
@negativekelvin I tried on another chip and burned `DISABLE_DL_ENCRYPT` manually. Doing so solved my problem and the error message is no longer there

closed 2020-11-26 by achraf-boussaada